DFKI-NI/mir_robot #1: About the the laser scan of the model in gazebo

xiangxiang123456, 2018-09-26:
Hello, author! 
When I launch mir_maze_world.launch and click the start button, it seems that the data of the laser scan is not correct. Data has always been shown to be "1.0" or "-inf". I would like to consult you about this problem. Thank you!

mintar, 2018-09-27:
That is strange. In my tests this did not happen. I am currently on vacation and will check this out when I get back.

mintar, 2018-10-08:
I just tried it and everything looks normal. Since you already closed this issue, I assume the problem is solved for you, too.

```
$ roslaunch mir_gazebo mir_maze_world.launch
$ rostopic echo -n1 /scan  
header: 
  seq: 353
  stamp: 
    secs: 28
    nsecs: 320000000
  frame_id: "back_laser_link"
angle_min: -2.3561899662
angle_max: 2.3561899662
angle_increment: 0.00872662942857
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 29.0
ranges: [2.118232250213623, 2.134587049484253, 2.1255643367767334, 2.123037099838257, 2.130375385284424, 2.1107029914855957, 2.128129243850708, 2.140658378601074, 2.1382999420166016, 2.1324946880340576, 2.125605821609497, 2.1385717391967773, 2.148759126663208, 2.143369674682617, 2.135503053665161, 2.1218411922454834, 2.1377458572387695, 2.15207576751709, 2.1421897411346436, 2.1377933025360107, 2.1335928440093994, 2.156099557876587, 2.1604838371276855, 2.1672089099884033, 2.179746627807617, 2.167895793914795, 2.162435293197632, 2.1683316230773926, 2.199090003967285, 2.1949846744537354, 2.1995058059692383, 2.2081403732299805, 2.221367359161377, 2.214350461959839, 2.229611396789551, 2.212204933166504, 2.2290825843811035, 2.2183027267456055, 2.244197130203247, 2.2532339096069336, 2.261500597000122, 2.2581493854522705, 2.2664599418640137, 2.2864959239959717, 2.2825698852539062, 2.2938103675842285, 2.288411855697632, 2.2956020832061768, 2.3420591354370117, 2.3185205459594727, 2.3345391750335693, 2.3495659828186035, 2.362271547317505, 2.3712964057922363, 2.371367931365967, 2.379747152328491, 2.4135091304779053, 2.41119122505188, 2.426671028137207, 2.428227663040161, 2.4512152671813965, 2.4759654998779297, 2.4809587001800537, 2.492811679840088, 2.5055365562438965, 2.516127347946167, 2.520406723022461, 2.5294995307922363, 2.5393424034118652, 2.556929588317871, 2.5906982421875, 2.6021783351898193, 2.624302387237549, 2.6215031147003174, 2.6471946239471436, 2.6544079780578613, 2.6848361492156982, 2.7046000957489014, 2.6968162059783936, 2.717867612838745, 2.7706665992736816, 2.7762415409088135, 2.8030083179473877, 2.797773838043213, 2.8493664264678955, 2.873110294342041, 2.8893210887908936, 2.9039344787597656, 2.931823253631592, 2.937793493270874, 2.978701114654541, 3.008256196975708, 3.0271542072296143, 3.062617540359497, 3.103529691696167, 3.117196559906006, 3.1402077674865723, 3.168971300125122, 3.2094993591308594, 3.251033067703247, 3.2907605171203613, 3.291536331176758, 3.343796730041504, 3.3715360164642334, 3.4274685382843018, 3.456854820251465, 3.4743337631225586, 3.514113187789917, 3.5700695514678955, 3.609253168106079, 3.6601874828338623, 3.68784236907959, 3.72294545173645, 3.7720842361450195, 3.8262665271759033, 3.886568546295166, 3.9290051460266113, 3.977846622467041, 4.05377721786499, 4.1021246910095215, 4.171630382537842, 4.243593692779541, 4.298753261566162, 4.348803997039795, 4.417950630187988, 4.515266418457031, 4.56553840637207, 4.661907196044922, 4.727895736694336, 4.846011638641357, 4.908377647399902, 5.014745235443115, 5.125690937042236, 5.222495079040527, 5.317698955535889, 5.427479267120361, 5.5271077156066895, 5.6706976890563965, 5.793872356414795, 5.918496608734131, 6.054089069366455, 6.209200382232666, 6.368007183074951, 6.523158073425293, 6.715065956115723, 6.903368949890137, 7.085410118103027, 7.312719345092773, 7.508185863494873, 7.741258144378662, 7.994043827056885, 8.259462356567383, 8.526810646057129, 8.812872886657715, 9.152398109436035, 9.486310005187988, 9.77011489868164, 9.761271476745605, 9.742576599121094, 9.728898048400879, 9.69291877746582, 9.685714721679688, 9.655826568603516, 9.659469604492188, 9.639558792114258, 9.637975692749023, 9.615714073181152, 9.590508460998535, 9.615538597106934, 9.610030174255371, 9.59945011138916, 9.592260360717773, 9.577767372131348, 9.57090950012207, 4.01645565032959, 4.009784698486328, 4.00575590133667, 4.003983020782471, 4.013416767120361, 4.016206741333008, 4.00447416305542, 4.001724720001221, 4.007286071777344, 3.998131036758423, 4.00667142868042, 4.029260635375977, 3.9997613430023193, 4.007713317871094, 4.03181791305542, 4.015185356140137, 4.021133899688721, 4.031152725219727, 4.024902820587158, 4.024942874908447, 4.029960632324219, 4.052312850952148, 4.045283794403076, 4.044433116912842, 4.0491204261779785, 4.05939245223999, 4.054620265960693, 4.057176113128662, 4.068199157714844, 4.086193084716797, 4.083792686462402, 4.108575820922852, 4.101598262786865, 4.100880146026611, 4.093325614929199, 4.108699798583984, 4.140781402587891, 4.144482135772705, 4.153148651123047, 4.167515277862549, 4.16301155090332, 4.19837760925293, 4.189143180847168, 4.213865756988525, 4.217899322509766, 4.246091842651367, 4.2614569664001465, 4.252127170562744, 4.270252227783203, 4.27981424331665, 4.306707859039307, 4.309825897216797, 4.338870525360107, 4.3686394691467285, 4.378902435302734, 4.356312274932861, 4.388303756713867, 4.418210506439209, 4.455902576446533, 4.454675674438477, 4.467686653137207, 4.478784084320068, 4.51624059677124, 4.5403056144714355, 4.541219711303711, 4.586543560028076, 4.596199989318848, 4.6231913566589355, 4.642734050750732, 4.678833484649658, 4.695813179016113, 4.720130443572998, 4.759897232055664, 4.781935214996338, 4.812685966491699, 4.830791473388672, 4.8516130447387695, 4.881672382354736, 4.925502777099609, 4.941622734069824, 4.976891040802002, 5.001030445098877, 5.0398759841918945, 5.085338115692139, 5.125948429107666, 5.162121772766113, 5.16978645324707, 5.222858905792236, 5.2 [... cut ...] 0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

xiangxiang123456, 2018-10-08:
> I just tried it and everything looks normal. Since you already closed this issue, I assume the problem is solved for you, too.
>
> ```
> $ roslaunch mir_gazebo mir_maze_world.launch
> $ rostopic echo -n1 [...]

It seems something wrong with my ros, it can run in my partner's computer. Anyway, thank you very much.

kajep09, 2019-05-16:
I have the same problem when building everything from source and running the tutorials as explained in the readme. There is no usable data on any of the scan topics (see output below).
I thought it might be a gazebo issue, so I replaced my default gazebo7 with a new gazebo9, but the problem is the same.

```
$ rostopic echo -n1 /scan 
header: 
  seq: 1476
  stamp: 
    secs: 118
    nsecs: 160000000
  frame_id: "front_laser_link"
angle_min: -2.3561899662
angle_max: 2.3561899662
angle_increment: 0.00872662942857
time_increment: 0.0
scan_time: 0.0
range_min: 0.0500000007451
range_max: 29.0
ranges: [-inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf, -inf]
intensities: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1 [... cut ...] 0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
---
```

mintar, 2019-05-16:
I assume it's an incompatibility between Gazebo and your graphics card. Which graphics card is it? I'm always running Gazebo on PCs with a NVIDIA card, since Gazebo sometimes produces these weird bugs on other graphics cards.

I suspect the following two things:

1. The problem is not specific to the mir_gazebo package; you'll probably get it with any other laser scanner Gazebo simulation as well on your PC.
2. The problem will probably go away if you run the same software on a PC with an NVIDIA graphics card.

kajep09, 2019-05-29:
Hi mintar,

Thank you for the quick response. It seems like you were right about the graphics card / drivers. I tried on another laptop and there it works fine, which prompted me to update my nvidia drivers on my own laptop afterwards - and now it works here too.