DFKI-NI/mir_robot #2: Unable to locate package

e-compton, 2018-10-08:
Hi I'm trying to install the binary version of this package, but can't seem to find it in the ubuntu or debian package repositories. Is there a repository that needs to be added to my source list?

Have tried:
```
sudo apt install ros-indigo-mir-robot
sudo apt install ros-jade-mir-robot
```

closed 2018-10-08 by mintar

Hazem9806, 2021-04-28:
Is it available for ROS noetic?
I tried 
`sudo apt install ros-noetic-mir-robot`
but it doesn't work

mintar, 2021-04-28:
Yes, it is available. Your command is also correct, I just verified that it works. Make sure that you're running Ubuntu 20.04, that you've enabled the apt repository and that you've run `sudo apt update` before.

commit 8325ebf0 (2018-10-08): Update README

Closes #2.