DFKI-NI/mir_robot #4: The negative average delay

thanhthtran, 2018-11-07:
Hi,
i try to connect to the real MIR 100 robot thourgh the driver. However, it shows the message in tf_monitor.

Frame: back_laser_link published by unknown_publisher Average Delay: -3597.61 Max Delay: 0
Frame: base_footprint published by unknown_publisher Average Delay: -3597.49 Max Delay: 0
Frame: base_link published by unknown_publisher Average Delay: -3597.61 Max Delay: 0
Frame: front_laser_link published by unknown_publisher Average Delay: -3597.61 Max Delay: 0
Frame: imu_link published by unknown_publisher Average Delay: -3597.61 Max Delay: 0
Frame: odom published by unknown_publisher Average Delay: -3597.67 Max Delay: 0

All Broadcasters:
Node: unknown_publisher 106.839 Hz, Average Delay: -3597.58 Max Delay: 0

i try to go to the settings on the web interface to set the current time of the robot but it does not work. As a result, there is no trasnformation from some frame in RVIZ

thanhthtran, 2018-11-07:
Thank you for your response. I following your instruction to change the time. it know just only have -0.2s. However, when i run the mir_driver launch file. CHecking with the tf monitor, there are a big delay for some frame. Is it normal?
Frames:
Frame: /back_laser_link published by unknown_publisher Average Delay: 54.5959 Max Delay: 55.2053
Frame: /base_footprint published by unknown_publisher Average Delay: 68.0524 Max Delay: 71.866
Frame: /base_link published by unknown_publisher Average Delay: 55.2006 Max Delay: 55.504
Frame: /bl_caster_rotation_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /bl_caster_wheel_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /br_caster_rotation_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /br_caster_wheel_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /fl_caster_rotation_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /fl_caster_wheel_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /fr_caster_rotation_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /fr_caster_wheel_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /front_laser_link published by unknown_publisher Average Delay: 57.408 Max Delay: 71.1888
Frame: /imu_link published by unknown_publisher Average Delay: 54.9379 Max Delay: 55.4139
Frame: /left_wheel_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: /odom published by unknown_publisher Average Delay: 71.0557 Max Delay: 73.0519
Frame: /right_wheel_link published by unknown_publisher Average Delay: 67.7905 Max Delay: 71.9867
Frame: back_laser_link published by unknown_publisher Average Delay: -0.278159 Max Delay: 0
Frame: base_footprint published by unknown_publisher Average Delay: -0.157837 Max Delay: 0
Frame: base_link published by unknown_publisher Average Delay: -0.278183 Max Delay: 0
Frame: bl_caster_rotation_link published by unknown_publisher Average Delay: 0.017524 Max Delay: 0.0491508
Frame: bl_caster_wheel_link published by unknown_publisher Average Delay: 0.0175249 Max Delay: 0.0491514
Frame: br_caster_rotation_link published by unknown_publisher Average Delay: 0.0175256 Max Delay: 0.0491518
Frame: br_caster_wheel_link published by unknown_publisher Average Delay: 0.0175262 Max Delay: 0.0491521
Frame: fl_caster_rotation_link published by unknown_publisher Average Delay: 0.0175267 Max Delay: 0.0491524
Frame: fl_caster_wheel_link published by unknown_publisher Average Delay: 0.017527 Max Delay: 0.0491526
Frame: fr_caster_rotation_link published by unknown_publisher Average Delay: 0.0175273 Max Delay: 0.0491529
Frame: fr_caster_wheel_link published by unknown_publisher Average Delay: 0.0175279 Max Delay: 0.0491532
Frame: front_laser_link published by unknown_publisher Average Delay: -0.278073 Max Delay: 0
Frame: imu_frame published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: imu_link published by unknown_publisher Average Delay: -0.277968 Max Delay: 0
Frame: left_wheel_link published by unknown_publisher Average Delay: 0.0175283 Max Delay: 0.0491536
Frame: odom published by unknown_publisher Average Delay: 24.6529 Max Delay: 58.7
Frame: right_wheel_link published by unknown_publisher Average Delay: 0.0175292 Max Delay: 0.049154
Frame: surface published by unknown_publisher(static) Average Delay: 0 Max Delay: 0

All Broadcasters:
Node: unknown_publisher 181.349 Hz, Average Delay: 14.0561 Max Delay: 73.0519
Node: unknown_publisher(static) 1e+08 Hz, Average Delay: 0 Max Delay: 0

thanhthtran, 2018-11-07:
And in the terminal where the mir_driver launch file run it shows this WARN:
[WARN] [1541590273.429593]: Could not process inbound connection: topic types do not match: [*] vs. [mir_actions/RelativeMoveActionFeedback]{'topic': '/relative_move_action/feedback', 'tcp_nodelay': '0', 'md5sum': '429713dfca2cc916179fc3088546a87b', 'type': '*', 'callerid': '/mir_auto_bagger'}

mintar, 2018-11-07:
Regarding your second error - you can comment out the following lines for now, if necessary:

https://github.com/dfki-ric/mir_robot/blob/54be543600058477268c93f4be25f51050749de4/mir_driver/nodes/mir_bridge.py#L185-L187

Let's first try to solve your other problem, then come back to this.

thanhthtran, 2018-11-07:
Thank you for your response again. Would you have any idea to solve the tf and delay ?

mintar, 2018-11-07:
Regarding your other problem (big delays in tf_monitor): No, this is not normal. Any delay bigger than 0.5 seconds or so will cause big problems; preferably the delay should be even lower.

Something very funny is going on: You have TF frames without a leading '/', which mostly don't have a big TF delay, and ones with a leading '/' that have a huge delay. I'll try this on our MiR and report back later.

thanhthtran, 2018-11-07:
Hi again,
I would give you answer tomorrow when i come back to my university ‘s lab.

thanhthtran, 2018-11-08:
A quick update: after setting the time again on the mir and change to the right public for odom ( in my case it is /odom not /odom_comb), the tf delay drops to 0.2-0.3s for all frame. Thank you for your help :)

mintar, 2018-11-08:
Glad to be of help! :)

closed 2018-11-08 by mintar