DFKI-NI/mir_robot #9: the execution can't succeed.

marooncn, 2018-12-23:
I do the simulation step by step, but the execution can't succeed as the following:
[image]
I think there maybe some error in the model or parameter setting, Is it?

mintar, 2018-12-23:
Thank you for your bug report!

That seems to be a bug in the [eband_local_planner](http://wiki.ros.org/eband_local_planner): It tries to take a shortcut through the wall. I don't know what causes this. You can try one of the other local planners by changing the `local_planner:=eband` parameter. Local planner can be either `dwa`, `eband`, `base`, `teb` or `pose`. However, the other local planners have different problems, and eband is the one that worked best for me.

In general, the local planner is the part of the navigation config that I'm not yet happy with. If you find better parameters for any of the local planners, please let me know.

marooncn, 2018-12-23:
Hi, @mintar, I change the local planner to dwa, but it still doesn't work:
[image]
It seems like two other rectangular body tries to move rather the robot self. So it's judged to stuck while the robot is free.

mintar, 2018-12-25:
That's weird, I've never seen anything like that. What are those "extension bodies"? Maybe something goes wrong with your simulation of the laser scans.

mintar, 2019-01-22:
This was probably fixed by #13 .

closed 2019-01-22 by mintar