DFKI-NI/mir_robot #14: Some floor camera topics missing

chbloca, 2019-01-22:
Hello there! I suggest you an improvement to your package.

Here is a list of the topics related to floor camera obtained with
`roslaunch mir_driver mir.launch`

> /camera_floor/background
/camera_floor/depth/parameter_descriptions
/camera_floor/depth/parameter_updates
/camera_floor/depth/points
/camera_floor/filter/parameter_descriptions
/camera_floor/filter/parameter_updates
/camera_floor/floor
/camera_floor/obstacles
/camera_floor/transform/parameter_descriptions
/camera_floor/transform/parameter_updates

And here without your launcher, using
`export ROS_MASTER_URI=http://192.168.12.20:11311`

> /camera_floor/background
/camera_floor/camera_info
/camera_floor/driver/color/camera_info
/camera_floor/driver/color/image_raw
/camera_floor/driver/color/image_raw/compressed
/camera_floor/driver/color/image_raw/compressed/parameter_descriptions
/camera_floor/driver/color/image_raw/compressed/parameter_updates
/camera_floor/driver/depth/camera_info
/camera_floor/driver/depth/color/points
/camera_floor/driver/depth/image_rect_raw
/camera_floor/driver/depth/image_rect_raw/compressed
/camera_floor/driver/depth/image_rect_raw/compressed/parameter_descriptions
/camera_floor/driver/depth/image_rect_raw/compressed/parameter_updates
/camera_floor/driver/extrinsics/depth_to_color
/camera_floor/driver/parameter_descriptions
/camera_floor/driver/parameter_updates
/camera_floor/filter/parameter_descriptions
/camera_floor/filter/parameter_updates
/camera_floor/filter/visualization_marker
/camera_floor/floor
/camera_floor/obstacles
/camera_floor/transform/parameter_descriptions
/camera_floor/transform/parameter_updates
/camera_top/background
/camera_top/obstacles

There are missing topics being recovered from your bridging launcher.

chbloca, 2019-01-22:
Some topics only start publishing after the robot is executing a mission, going from A to B.

chbloca, 2019-02-08:
Actually, I cannot visualize on RViz the camera topics with your launcher. Is there any way to solve this issue?

mintar, 2019-03-19:
I haven't gotten around to adding the topics yet and testing that everything works. You can easily do it yourself by adding some lines around here:

https://github.com/dfki-ric/mir_robot/blob/72fe47e514753e3f9ca3a7949f7aec76c816e629/mir_driver/nodes/mir_bridge.py#L112

chbloca, 2019-03-19:
Here are the additions, everything is working OK:

```
from sensor_msgs.msg import Imu, LaserScan, PointCloud2, Range, Image, CompressedImage, CameraInfo

TopicConfig('camera_floor/driver/color/camera_info', CameraInfo),
              TopicConfig('camera_floor/driver/color/image_raw', Image),
              TopicConfig('camera_floor/driver/color/image_raw/compressed', CompressedImage),
              TopicConfig('camera_floor/driver/color/image_raw/compressed/parameter_descriptions', ConfigDescription),
              TopicConfig('camera_floor/driver/color/image_raw/compressed/parameter_updates', Config),
              TopicConfig('camera_floor/driver/depth/camera_info', CameraInfo),
              TopicConfig('camera_floor/driver/depth/color/points', PointCloud2),
              TopicConfig('camera_floor/driver/depth/image_rect_raw', Image),
              TopicConfig('camera_floor/driver/depth/image_rect_raw/compressed', CompressedImage),
              TopicConfig('camera_floor/driver/depth/image_rect_raw/compressed/parameter_descriptions', ConfigDescription),
              TopicConfig('camera_floor/driver/depth/image_rect_raw/compressed/parameter_updates', Config),
              TopicConfig('camera_floor/depth/parameter_descriptions', ConfigDescription),
              TopicConfig('camera_floor/depth/parameter_updates', Config),
              TopicConfig('camera_floor/depth/points', PointCloud2),
              TopicConfig('camera_floor/driver/parameter_descriptions', ConfigDescription),
              TopicConfig('camera_floor/driver/parameter_updates', Config),
```

chbloca, 2019-03-23:
> Here are the additions, everything is working OK:
>
> ```
> from sensor_msgs.msg import Imu, LaserScan, PointCloud2, Range, Image, CompressedImage, CameraInfo [...]

However, I notice a more lagged connection

mintar, 2019-03-26:
I'm reopening this so I remember to put the new topics in at some point.

chbloca, 2019-04-08:
By the way, I tried to implement this and an error shows up:
[image]
Just in case I did a `chmod +x mir_bridge.py` and a catkin_make. It looks like this:
```
#!/usr/bin/env python
import rospy

import copy
import sys
from collections import Iterable

from mir_driver import rosbridge
from rospy_message_converter import message_converter

from actionlib_msgs.msg import GoalID, GoalStatusArray
from diagnostic_msgs.msg import DiagnosticArray, DiagnosticStatus
from dynamic_reconfigure.msg import Config, ConfigDescription
from geometry_msgs.msg import PolygonStamped, Pose, PoseArray, PoseStamped, PoseWithCovarianceStamped, Twist
from mir_actions.msg import *
from mir_msgs.msg import *
from move_base_msgs.msg import MoveBaseActionFeedback, MoveBaseActionGoal, MoveBaseActionResult, MoveBaseFeedback, MoveBaseResult
from nav_msgs.msg import GridCells, MapMetaData, OccupancyGrid, Odometry, Path
from rosgraph_msgs.msg import Log
from sensor_msgs.msg import Imu, LaserScan, PointCloud2, Range, Image, CompressedImage, CameraInfo
from std_msgs.msg import Float64, String
from tf.msg import tfMessage
from visualization_msgs.msg import Marker, MarkerArray

tf_prefix = ''

class TopicConfig(object):
    def __init__(self, topic, topic_type, latch=False, dict_filter=None):
        self.topic = topic
        self.topic_type = topic_type
        self.latch = latch
        self.dict_filter = dict_filter

# remap mir_actions/MirMoveBaseAction => move_base_msgs/MoveBaseAction
def _move_base_feedback_dict_filter(msg_dict):
    # filter out slots from the dict that are not in our message definition
    # e.g., MiRMoveBaseFeedback has the field "state", which MoveBaseFeedback doesn't
    filtered_msg_dict = copy.deepcopy(msg_dict)
    filtered_msg_dict['feedback'] = {key: msg_dict['feedback'][key] for key in MoveBaseFeedback.__slots__}
    return filtered_msg_dict

def _move_base_result_dict_filter(msg_dict):
    filtered_msg_dict = copy.deepcopy(msg_dict)
    filtered_msg_dict['result'] = {key: msg_dict['result'][key] for key in MoveBaseResult.__slots__}
    return filtered_msg_dict

def _tf_dict_filter(msg_dict):
    filtered_msg_dict = copy.deepcopy(msg_dict)
    for transform in filtered_msg_dict['transforms']:
        transform['child_frame_id'] = tf_prefix + '/' + transform['child_frame_id'].strip('/')
    return filtered_msg_dict

def _prepend_tf_prefix_dict_filter(msg_dict):
    #filtered_msg_dict = copy.deepcopy(msg_dict)
    if not isinstance(msg_dict, dict):   # can happen during recursion
        return
    for (key, value) in msg_dict.iteritems():
        if key == 'header':
            try:
                # prepend frame_id
                frame_id = value['frame_id'].strip('/')
                if (frame_id != 'map'):
                    # prepend tf_prefix, then remove leading '/' (e.g., when tf_prefix is empty)
                    value['frame_id'] = (tf_prefix + '/' + frame_id).strip('/')
                else:
                    value['frame_id'] = frame_id
[... 274 lines cut ...]

        print 'Publishers:'
        for (topic_name, topic_type, has_publishers, has_subscribers) in topics:
            if has_publishers:
                print ' * %s [%s]' % (topic_name, topic_type)

        print '\nSubscribers:'
        for (topic_name, topic_type, has_publishers, has_subscribers) in topics:
            if has_subscribers:
                print ' * %s [%s]' % (topic_name, topic_type)

        return topics

def main():
    rospy.init_node('mir_bridge')
    MiR100Bridge()
    rospy.spin()

if __name__ == '__main__':
    try:
        main()
    except rospy.ROSInterruptException:
        pass
```
I have no idea where the error might come from..

mintar, 2021-06-11:
I have just pushed a large update to the `melodic` and `noetic` branches. This issue is fixed now.

closed 2021-06-11 by mintar