DFKI-NI/mir_robot #17: topic 'mir_status' is not published by the MiR!

dzoberti, 2019-02-19:
I am working on MIR100 with software 2.2.2. When starting `mir.launch`, I get a warning that the `mir_status` topic is not published by MiR and it is true. However I want to get this vital information, for example information about the battery. I have noticed that there exists a topic called `robot_status` with the message type  `mirMsgs/RobotStatus` which I cannot read:

```
$ rostopic echo /robot_status
ERROR: Cannot load message class for [mirMsgs/RobotStatus]. Are your messages built?
```
I can't get further than this and I would appreciate any help.

mintar, 2019-02-19:
I'm currently still working with MiR software 1.9, but am planning to upgrade soon. :)

I've pushed the messages from version 2.0.2 (should hopefully be the same as 2.2.2) to a new branch called `mir_msgs-2.0.2` in my personal fork:

https://github.com/mintar/mir_robot/tree/mir_msgs-2.0.2

I've renamed `mirMsgs` to `mir_msgs` to conform with standard naming practices. This means that if you connect to the MiR's roscore directly, you will still get the same error message about messages not being built. However, if you use the `mir_bridge.py` instead, it should work if you first add the following line:

```python
 TopicConfig('robot_status', RobotStatus),
```

... around here:

https://github.com/dfki-ric/mir_robot/blob/0041b41bb661d32a34c266d0535acf1c78cf61b9/mir_driver/nodes/mir_bridge.py#L145

Once I update to MiR software 2, I'll also update the code in this repo properly.

dzoberti, 2019-02-20:
It works!! Thank you so much.

closed 2019-02-20 by dzoberti