DFKI-NI/mir_robot #20: Client [/mir_auto_bagger] wants topic /move_base/goal to have datatype/md5sum

chbloca, 2019-03-14:
Hello Mintar.
When I use NavGoal in RViz when executing gmapping and move_base nodes it comes the following error:

> [ERROR] [1552579781.696460056]: Client [/mir_auto_bagger] wants topic /move_base/goal to have datatype/md5sum [*/8ac6f5411618f50134619b87e4244699], but our version has [move_base_msgs/MoveBaseActionGoal/660d6895a1b9a16dce51fbdd9a64a56b]. Dropping connection.

The procedure I follow is:

`export ROS_MASTER_URI=http://192.168.12.20:11311`
`export ROS_IP=192.168.12.246`

Then I roslaunch the following script to initialize mainly RViz, gmapping and move_base nodes.
Then in RViz I try to use NavGoal but it does not work due to the error that appeared in a message on the same terminal as the roslaunch command. I need those nodes in order to run another package.

```
<launch>

  <param name="use_sim_time" value="false" />

  <node pkg="laser_filters" type="scan_to_scan_filter_chain" name="laser_filter">
      <remap from="scan" to="f_scan" />
      <rosparam command="load" file="$(find laser_filters)/laserscan_filter.yaml" />
  </node>

  <arg name="scan_topic" default="/scan_filtered" />
  <node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">

    <param name="odom_frame" value="odom"/>
    <param name="base_frame" value="base_link"/>
    <param name="map_frame" value="map"/>

    <!-- Process 1 out of every this many scans (set it to a higher number to skip more scans)  -->
    <param name="throttle_scans" value="1"/>

    <param name="map_update_interval" value="2.0"/> <!-- default: 5.0 -->

    <!-- The maximum usable range of the laser. A beam is cropped to this value.  -->
    <param name="maxUrange" value="5.0"/>

    <!-- The maximum range of the sensor. If regions with no obstacles within the range of the sensor should appear as free space in the map, set maxUrange < maximum range of the real sensor <= maxRange -->
    <param name="maxRange" value="10.0"/>

    <param name="sigma" value="0.05"/>
    <param name="kernelSize" value="1"/>
    <param name="lstep" value="0.05"/>
    <param name="astep" value="0.05"/>
    <param name="iterations" value="5"/>
    <param name="lsigma" value="0.075"/>
    <param name="ogain" value="3.0"/>
    <param name="minimumScore" value="0.0"/>
    <!-- Number of beams to skip in each scan. -->
    <param name="lskip" value="0"/>

    <param name="srr" value="0.01"/>
    <param name="srt" value="0.02"/>
    <param name="str" value="0.01"/>
    <param name="stt" value="0.02"/>

    <!-- Process a scan each time the robot translates this far  -->
    <param name="linearUpdate" value="0.1"/>

    <!-- Process a scan each time the robot rotates this far  -->
    <param name="angularUpdate" value="0.05"/>

    <param name="temporalUpdate" value="0.1"/>
    <param name="resampleThreshold" value="0.5"/>

    <!-- Number of particles in the filter. default 30        -->
    <param name="particles" value="10"/>

    <!-- Initial map size  -->
    <param name="xmin" value="-5.0"/>
    <param name="ymin" value="-5.0"/>
    <param name="xmax" value="5.0"/>
    <param name="ymax" value="5.0"/>

    <!-- Processing parameters (resolution of the map)  -->
    <param name="delta" value="0.1"/>

    <param name="llsamplerange" value="0.01"/>
    <param name="llsamplestep" value="0.01"/>
    <param name="lasamplerange" value="0.005"/>
    <param name="lasamplestep" value="0.005"/>

    <remap from="scan" to="$(arg scan_topic)"/>
  </node>

  <arg name="namespace" value=""/>
  <node pkg="move_base" type="move_base" respawn="false" name="move_base_node" output="screen">
    <param name="footprint_padding" value="0.01" />
    <param name="controller_frequency" value="5.0" />
    <param name="controller_patience" value="3.0" />
    <param name="oscillation_timeout" value="30.0" />
    <param name="oscillation_distance" value="0.5" />
    <param name="planner_patience" value="1" />
    <param name="controller_patience" value="1" /> 
    <remap from="cmd_vel" to="mobile_base/commands/velocity"/>
    <param name="recovery_behavior_enabled" value="false" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/base_local_planner_params.yaml" command="load" />  
    <param name="global_costmap/global_frame" value="$(arg namespace)/map"/>
    <param name="global_costmap/robot_base_frame" value="$(arg namespace)/base_link"/>
    <param name="global_costmap/laser_scan_sensor/sensor_frame" value="/$(arg namespace)/base_laser_link"/>
    <param name="global_costmap/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>    
    <param name="local_costmap/global_frame" value="$(arg namespace)/odom"/>
    <param name="local_costmap/robot_base_frame" value="$(arg namespace)/base_link"/>
    <param name="local_costmap/laser_scan_sensor/sensor_frame" value="$(arg namespace)/front_laser_link"/>
    <param name="local_costmap/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>
    <param name="local_costmap/obstacle_layer/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>
  </node>

  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find rrt_exploration_tutorials)/launch/includes/rviz_config/simple.rviz">
  </node>

</launch>
```

Is there any idea of why this happens?

In addition, why by using `roslaunch mir_driver mir.launch`, the opening of RViz is much quicker than just by using `export ROS_MASTER_URI=http://192.168.12.20:11311` and `export ROS_IP=192.168.12.246`?
By the way, I am trying to access the topics from the camera by using `roslaunch mir_driver mir.launch` but they are not available for some reason. Although by using `export ROS_MASTER_URI=http://192.168.12.20:11311` and `export ROS_IP=192.168.12.246` they indeed are available.

Thanks for your time.

mintar, 2019-03-19:
If you connect directly to the MiR ROS master (192.168.12.20), the ROS message definitions in your external PC have to be *exactly* the same as on the internal MiR PC. That is what causes the error message that you're getting (about the wrong md5sum).

If you use the `mir_driver`, this does not happen, because the `mir_driver/mir_bridge.py` node doesn't subscribe directly to the internal PC's ROS master; instead, all messages are converted into a Python dictionary and back, so minor differences in the message definitions don't matter.

TL;DR: The only practical way to avoid the md5sum error is to use the `mir_driver`.

closed 2019-03-19 by mintar

chbloca, 2019-03-19:
All right, thank you very much for your info 👍

mintar, 2019-04-10:
I have just updated the mir_actions and mir_msgs packages to the MiR software version 2.3.1. The MirMoveBase action now looks like this:

https://github.com/dfki-ric/mir_robot/blob/7682e991e0efb66b1fcfbf0a4c2b42f9a837235a/mir_actions/action/MirMoveBase.action

This action definition hasn't changed between MiR software versions 1.9.13 and 2.3.1, so as long as you use any MiR software version within that range, you should be ok. In 2.4.0, the action has changed, which causes problems.

If you use this new action definition, the message `mir_actions/MirMoveBaseActionGoal` has md5sum `8ac6f5411618f50134619b87e4244699`, which is exactly what the client wants (see the error message in your first post).

This is what you need to do if you want to send move_base goals to the MiR directly, without using the mir_bridge:

* clone and compile the latest version of this repo
* change your move_base client so that it sends `mir_actions/MirMoveBase`, not `move_base_msgs/MoveBase` action goals. The only field in that message that you need to fill out is `target_pose`, just like in the original MoveBase action.

chbloca, 2019-04-10:
All right, thank you very much for the support!
How can I implement your second point? How can I modify:
```
<node pkg="move_base" type="move_base" respawn="false" name="move_base_node" output="screen">
    <param name="footprint_padding" value="0.01" />
    <param name="controller_frequency" value="5.0" />
    <param name="controller_patience" value="3.0" />
    <param name="oscillation_timeout" value="30.0" />
    <param name="oscillation_distance" value="0.5" />
    <param name="planner_patience" value="1" />
    <param name="controller_patience" value="1" /> 
    <remap from="cmd_vel" to="mobile_base/commands/velocity"/>
    <param name="recovery_behavior_enabled" value="false" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/base_local_planner_params.yaml" command="load" />  
    <param name="global_costmap/global_frame" value="$(arg namespace)/map"/>
    <param name="global_costmap/robot_base_frame" value="$(arg namespace)/base_link"/>
    <param name="global_costmap/laser_scan_sensor/sensor_frame" value="/$(arg namespace)/base_laser_link"/>
    <param name="global_costmap/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>    
    <param name="local_costmap/global_frame" value="$(arg namespace)/odom"/>
    <param name="local_costmap/robot_base_frame" value="$(arg namespace)/base_link"/>
    <param name="local_costmap/laser_scan_sensor/sensor_frame" value="$(arg namespace)/front_laser_link"/>
    <param name="local_costmap/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>
    <param name="local_costmap/obstacle_layer/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>
  </node>
```
Because this is the only node which is accessing the move_base node.

mintar, 2019-04-10:
I don't understand your question. Also, I guess this is out of scope for this package.