DFKI-NI/mir_robot #21: How to use ros to send a goal, which is defined in mir's map?

flymaxty, 2019-03-25:
Hi all,
I'm trying to use rviz to send a goal to real mir robot, but it does not move. Looks like mir can not use ros to set a goal directly. Is there any way to send a goal to real mir robot in ros? Like mir mission etc..

Thank you!

mintar, 2019-03-25:
It's possible to set a goal directly from RViz.

* Follow the steps described under "Running the driver on the real robot" in the README. Especially, you have to make sure that you launch the "Planner" (i.e., `move_base` and `amcl` on the robot). If it isn't started, when you start the mir_driver, you'll get a warning that the topic `move_base_simple/goal` is not yet subscribed to by the MiR.
* Start RViz, for example like this: `rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz`
* In RViz, go to "Panels -> Tool Properties" and make sure that "2D Nav Goal -> Topic" is set to `move_base_simple/goal`.
* Run `rostopic info move_base_simple/goal` and make sure that the MiR driver is subscribed to that topic, and RViz is publishing to it.
* In RViz, make sure that "Global Options -> Fixed Frame" is set to "map".
* In RViz, click "2D Pose Estimate", then click on the map.

rsr152, 2019-04-23:
Hi all,

I am new to the ROS framework (~2 weeks). I have been able to install ROS Kinetic in Ubuntu 16.4, run the MiR simulation in Gazebo (thanks @mintar!) and am now looking to send velocity and pose commands to the physical robot:

1. I have been able to publish to the /cmd_vel topic from the command line and move the robot forward/back and turn it clockwise/counter-clockwise using commands such as:
_rostopic pub /cmd_vel geometry_msgs/Twist -r 1 -- '[0.05, 0.0, 0.0]' '[0.0, 0.0, 0.0]'_

2. How can I get the robot to get to a particular pose (x, y, theta) from the command line?

3. In order to execute @mintar's suggestion above re: rviz, I assume I will have to map out the environment (room) first.  Is that right and if so, what is the best way to do that?

4. Finally, the robot's **position** in real life is consistent with the position returned by /robot_pose but its **orientation** is not.  Any thoughts on how to calibrate it accurately?

I would appreciate any help.  Thank you in advance!

rsr152, 2019-05-16:
Hi all,

I am seeing the same behavior -- the (real) robot does not move after clicking on a destination in RViz.  All the settings, publications and subscriptions are in line with what @mintar has outlined above (screenshots attached).  The map displayed in RViz as well as the position of the robot in the map are accurate.

The only step I did differently was:
In RViz, I clicked "2D Nav Goal", then click on the map, as I have been doing in Gazebo.
I have not yet tried clicking on "2D Pose Estimate" and then clicking on the map because I wanted to be cautious and not have the 65kg robot run crazy in the lab :)  

Appreciate any help. As always, thank you!

Rahul

[image]
[image]
[image]
[image]

mintar, 2019-05-22:
1. Which MiR software version do you use?
2. Are the clocks synchronized? You can check by running `rosrun tf tf_monitor`, as described in the README.

mpatil2716, 2019-05-22:
Hi all,

I am also seeing the same behavior too -- i am using simulation robot and it does not move after clicking on a destination in RViz. I followed all the instruction given in README and all the settings, publications and subscriptions are in line with what @mintar has outlined above (screenshots attached). The map displayed in RViz as well as the position of the robot in the map are accurate.

[image]
[image]
[image]

Also is anyone has MiR500 or 1000 URDF package?
Appreciate any help. As always, thank you!

rsr152, 2019-05-30:
> 1. Which MiR software version do you use?
> 2. Are the clocks synchronized? You can check by running `rosrun tf tf_monitor`, as described in the README.

1. I use MiR software 2.6.2
2. I checked the clocks synchronization and they seem to be OK with a max delay of 0.08s (screenshot attached)

[image]

mintar, 2019-06-03:
Could you give it a try with MiR software 2.3.1 or earlier? There is currently an incompatibility with MiR software versions greater than 2.3.1.

closed 2019-08-07 by mintar