DFKI-NI/mir_robot #23: Could not find a package configuration file provided by "costmap_queue"

chbloca, 2019-04-08:
```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "costmap_queue"
  with any of the following names:

    costmap_queueConfig.cmake
    costmap_queue-config.cmake

  Add the installation prefix of "costmap_queue" to CMAKE_PREFIX_PATH or set
  "costmap_queue_DIR" to a directory containing one of the above files.  If
  "costmap_queue" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  mir_robot/mir_dwb_critics/CMakeLists.txt:6 (find_package)

-- Could not find the required component 'costmap_queue'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "costmap_queue"
  with any of the following names:

    costmap_queueConfig.cmake
    costmap_queue-config.cmake

  Add the installation prefix of "costmap_queue" to CMAKE_PREFIX_PATH or set
  "costmap_queue_DIR" to a directory containing one of the above files.  If
  "costmap_queue" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  mir_robot/mir_dwb_critics/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/chbloca/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/chbloca/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
chbloca@chbloca-ThinkPad-T470s:~/catkin_ws$ 
```
How come does this error appear during compilation since I have done previously:
`$ sudo apt-get install ros-kinetic-navigation`

chbloca, 2019-04-08:
```
chbloca@chbloca-ThinkPad-T470s:~/catkin_ws/src$ rosdep install --from-paths ./ -i -y --rosdistro kinetic
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
dodo_detector_ros: Cannot locate rosdep definition for [kinect2_bridge]

```

mintar, 2019-04-08:
To avoid the `kinect2_bridge` error, you should run that `rosdep install` command inside the `mir_robot` folder.

chbloca, 2019-04-08:
Many thanks as always for your help @mintar. It is working well now.

closed 2019-04-08 by chbloca