DFKI-NI/mir_robot #24: When adding camera topics, lag is huge

chbloca, 2019-04-09:
Hello. I have been testing the addition of the camera topics to the mir_robot/mir_driver/nodes/mir_bridge.py. The problem comes when there is a huge lag on the streaming of the topic /camera_floor/driver/color/image_raw so does /camera_floor/driver/depth/color/points when doing the ROS bridge with roslaunch mir_driver mir.launch. I have not forgotten to comment also the other available topics which I do not need.
Why is this happening? Is there a solution to this?

mintar, 2019-04-09:
The lag is because the images and point clouds  are very large, high-frequency messages, so there is just a lot of data to be transferred. The rosbridge internally uses JSON to transfer the messages from the MiR PC to the host PC, which blows up the message size by around 5x and adds significant computational overhead on both the MiR PC and host PC side (compared to directly subscribing to the ROS messages). This is the public API of the MiR, so nothing can be done about this on the mir_driver side.

Regarding your problem, there are several ways you could solve this, all of which are kind of hacky. The easiest way I can think of is:

* Don't use the mir_driver, but connect to the MiR's roscore directly.
* Don't use the move_base action; instead, publish the target pose onto the `move_base_simple/goal` topic. The drawback is that you won't get feedback when the robot has reached its goal, or when it was unable to do so; you probably have to monitor the odometry and tf topics to figure out when the robot has stopped, along with some timeouts.

chbloca, 2019-04-09:
Besides what you commented, I think the root of the problem is in this launcher when I try to do the gmapping. Since the global costmap is not published by the MiR, I have to force such publication by running the move_base node separately:
```
<launch>

  <param name="use_sim_time" value="false" />

  <arg name="scan_topic" default="/f_scan" />
  <node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">

    <param name="odom_frame" value="odom"/>
    <param name="base_frame" value="base_link"/>
    <param name="map_frame" value="map"/>

    <!-- Process 1 out of every this many scans (set it to a higher number to skip more scans)  -->
    <param name="throttle_scans" value="1"/>

    <param name="map_update_interval" value="2.0"/> <!-- default: 5.0 -->

    <!-- The maximum usable range of the laser. A beam is cropped to this value.  -->
    <param name="maxUrange" value="5.0"/>

    <!-- The maximum range of the sensor. If regions with no obstacles within the range of the sensor should appear as free space in the map, set maxUrange < maximum range of the real sensor <= maxRange -->
    <param name="maxRange" value="10.0"/>

    <param name="sigma" value="0.05"/>
    <param name="kernelSize" value="1"/>
    <param name="lstep" value="0.05"/>
    <param name="astep" value="0.05"/>
    <param name="iterations" value="5"/>
    <param name="lsigma" value="0.075"/>
    <param name="ogain" value="3.0"/>
    <param name="minimumScore" value="0.0"/>
    <!-- Number of beams to skip in each scan. -->
    <param name="lskip" value="0"/>

    <param name="srr" value="0.01"/>
    <param name="srt" value="0.02"/>
    <param name="str" value="0.01"/>
    <param name="stt" value="0.02"/>

    <!-- Process a scan each time the robot translates this far  -->
    <param name="linearUpdate" value="0.1"/>

    <!-- Process a scan each time the robot rotates this far  -->
    <param name="angularUpdate" value="0.05"/>

    <param name="temporalUpdate" value="0.1"/>
    <param name="resampleThreshold" value="0.5"/>

    <!-- Number of particles in the filter. default 30        -->
    <param name="particles" value="10"/>

    <!-- Initial map size  -->
    <param name="xmin" value="-5.0"/>
    <param name="ymin" value="-5.0"/>
    <param name="xmax" value="5.0"/>
    <param name="ymax" value="5.0"/>

    <!-- Processing parameters (resolution of the map)  -->
    <param name="delta" value="0.1"/>

    <param name="llsamplerange" value="0.01"/>
    <param name="llsamplestep" value="0.01"/>
    <param name="lasamplerange" value="0.005"/>
    <param name="lasamplestep" value="0.005"/>

    <remap from="scan" to="$(arg scan_topic)"/>
  </node>

  <arg name="namespace" value=""/>
  <node pkg="move_base" type="move_base" respawn="false" name="move_base_node" output="screen">
    <param name="footprint_padding" value="0.01" />
    <param name="controller_frequency" value="5.0" />
    <param name="controller_patience" value="3.0" />
    <param name="oscillation_timeout" value="30.0" />
    <param name="oscillation_distance" value="0.5" />
    <param name="planner_patience" value="1" />
    <param name="controller_patience" value="1" /> 
    <remap from="cmd_vel" to="mobile_base/commands/velocity"/>
    <param name="recovery_behavior_enabled" value="false" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find rrt_exploration_tutorials)/param/base_local_planner_params.yaml" command="load" />  
    <param name="global_costmap/global_frame" value="$(arg namespace)/map"/>
    <param name="global_costmap/robot_base_frame" value="$(arg namespace)/base_link"/>
    <param name="global_costmap/laser_scan_sensor/sensor_frame" value="/$(arg namespace)/base_laser_link"/>
    <param name="global_costmap/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>    
    <param name="local_costmap/global_frame" value="$(arg namespace)/odom"/>
    <param name="local_costmap/robot_base_frame" value="$(arg namespace)/base_link"/>
    <param name="local_costmap/laser_scan_sensor/sensor_frame" value="$(arg namespace)/front_laser_link"/>
    <param name="local_costmap/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>
    <param name="local_costmap/obstacle_layer/laser_scan_sensor/topic" value="/$(arg namespace)/$(arg scan_topic)"/>
  </node>

  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find rrt_exploration_tutorials)/launch/includes/rviz_config/simple.rviz">
  </node>

</launch>
```
This creates this extra /move_base related topics:

> /move_base_node/NavfnROS/plan
> /move_base_node/TrajectoryPlannerROS/global_plan
> /move_base_node/TrajectoryPlannerROS/local_plan
> /move_base_node/global_costmap/costmap
> /move_base_node/global_costmap/costmap_updates
> /move_base_node/global_costmap/footprint
> /move_base_node/global_costmap/inflation_layer/parameter_descriptions
> /move_base_node/global_costmap/inflation_layer/parameter_updates
> /move_base_node/global_costmap/obstacle_layer/parameter_descriptions
> /move_base_node/global_costmap/obstacle_layer/parameter_updates
> /move_base_node/global_costmap/parameter_descriptions
> /move_base_node/global_costmap/parameter_updates
> /move_base_node/global_costmap/static_layer/parameter_descriptions
> /move_base_node/global_costmap/static_layer/parameter_updates
> /move_base_node/local_costmap/footprint

I actually need /move_base_node/global_costmap/costmap from the MiR, but there is no way to retrieve it without running this node separately...

mintar, 2019-04-10:
Umm... if you run a separate instance of move_base internally, the costmap won't be the same as the one from the MiR, so this makes no sense.

The MiR publishes the costmap on the topic `move_base_node/local_costmap/obstacles`, but it has type `nav_msgs/GridCells`, not the more common `nav_msgs/OccupancyGrid`. There are a couple more `GridCells` topics you can comment in in `mir_bridge.py`, if you want them.

The MiR runs Kinetic, but many parts were branched off and heavily customized in Indigo (or Fuerte), so move_base and the rest of the navigation stack seem to resemble Indigo more than Kinetic.

chbloca, 2019-04-10:
All right. Then I cannot use the costmaps since a package I am using is not compatible with such kind of messages. That is why I have to run a move_base additional node in my computer. Thanks for the information :)

closed 2019-04-10 by chbloca