DFKI-NI/mir_robot #26: Would this package work for mir200?

teover, 2019-04-23:
If not, which changes would need to be made to make it work?

mintar, 2019-04-23:
I think it should work without modifications, because the Mir 100 and 200 are very similar from the software point of view.

kurtgeebelen, 2019-04-30:
Same question, but then for MIR 500. Would this package also work for that robot?

mintar, 2019-04-30:
I honestly don't know, I've never worked with a MiR 500. So the only option is to try. :)

mintar, 2019-04-30:
And let me know if it works!

teover, 2019-05-03:
Made my own solution using the MiR REST API as I only needed position, battery feedback and the ability to send missions to MiR200. 
```

import requests
import json

# Robot position mission json for MiR200, can be found by requesting mission list from REST api
go_to_robot = {
    "mission_id": "YOUR MISSION ID, SHOULD LOOK LIKE THE ONE UNDER",
    "message": "Go to robot",
    "parameters": [],
    "priority": 0
}

# Storage mission json
go_to_storage = {
    "mission_id": "112434fa-674s-11e9-sg4e-dsgdg1132d32",
    "message": "Go to storage",
    "parameters": [],
    "priority": 0
	
# API URL
url_mission_queue = "http://mir.com/api/v2.0.0/mission_queue"
url_get_status = "http://mir.com/api/v2.0.0/status"

headers = {
        'accept': 'application/json',
        'Authorization': 'Basic YOUR KEY',
        'Accept-Language': 'en_US',
        'Content-Type': 'application/json'
        }
		
# Check battery status
def checkMiRBattery():
    r = requests.get(url_get_status, headers=headers).json()
    return r["battery_percentage"]

# Check if robot is at a position, input position is a dict with x and y
def checkIfAtPosition(position):
    r = requests.get(url_get_status, headers=headers).json()
    mir_pos_x = r["position"]["x"]
    mir_pos_y = r["position"]["y"]
    # Treshold for accepting the postion in meters
    t = 0.1
    if (position["x"]-t < mir_pos_x < position["x"]+t and position["y"]-t < mir_pos_y < position["y"]+t):
        return True
    else:
        return False

# takes input string mission, mission is a mission json as above
def addMissionToQueue(mission):
    r = requests.post(url_mission_queue, data=json.dumps(mission), headers=headers)
    if r.status_code == 201:
        return True
    else:
        print "Error: " + str(r.status_code)
        return False
```

closed 2019-05-03 by teover

kurtgeebelen, 2020-02-10:
Hello mintar,

It took a bit longer than expected (to find the time to test it), but we got the package working on our MIR500. As mentioned in other posts, we had to make a small change to mirbridge.py:
https://github.com/dfki-ric/mir_robot/issues/45#issuecomment-584404158