DFKI-NI/mir_robot #31: Using package with MIR 200

kevinjames31, 2019-07-29:
Is it possible to get this package working with the MiR 200? Can't seem to get the robot to receive published commands from my PC

mintar, 2019-08-07:
I haven't tested this package with a MiR200, but I am 90% sure that it should be working. The MiR100 and MiR200 are almost identical. Your connection problems could be the same as #30 . Does the node never get past the "still waiting for connection to 192.168.12.20" warning?

kevinjames31, 2019-08-07:
It actually works with the MiR200. I had initially set the ROS master as the robot IP but when I changed it to the Jetson's IP from which I am running your package the robot responds to commands. May I ask exactly how this is being accomplished even though there should really only be one ROS master? Sorry but I'm new to most of this.

closed 2020-06-29 by mintar