DFKI-NI/mir_robot #34: Which topic to publish actions?

NPMN, 2019-08-13:
Hi guys!!

I have made a script which through python library requests and RestAPI retrieved the x,y and theta coordinates from position markers made in the map of the mir. 
The rest of my plan is to take these coordinates and publish them over an simple server/client action and my question is to which topic to publish these values to??
I have looked over the topics /robot_pose, /amcl_pose, move_base_node/current_goal and move_base/feedback and which all publishes the x,y, theta of the robot.

I have also taken a look at which kind of messages these topics publishes and the two which i think is the most relevant is the messages of move_base_node/current_goal and move_base/feedback which publishes messages of the type geometry_msgs/PoseStamped and MoveBaseActionFeedback.

Any help would be appreciated! Thank you in advance

mintar, 2019-08-13:
You should create a ROS actionlib client for the `move_base` action. See this tutorial:

http://wiki.ros.org/navigation/Tutorials/SendingSimpleGoals

The example is in C++, but you can equally use python. See here:

http://wiki.ros.org/actionlib_tutorials/Tutorials/Writing%20a%20Simple%20Action%20Client%20%28Python%29

NPMN, 2019-08-13:
Thank you very much @mintar. I will take a look at this and come back to you if any problem would arise :)

NPMN, 2019-08-16:
Hello @mintar I have a question about the move_base action. I looked at the documentation of this action and it only sends out a goal and a feedback definition. I followed the tutorials and have written an actionlib, but wonder if i need to use a publisher to publish this goal message or will this action client to that for me and will this action "publish" it to move_base/goal topic??

Also when running my actionlib client node, the steps to run this is as simple to only connect to mir, run mir_bridge and then the actionlib client? I understand that the action server can take some time to be connected to the action client and a reponse.

I will attached the testing actionlib client node below. Not sure if a while loop is needed to go to the desired pose.
`#!/usr/bin/env python

import rospy
import actionlib
import actionlib_msgs.msg
from move_base_msgs.msg import *
from GetQuaternionFromAngle import *
import os #Not used...

def initPose():
    rospy.loginfo("Retrieving List from other program...\n")
    filename='pList.txt'
    _2DPoseList=ReadTxtFileIntoList(filename)
    
    
    return _2DPoseList
        

def mir_client():
   

############################Getting Pose######################################################

  _2DPoseList=initPose() #
  

##############################################################################################

#Creates the SimpleActionClient from actionlib and passing the type of action := MoveBaseAction to the constructor.
  MiRClient=actionlib.SimpleActionClient('move_base',move_base_msgs.msg.MoveBaseAction)

  rospy.loginfo("Waiting for server to start")
  #Waits until the action server from MiR has started up and started listening for goals.
  MiRClient.wait_for_server(rospy.Duration(10))
 
  rospy.loginfo("Action server started!")

  #Send a specific goal to the robot. As of know, just put anyone of the positions generated from GetQuaternionFromAngle.py
  goal=move_base_msgs.msg.MoveBaseGoal()

  goal.target_pose.header.frame_id = "base_link" #Which frame is here?? map? base_link?
  goal.target_pose.header.stamp = rospy.Time.now()

  goal.target_pose.pose.position.x=_2DPoseList[2][1] #
  goal.target_pose.pose.position.y=_2DPoseList[2][2] #
  goal.target_pose.pose.position.z= 0.0 #

  goal.target_pose.pose.orientation.x=_2DPoseList[2][3] #
  goal.target_pose.pose.orientation.y=_2DPoseList[2][4] #
  goal.target_pose.pose.orientation.z=_2DPoseList[2][5] #
  goal.target_pose.pose.orientation.w=_2DPoseList[2][6] #

  rospy.loginfo("Sending Goal Pose")
  MiRClient.send_goal(goal)

  
  IsFinished=MiRClient.wait_for_result(rospy.Duration(60))
  rospy.loginfo(IsFinished)
   
  #Testing...

def main():
    try:
     rospy.loginfo("Running...")
     rospy.init_node('ExecuteMiRAction')
    
     mir_client()
 
  except rospy.ROSInterruptException:
     print("program interrupted before completion")

if __name__=='__main__':
   
   try:
       main()
   except:
      print("Main could not execute!!")
  

 

  

   `

mintar, 2019-08-16:
The action client will handle the communication over the 5 topics (cancel, feedback, goal, result, status). You do not need to create a publisher or subscriber yourself.

The `frame_id` should be `map`.

Check the indentation of your code again (it's wrong). In Python, indentation matters! Just run `pylint myprogram.py` to see the errors.

The steps necessary to run it are in the README, section "Running the driver on the real robot".

NPMN, 2019-08-19:
Hi @mintar!
I followed your instructions and made an action client. I am connected to the mir interface and has pressed play. 
I launch the rosbridge launch file and my action client node and i can see that the `mir_bridge` is starting to stream messages on `move_base` topics. 
The result from running the client node is that is starting and goes to the end but i don't get the robot to move. 
I also checked the tf_monitor and got an average delay: 2.78835/max delay 2.83468 for the /base_footprint. I also went to the setting of time on the robot which were correct. 

Is there any more troubleshooting to get the robot to go? 

I will attached my result below!
[image]
[image]
[image]

my action script:

`#!/usr/bin/env python
"""This script will generate an action for the MiR robot to move to a specific pose"""

from os import path
import rospy
import actionlib
import move_base_msgs.msg
from GetQuaternionFromAngle import ReadTxtFileIntoList

def get_poses():
    """This function will recieve the poses from a .txt, take a look at GetQuaternionFromAngle.py"""

    if path.exists("pList.txt"):
        rospy.loginfo("Retrieving List from other program...\n")
        filename = path.realpath("pList.txt")
        _2DPoseList = ReadTxtFileIntoList(filename)
        return _2DPoseList
    else:
        print "File don't exists"
        return None

def feedback_cb(msg):
    """ sending feedback of the action"""
    print "Feedback recieved: ", msg

def mir_client():
    """The action client """

    _2DPoseList = get_poses()

    if _2DPoseList != None:

        #Creates the SimpleActionClient
        mir = actionlib.SimpleActionClient('move_base', move_base_msgs.msg.MoveBaseAction)

        rospy.loginfo("Waiting for server to start")
        #Waits until the action server from MiR has started up and started listening for goals.
        mir.wait_for_server(rospy.Duration(10))

        rospy.loginfo("Action server started!")

        #Send a goal to robot.
        goal = move_base_msgs.msg.MoveBaseGoal()

        goal.target_pose.header.frame_id = "map" #map frame
        goal.target_pose.header.stamp = rospy.Time.now()

        goal.target_pose.pose.position.x = float(_2DPoseList[3][1]) #
        goal.target_pose.pose.position.y = float(_2DPoseList[3][2]) #
        goal.target_pose.pose.position.z = 0.0 #

        goal.target_pose.pose.orientation.x = float(_2DPoseList[3][3]) #
        goal.target_pose.pose.orientation.y = float(_2DPoseList[3][4]) #
        goal.target_pose.pose.orientation.z = float(_2DPoseList[3][5]) #
        goal.target_pose.pose.orientation.w = float(_2DPoseList[3][6]) #

        rospy.loginfo("Sending Goal Pose")

        mir.send_goal(goal) #feedback_cb=feedback_cb

        IsFinished = mir.wait_for_result(rospy.Duration(10)) #Longer duration here???
        rospy.loginfo(IsFinished)

        return mir.get_result()
    else:
        print '_2DPoseList is equal to None!'
        return None

def main():
    """ the ROS node executing the action client """
    try:
        rospy.init_node('mir_action_client')
        rospy.loginfo("Running...")
        rospy.loginfo("mir_client is about to start")

        result = mir_client()
        print 'Result is: ', result
        rospy.loginfo("mir_client is finished")
    except rospy.ROSInterruptException as e:
        print "Something is wrong: ", e

if __name__ == '__main__':

    try:
        main()
    except rospy.ROSInterruptException:
        print "Main could not execute!!"
`

mintar, 2019-08-26:
Which MiR software version are you using? This repository is only compatible with version 2.3 and earlier. The problem you are describing sounds exactly like the problems caused by a later version of the MiR software.

NPMN, 2019-08-26:
Yes, that was my initial though. I'm using the 2.7 version. Thank you for all your help @mintar, but i will solve my problems of positioning with Rest API.

closed 2019-08-26 by NPMN