DFKI-NI/mir_robot #35: Dead reckoning with a MiR100

rsr152, 2019-08-21:
Hi all,

My MiR100 robot (like all others) is really "hesitant" while docking and I am trying to use dead reckoning for the last meter or so of travel so it can dock faster.  Even a couple of seconds saved per docking action is a big saving for me.

However, dead reckoning is affected by errors (due to slippage or bumps or uneven surfaces) that may worsen as the robot travels more.  Does any one have any ideas on what other kind of feedback I can use in an indoor environment in addition to the odometry?  Triangulation using transmitters/receivers or GPS is not an option.  Is there any way to extract localization data using the MiR's "out of the box" sensors?  Using the camera to recognize fiducials to estimate pose is an option but I am looking for something quick and relatively easier than that for now.

I know the coordinates of different locations on my map in a world frame, and I know the pose of the robot as it moves, also in the world frame.  Can I use any capability in the /tf2 library to get another **estimate** of the robot's current pose. I can then try fusing this new data with information that I am getting from the /odom topic?

Would appreciate any ideas - Thanks!

mintar, 2019-08-26:
You can use tf to get the transform from the "map" frame to the "base_footprint" frame. This is the result of the [AMCL](http://wiki.ros.org/amcl) localization (i.e., the sensor fusion from laser scanners, IMU and odometry). This is what the MiR uses internally as well. You don't need to do any fusion with the /odom topic yourself, since it's already included.

rsr152, 2019-09-04:
I started exploring the use of tf but wanted to get more input on whether it would help at all with better estimation of robot pose in a situation where I want to use dead reckoning to get to a target pose (using only encoder feedback, no camera feedback).

My controller today extracts the current pose of the MiR using an API call and guide it to its target pose (target pose has been defined earlier, such as a waypoint).  All these coordinates (x, y, theta) and (xtarget, ytarget, thetatarget) are in the "map" frame.  If the robot is "suffering" from encoder errors, I feel using tf will not help overcome these.  There will be a need to capture feedback from other sensors like an overhead camera, or an onboard camera.

Am I missing something?  Appreciate your thoughts...

mintar, 2019-09-05:
As I said, the tf pose (which should be the same that you get from the MiR via an API call) already contains a sensor fusion between odometry (= encoders), IMU and laser scanners. If you have a normal amount of encoder slip and IMU drift, this is usually corrected by referencing the laser scans to the recorded map. The onboard camera is not useful for localization, since it is pointed downwards to the floor. If you get a lot of localization errors, I would recommend re-recording the map; this usually helps. If you have a highly dynamic environment, where the map changes drastically all the time, you might have to integrate an external positioning system.

rsr152, 2019-09-25:
Thanks a lot for the suggestions.  This is working well for now!

closed 2019-09-25 by rsr152

stevemartinov, 2021-04-24:
@mintar I think you need to read the question a bit more careful before replying with the same comment as you replied before.

@rsr152 Yes, that is how AMRs perform the last meter drive in. You first use AMCL to navigate to some pose then when you have arrived, you can turn off the AMCL and slowly give commands directly to the motor controllers to move forward/backward X meters. 1 meter is nothing for wheel encoders and you will probably get mm accuracy