DFKI-NI/mir_robot #38: publishing to cmd_vel doesn't move the mir

IanCloutier, 2019-09-26:
Hello,

I've been having this issue where I cannot get the mir to move by publishing to the cmd_vel topic with the following command:

"rostopic pub /cmd_vel geometry_msgs/Twist -r 100 -- '[0.0, 0.0, 0.0]' '[0.0, 0.0, 0.2]'" 

after issuing this command, if I echo the cmd_vel topic I see these values coming back, but the MiR isnt moving (it is in a green idle state). I have already checked the list of pub/sub topics in mir_bridge.py, and the topic cmd_vel is in the SUB_TOPICS section and not the PUB_TOPICS section. Any help would be appreciated.

Thanks!

IanCloutier, 2019-10-01:
Hi! This seems to be an issue involving Mir software 2.7.0 and above. I rolled my software back to 2.6.2 and it worked fine.

mintar, 2019-10-04:
Great to hear that you've solved it. This repo only fully supports MiR software 2.3.1 and earlier; on later versions, sending move_base actions doesn't work.

closed 2019-10-04 by mintar

kurtgeebelen, 2020-02-10:
I posted a solution to this issue here:
https://github.com/dfki-ric/mir_robot/issues/45#issuecomment-584404158