DFKI-NI/mir_robot #39: Post correct x, y and orientation values to MiR100 and MiR200

vikram-gopinath, 2019-10-03:
Hi Mintar, 

My MiR100 and MiR200 operate in dynamic environments and constantly lose localization. Could you share how I should go about updating the correct x, y and orientation values (more accurate values which I have obtained by other means). I am guessing I would have a publisher node to constantly publish the correct values and overwrite the x, y and orientation values on the MiR. Looking forward to your reply.

Best,
Vikram

rsr152, 2019-10-21:
Vikram, one way to do it is to use the MiR's API to write the correct position and orientation to the robot.  

The command is as follows:
Rest Command: 192.168.xx.yy/api/v2.0.0/status

Raw text (0,0,0 is the example here):
{
"position":
{ "y": 0, "x": 0, "orientation": 0}
}

You'll have to add the other syntax for the API call from your code.

If you have done it some other way, please share.

vikram-gopinath, 2020-01-09:
Thanks for the response. Used REST API as you described as it was way easier than using ROS.