DFKI-NI/mir_robot #41: I have simple question

kyunghwan, 2019-11-06:
I have simple question, is this the official source code for Mir Robot Company?

planthaber, 2019-11-06:
No.

This is the code the German Research Center for Artificial Intelligence (DFKI) created to use the MIR Robots with ROS.

I am not the developer but @mintar might tell you more.

mintar, 2019-11-06:
That's correct. We are not affiliated with MiR (the company), and the code in this repo is not supported by MiR.

closed 2019-11-06 by mintar

planthaber, 2019-11-06:
@mintar I think it is a good idea to add that to the repository description

mintar, 2019-11-06:
Done: f6b9a4e61f79f3966cab039f47950cfb628e51f6.

planthaber, 2019-11-06:
I meant the description for the github repository on the github repository page itself:
Changed it for you ;-) 

    ROS support for the MiR100 Robot. This is a community project to use the MiR Robots with ROS. It is not affiliated with Mobile Industrial Robots.