DFKI-NI/mir_robot #45: Using ROS with 2.8 firmware

FabianPG11, 2020-02-04:
Dear all, 
I have a question regarding the ROS integration with MiR100 running robot software versions of 2.7 and 2.8. 
As already mentioned in the previous issues, the ROS wrapper available in this repository is not compatible with versions older than 2.3. I was wondering what causes the problem and whether there is a fix for it.
Currently, we run the driver and the robot gets connected successfully, however, nothing happens after commanding velocities through /cmd_vel topic.
I would highly appreciate any help.

kurtgeebelen, 2020-02-10:
Hello,

We contacted MIR for this issue. The problem is that the message type for /cmd_vel should be TwistStamped in stead of Twist on line 208 in mir_driver/nodes/mir_bridge.py . 

Not sure if this changed in the MIR software (only tested with latest version), but this solved the problem for us, on a MIR 500, and I assume it will solve it on a MIR100/200 too.

FabianPG11, 2020-02-11:
Thanks for the reply. We have tested it on MiR100 running robot software version 2.7 and it worked.

rsr152, 2020-02-28:
Thanks.  I am looking to measure the **deceleration** of a MiR 100 robot in an emergency stop  (e-stop) situation.  I am running software v 2.8.1 but can roll back to v2.8.0 if commanded velocities don’t t move the robot.

1. Is there a way to measure deceleraton directly, or do we have to calculate it using the velocities v1 and v2 logged at times t1 and t2, and calculating acceleration = (v2-v1)/(t2-t1)

2.  Any ideas on how we can force the MiR 100 robot to come to an emergency stop **without** introducing a sudden physical barrier in its path?  If we command the robot to travel along a straight line at say 1.0m/s and then command a velocity of 0, will it force an e-stop?

My understanding is that from the command line, this would translate to (would appreciate if someone could validate the command)  :
rostopic pub /cmd_vel geometry_msgs/TwistStamped -r 5 -- '[1.0, 0.0, 0.0]' '[0.0, 0.0, 0.0]'
followed by 
rostopic pub /cmd_vel geometry_msgs/TwistStamped -r 5 -- '[0.0, 0.0, 0.0]' '[0.0, 0.0, 0.0]'

Would welcome any other ideas to force an e-stop !

Thanks as always!

rsr152, 2020-02-28:
> Hello,
>
> We contacted MIR for this issue. The problem is that the message type for /cmd_vel should be TwistStamped in stead of Twist on line 208 in mir_driver/nodes/mir_bridge.py .
>
> Not sure if this [...]

Hi all,

I was troubleshooting my problem and had a few  updates.  The result is setill the same (Nothing happens after commanding velocities through /cmd_vel topic.) 

1. I made the change suggested by @kurtgeebelen in line 208 in **mir_driver/nodes/mir_bridge.py** 
2. I also added the line 
**geometry_msgs/TwistStamped twist**
in the file ~/catkin_ws/src/mir_robot/mir_msgs/msg/ExternalRobot.msg
(not sure if I should have)
3. _roslaunch mir_driver mir.launch_ completes initialization successfully
4.  rosmsg show geometry_msgs/Twist returns the expected result
_geometry_msgs/Vector3 linear
  float64 x
  float64 y
  float64 z
geometry_msgs/Vector3 angular
  float64 x
  float64 y
  float64 z_
5. rosmsg show geometry_msgs/TwistStamped returns:
_std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
geometry_msgs/Twist twist
  geometry_msgs/Vector3 linear
    float64 x
    float64 y
    float64 z
  geometry_msgs/Vector3 angular
    float64 x
    float64 y
    float64 z_
(I was expecting the same result as Twist)

6. A sample velocity command 
**rostopic pub /cmd_vel geometry_msgs/TwistStamped -r 5 -- '[1.0, 0.0, 0.0]' '[0.0, 0.0, 0.0]'** 
errors out with this message:
_Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rostopic", line 35, in <module>
    rostopic.rostopicmain()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 2103, in rostopicmain
    _rostopic_cmd_pub(argv)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1789, in _rostopic_cmd_pub
    argv_publish(pub, msg_class, pub_args, rate, options.once, options.verbose, substitute_keywords=options.substitute_keywords)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1813, in argv_publish
    publish_message(pub, msg_class, pub_args, rate, once, verbose=verbose, substitute_keywords=substitute_keywords)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1663, in publish_message
    _fillMessageArgs(msg, pub_args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1701, in _fillMessageArgs
    genpy.message.fill_message_args(msg, pub_args, keys=keys)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 550, in fill_message_args
    _fill_message_args(msg, msg_args, keys, '')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 508, in _fill_message_args
    _fill_val(msg, f, v, keys, prefix)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 429, in _fill_val
    _fill_message_args(def_val, v, keys, prefix=(prefix+f+'.'))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 508, in _fill_message_args
    _fill_val(msg, f, v, keys, prefix)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 429, in _fill_val
    _fill_message_args(def_val, v, keys, prefix=(prefix+f+'.'))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 510, in _fill_message_args
    raise ValueError("invalid msg_args type: %s"%str(msg_args))
ValueError: invalid msg_args type: 0.0_

I'd appreciate any troubleshooting suggestions!

Thanks!

kurtgeebelen, 2020-02-28:
You're not publishing a TwistStamped message in the correct way. Try this:

rostopic pub -r 5 /cmd_vel geometry_msgs/TwistStamped "{header: auto, twist:{linear:{x: 1, y: 0, z: 0}, angular:{x: 0, y: 0, z: 0}}}"

FabianPG11, 2020-03-02:
Hey, 
I was wondering if there is a way to control mir100 sending goals for this version (2.7). In my case, I can command it with the topic /cmd_vel, however, it does not move with a simple goal navigation node. I tried with move_base and relative_move_action but none of them worked.
Any hint for this?

Thanks.

rsr152, 2020-03-03:
Thanks a lot - It worked from the command line!

Is the change in mir_bridge.py the only change needed in the entire code base? Based on my limited understanding of ROS, I suspect there are other files (such as catkin_ws/src/mir_robot/mir_msgs/msg/ExternalRobot.msg) that will need to be modified, because the Twist message has a different structure from TwistStamped.

At a high level, I am trying to do this:
If (user wants to manually command a velocity) then
_- publish TwistStamped message (as above)_
elseif (user wants to run mission X) then
_- (invoke API to complete the mission)_
endif

I was able to do this using the Twist message in earlier versions of the MiR software.

FaseehCS, 2020-03-26:
Hi I am also using MIR software version 2.8 and managed to move it using topic /cmd_vel by changing the msg type as described above. I am having the same issue with the topic /move_base and relative_move_action. Anyone who has used this, kindly comment on this. Thank you

kurtgeebelen, 2020-03-26:
Hi all,

we had the same issue in trying to send a goal (x,y,orientation) to the MIR, and it didn't react or when it did react it didn't navigate to the correct position. We are now using the rest API to have it navigate to a desired posiiton. The position can then also be send from a script. Below is an example python script that we use for this. In the MIR itself you have to create position and a mission to drive to this position. The coordinates of this position are thus changed by this script (set_position)

`import requests
import json
import time
import sys

 

# MIR settings
mir_ip_address = "192.168.50.123"

 

#generating the authorization key see restapi documentation
mir_authorization_Distributor = "your authorization code"
mir_headers = {
        "accept": "application/json",
        "Accept-Language": "en_US",
        "content-type": "application/json",
        "Authorization": mir_authorization_Distributor,
        }

 

# ----------------------------------- Get MIR State ---------------------------------------#
# retrieve the state number
def get_robot_state():
    # Request the status
    response = requests.get(mir_url + "status", headers=mir_headers)
    # Take out the state number from the code
    code = response.json()
    #state = code['state_id']
    # Return the state number
    return code

 

# ----------------------------------- Send MIR position --------------------------------------#
# start mission to specific location
# retrieve mission_id via output from get missions command

 

def set_position(GUID, x, y , orientation):

 

    data = {"pos_x": x, "pos_y": y, "orientation": orientation}

 

    # Send mission to mir
    response = requests.put(mir_url + "positions/"+GUID, headers=mir_headers, data=json.dumps(data))
    return response.json()    

 

# ----------------------------------- Send MIR on mission --------------------------------------#
# start mission to specific location
# retrieve mission_id via output from get missions command

 

def set_mission(GUID):
    data = {"mission_id": GUID}

 

    # write to log
    print("mir send on mission: " + GUID)

 

    # Send mission to mir
    response = requests.post(mir_url + "mission_queue", headers=mir_headers, data=json.dumps(data))
    return response.json()

 

if __name__ == "__main__":
    set_position("your guid", 5.0, 2.75 , 153)
    set_mission("your guid")`

FaseehCS, 2020-04-22:
> Hi all,
> 
> we had the same issue in trying to send a goal (x,y,orientation) to the MIR, and it didn't react or when it did react it didn't navigate to the correct position. We are now using the rest API to have it navigate to a desired posiiton. The position can then also be send from a script. Below is an example python script that we use for this. In the MIR itself you have to create position and a mission to drive to this position. The coordinates of this position are thus changed by this script (set_position)
> 
> ```python
> import requests
> import json
> import time
> import sys
> 
>  
> 
> # MIR settings
> mir_ip_address = "192.168.50.123"
> 
>  
> 
> #generating the authorization key see restapi documentation
> mir_authorization_Distributor = "your authorization code"
> mir_headers = {
>         "accept": "application/json",
>         "Accept-Language": "en_US",
>         "content-type": "application/json",
>         "Authorization": mir_authorization_Distributor,
>         }
> 
>  
> 
> 
> # ----------------------------------- Get MIR State ---------------------------------------#
> # retrieve the state number
> def get_robot_state():
>     # Request the status
>     response = requests.get(mir_url + "status", headers=mir_headers)
>     # Take out the state number from the code
>     code = response.json()
>     #state = code['state_id']
>     # Return the state number
>     return code
> 
>  
> 
> # ----------------------------------- Send MIR position --------------------------------------#
> # start mission to specific location
> # retrieve mission_id via output from get missions command
> 
>  
> 
> def set_position(GUID, x, y , orientation):
> 
>  
> 
>     data = {"pos_x": x, "pos_y": y, "orientation": orientation}
> 
>  
> 
>     # Send mission to mir
>     response = requests.put(mir_url + "positions/"+GUID, headers=mir_headers, data=json.dumps(data))
>     return response.json()    
> 
>  
> 
> # ----------------------------------- Send MIR on mission --------------------------------------#
> # start mission to specific location
> # retrieve mission_id via output from get missions command
> 
>  
> 
> def set_mission(GUID):
>     data = {"mission_id": GUID}
> 
>  
> 
>     # write to log
>     print("mir send on mission: " + GUID)
> 
>  
> 
>     # Send mission to mir
>     response = requests.post(mir_url + "mission_queue", headers=mir_headers, data=json.dumps(data))
>     return response.json()
> 
>  
> 
> if __name__ == "__main__":
>     set_position("your guid", 5.0, 2.75 , 153)
>     set_mission("your guid")
> ```

Hi Kurt, 
I have been working on MIR version 2.8.0.3 for a while now. I encountered the same problem. Since, this repo offers support for SW version 2.3 or earlier, I tried to fix the issues. The twistStamped msg solution as discussed above worked for  /cmd_vel topic. Thank you for that. For move_base goal actions no luck so far. I am switching to Rest API now. The above code is a nice start indeed but I was wondering if you have done something more sophisticated with this API like, motion planning, mapping, etc.? If you have, is it possible that you can upload a code snippet for that as well, given its not private, offcourse.
 
Thank you

mintar, 2020-06-30:
I don't have MiR Software 2.8 for testing. The latest I have is MiR software 2.6.0. From what I can tell, there are two main problems with using this driver for versions > 2.3.1:

1. The `cmd_vel` topic type was changed from `geometry_msgs/Twist` to `geometry_msgs/TwistStamped` in MiR software 2.7.
2. The `MirMoveBaseAction` was changed in MiR software 2.4.0. Sending MoveBaseAction goals over the `mir_bridge` no longer works.

I have just pushed a new branch [**melodic-2.8**](https://github.com/dfki-ric/mir_robot/compare/melodic-2.8?expand=1) where I'm attempting to fix these two things. The general idea behind the fixes is that the `mir_bridge` should continue to offer the standard topic types on the public ROS side, and translate them from/to the non-standard messages that the MiR uses:

topic                | (public) ROS side                       | (internal) MiR side
---------------------|-----------------------------------------|--------------------------------------
`cmd_vel`            | `geometry_msgs/Twist`                   | `geometry_msgs/TwistStamped`
`move_base/result`   | `move_base_msgs/MoveBaseActionResult`   | `mir_actions/MirMoveBaseActionResult`
`move_base/goal`     | `move_base_msgs/MoveBaseActionGoal`     | `mir_actions/MirMoveBaseActionGoal`
`move_base/feedback` | `move_base_msgs/MoveBaseActionFeedback` | `mir_actions/MirMoveBaseActionFeedback`

Based on the discussion in this issue, I'm relatively confident that 27652fa8 fixes the first problem (cmd_vel).

The commit 47c6aeb8 is my attempt at fixing the second issue (move_base). If you look at the commit, you'll see that lots of fields were added to the `MirMoveBaseAction` definition. I *guess* that the reason why the robot isn't moving is that we leave the `move_task` at its default value of 0, which means `BASE_MOVE`:

https://github.com/dfki-ric/mir_robot/commit/47c6aeb884a6ae83ea5d06cfcea5dbd15f481b7b#diff-f84e16efd5bf3d46b6e08d6c36495fdcR3-R8

... so I'm setting it to `GLOBAL_MOVE` here and hope that this fixes it:

https://github.com/dfki-ric/mir_robot/commit/47c6aeb884a6ae83ea5d06cfcea5dbd15f481b7b#diff-5fb6d7d4a121556dac4973fa51f0c2b2R38

Since I don't have access to the real robot (or MiR software 2.8) at the moment, I **could not test** these changes. So there is a significant chance that they won't work yet, but they should give you a starting point.

@BenziF @FaseehCS @FabianPG11 @rsr152 @kurtgeebelen : If you have the time, please test the branch `melodic-2.8` and report back if it fixes the two problems.

BenziF, 2020-07-01:
Hi mintar,

thank you very much for your quick reply. Due to COVID, we have limited access to our facilities, but next Tuesday I should be able to go and test the new branch of the code on our real robot (MiR software 2.8).

As a quick followup question, is there a way to synchronize the internal PC of the MiR from Ubuntu's command line? I did it using the web interface, but i need to press two buttons in it, one to get the time of my device and another one to save the info and setting it to the robot's internal system. As a result, when I run `$ rosrun tf tf_monitor` I get that the standard tf of the robot are fine (such as /back_laser_link, /imu_link), with avg delays of milliseconds, while the avg delay for /base_footprint is around 0.4 seconds. Is this acceptable, or is it gonna cause any issues with the transforms in your opinion? I have no idea why they decided to go with this interface, it's totally not convenient.

Anyway, I'll report back to you next Tuesday, and thanks again

mintar, 2020-07-01:
> As a quick followup question, is there a way to synchronize the internal PC of the MiR from Ubuntu's command line?

Not really. I have a working solution, but it's not very easy to install: We have an external PC integrated into our robot that is on the same wired network as the MiR PC; let's call it `sensors-pc`. That PC's clock is our reference clock. It's synced to an NTP clock whenever the `sensors-pc` has access to the internet. Next, I installed `chrony` on both the `sensors-pc` and the internal PC of the MiR, and set up the `sensors-pc` as the chrony server and the internal MiR PC as the chrony client. This way, the clocks on these systems always stay in sync without any manual interaction.

If you're wondering how you can install stuff on the internal MiR PC:
* connect a monitor and keyboard to the ports that are exposed on one corner of the MiR
* boot into a live USB linux system
* `chroot` into the MiR PC
* download `chrony_2.1.1-1ubuntu0.1_amd64.deb`, `libtomcrypt0_1.17-7ubuntu0.1_amd64.deb`, `libtommath0_0.42.0-1.2_amd64.deb` and `timelimit_1.8-1_amd64.deb` from a PC that has internet and install them in the `chroot` environment onto the MiR PC using `dpkg -i`
* set up `/etc/chrony/chrony.conf`

Or you do what I did and build your own MiR update image that installs chrony. But that's not easier than the previous way. :)

Alternatively, maybe there's a way to set the clock using the REST API, but I haven't looked into that.

> I did it using the web interface, but i need to press two buttons in it, one to get the time of my device and another one to save the info and setting it to the robot's internal system. As a result, [...]

It would be better if you could get it down to around 0.1 seconds, but that's not really easy using the manual method. I have only ever used the old interface, where they had only one button for setting the time.

> I have no idea why they decided to go with this interface, it's totally not convenient.

Probably because in the "standard use case" for the MiR (i.e., not using ROS), it doesn't matter if the clocks are not in sync. But I agree, this is a major PITA for our use case.

matthias-mayr, 2020-07-01:
@FaseehCS  and me tested the branch with our MIR. cmd_vel worked after applying the change in #56. 
Sending goals also worked when we created a move_base action.

One time it also happened that the "planner" module on MIR dies "unexpectedly" and the button in the web interface went into the grey state after sending a goal from RViz. As of now we can only recover from that by rebooting.
When we tried it again after rebooting and reconnecting, we could use RViz to send goals though.

When starting the MIR bridge we also stumbled upon #57.

Another change that was necessary is in #58: MIR uses tf2.

mintar, 2020-07-01:
Thanks for testing this and providing the pull requests! I've merged #56 and #57. I have my doubts about #58, but I'll post them over there.

Now that we know that this works, we'll have to think about how to merge it into the main `melodic` / `kinetic` / `noetic` branches. I don't plan on keeping the `melodic-2.8` branch around. On the other hand, I don't want just merge this and break the repo for people who are not running 2.8.

Before merging, we'll need some mechanism like the following:
* add a string parameter `mir_sw_version` that can be something like "2.3.1" or "2.8.0"
* add some checks like `if version_gte(mir_sw_version, '2.4.0'): ...`, and only apply the changes in the `melodic-2.8` branch where appropriate.

If any of you want to do that work and provide a pull request, that would be great! Otherwise it'll have to wait until I find the time.

BenziF, 2020-07-08:
Hi @mintar and everyone, yesterday i was able to test the robot (version 2.8) using the new melodic-2-8 branch but sadly still nothing works. I tested posting to `/cmd_vel` but the MiR didn't move. I also launched `roslaunch mir_navigation hector_mapping.launch`, `roslaunch mir_navigation move_base.xml with_virtual_walls:=false` and `rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz`, but it wasn't creating any map in rviz. I'm not aware if I should be launching anything different to  move the robot or to read info from the `/map` topic. 
The main issue that might be causing the fault, imho, is the time synchronization, since hector_mapping was complaining about the unability to transform from `base_footprint` to the other laser frames, probably because the avg delay is around 0.35 secs. Movebase too was complaining, send warnings in that regard. If you think this is the main cause, I'll have to find a way to synchronize the two PCs (which isn't a cakewalk), or I'll eventually need to revert back to version 2.3.1 of the MiR software. 
 Also, when i run `tf_monitor`, i don't see the `odom_comb` frame, as was written in the README, and also the topic itself `/odom_comb` isn't receiving any messages (I saw something similar in an open issue but there wasn't any solution to that).
I wasn't working on the robot using my PC, so i got some screenshots of the terminal as I launched `mir_driver mir.launch`, here below.
[image] 
[image]

If it's my fault and I'm not launching the correct files, then my bad and I kindly as you to correct me. If instead you can ensure me that the problem lies in the synch, I'll find a way out. Thanks for everything btw :)

mintar, 2020-07-08:
Thanks for testing and providing feedback! While 0.35 s time lag may cause some more subtle problems, I don't think that's the problem here. As long as the lag is below 10 s, the robot should at least move.

I have just merged #58 today. If I understand @matthias-mayr correctly, it's necessary to avoid the transform errors. So please pull the melodic-2.8 branch and try again.

Another thing that looks suspicious is the "topic 'move_base/feedback' is not published by the MiR!" warnings. Did you start the "planner" component in the MiR web interface?

BenziF, 2020-07-08:
Thank you for all the support! Okay, I suspected it too, given the large timeout value, but still wasn't 100% sure. 

Perfect, I'll try it next Tuesday! Sorry about the slow updates, but we still have limited access to the facilities.

> Another thing that looks suspicious is the "topic 'move_base/feedback' is not published by the MiR!" warnings. Did you start the "planner" component in the MiR web 

I went through most of the web interface and didn't find anything which resembled the "Service" -> "Configuration" -> "Launch menu" cited in the README, so I simply exported the master with `export ROS_MASTER_URI=http://192.168.12.20:11311` and using `rostopic list` I found out one `amcl` topic and many `move_base` topics. Still this doesn't give me full confirmation, so I'll look into the user manual to see if I can find a Launch menu. 

Also, quick followup question, do I need to push the Play button from the web interface? I tried both ways yesterday, as paused ( yellow LEDs) and active state (green LEDs) but still achieved nothing

matthias-mayr, 2020-07-08:
The layout of the web interface has changed. You can find it in "System --> Processes --> planner".

Although in our understanding it should have responded to the `/cmd_vel` commands. At least it worked for us before subscribing to `/tf_static`. I am also quite sure that we did not press the "play" button in the web interface when sending to `/cmd_vel`.

But #58 should be the reason why your transformations could not be resolved and therefore mapping on the ROS side should not work.

BenziF, 2020-07-08:
> The layout of the web interface has changed. You can find it in "System --> Processes --> planner".

Finally, thanks you, I don't know how I missed it. Now I can be 100% sure

>Although in our understanding it should have responded to the /cmd_vel commands. At least it worked for us before subscribing to /tf_static. I am also quite sure that we did not press the "play" button in the web interface when sending to /cmd_vel.

All I did was launching `roslaunch mir_driver mir.launch`, opening another terminal and publishing a random msg on `/cmd_vel`, WITHOUT exporting the ROS master, but from my understanding that's how it's supposed to work (please correct me if I'm wrong). This failing really makes me question if something is wrong, since it shouldn't need any transform in order to move via cmd_vel.

mintar, 2020-07-09:
I'm never in the green mode ("mission") when publishing goals from ROS, always either blue ("manual active") or yellow ("manual inactive"). Also remember that you must *not* do `export ROS_MASTER_URI=http://192.168.12.20:11311` when you are running the `mir_bridge`; use `export ROS_MASTER_URI=http://localhost:11311` or something to that effect instead.

Still, to check whether `move_base` has already been started on the robot it's a good idea to do what you did (something like `export ROS_MASTER_URI=http://192.168.12.20:11311 && rostopic info move_base/feedback`). Just make sure that `mir_bridge` is not also running on that ROS master.

BenziF, 2020-07-10:
> I'm never in the green mode ("mission") when publishing goals from ROS, always either blue ("manual active") or yellow ("manual inactive"). Also remember that you must not do export [...]

Okay all clear, thank you very much for the confirmation. 

As a last question, does the procedure described in the README for the case of _Gazebo (mapping)_ also works with the real robot? Namely, if I launch `hector_mapping.launch` `move_base.xml with_virtual_walls:=false` and `rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz` would I obtain the map computed by the real MiR with the possibility of sending goals with the 2D goals tool of rviz? I always too this for granted, so please correct me if I'm wrong

mintar, 2020-07-11:
When you follow the README section "Running the driver on the real robot", you'll see the map computed by the real MiR in RViz, and can also send goals from RViz. 

You could also run `hector_mapping.launch` on your PC when connected to the real robot, but you'll have to remap some topics. Otherwise, the map published by the real MiR and the map published by hector_mapping will conflict. Also, the resulting map will not be stored on the MiR, but on your external PC. You can save it as a PNG (`rosrun map_server map_saver`) and export it to the MiR via the MiR web interface, so that it will also be used by move_base on the MiR. But usually it's easier and better to just use the MiR's mapping functionality.

Likewise, when you run `roslaunch mir_navigation move_base.xml` or similar, you'll have to take care of remapping topics (so that the move_base action topics from the MiR and the local move_base don't conflict). It also works: your local move_base will send `cmd_vel` commands to the real MiR. I've done this before, when I had a very specific reason: I had mounted a very long part on the MiR (over 7 meters long), so I needed to make more changes to the move_base configuration. But otherwise, it's usually easier and better to just use the MiR's own move_base.

BenziF, 2020-07-17:
Hello everybody once again,

I was finally able to test on the real robot with the latest pull #58 . I was able to move the MiR by posting messages onto `/cmd_vel` continuously (the robot stops if the msgs aren't published constantly, which turns out to be positive for our interests), and by sending a goal to the `move_base` client though a simple script. This last command executes the same behaviors as if I were to give the robot a goal through the web interface apparently. 

Besides the good news, I had troubles with RViz, since even with the latest code in melodic-2.8 the map wouldn't show up regardless of my inputs. All I could see was the model of the mobile base, and a series of errors due to the absence of the front and back laser's frames. I'm able to receive the map, since the `/map` topic is receiving data, so maybe I'm just executing the wrong instructions? I'm not too familiar with the tool tbh, so all i did was launching `rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz` . Please correct me if I'm wrong. Also, from #60 it appears that some issues are still present with the tool, so I was mostly looking to move the MiR by simple move_base actions. Still, I'd really like to try out this feature. 

Above all, thank you for your continuous support!

matthias-mayr, 2020-07-20:
@BenziF: I just filed a pull request (#62) to resolve #60 and it worked with our MIR.
The issues you are describing sound like your transformations might not work. Make sure that you always synchronize time as described in the readme before connecting to your MIR. You can check your tf tree e.g. with `rosrun rqt_tf_tree rqt_tf_tree`. Make sure that you have transformations all the way from `map` through `base_link` to e.g. `front_laser_link`.

matthias-mayr, 2020-07-20:
I have another question that might be related to the new software version. When connecting to the MIR I get to
```
mir_bridge: [MiR100Bridge.__init__]: [/mir_bridge] trying to connect to 192.168.12.20:9090...
mir_bridge: ### ROS bridge connected ###
mir_bridge: [MiR100Bridge.__init__]: [/mir_bridge] ... connected.
```
quite fast, but then there is an ~20s delay until it progresses with
```
mir_bridge: Publishers:
mir_bridge:  * /LightCtrl/bms_data [mirMsgs/BMSData]
...
```
Is this always the case or could it be due to some changes?

mintar, 2020-07-20:
No, that's normal. It takes quite some time to get all the publishers and subscribers. I didn't debug where exactly the time lag occurs, but I guess it's all the back-and-forth of adding publishers and subscribers one by one via the websockets interface.

MartinJensen37, 2020-10-26:
I just started working on the MiR in the lab with the melodic-2.8 branch. I managed to connect and sync the robot with a 0.2s delay, but when I try to send goals to the MiR with `move_base_simple/goal` the mir_bridge simply comes with the message seen in the screenshot: 
[image]
I have checked that the planner is running on the MiR through the web-interface. Has anyone else stumbled upon the same problem?

mintar, 2020-10-26:
I'm still a bit confused which MiR versions now work and which don't.

@MartinJensen37 is running MiR software 2.8.3, and move_base isn't working (but cmd_vel is).

@BenziF @matthias-mayr @rsr152 : Could you please post which MiR software version exactly you're using (e.g. 2.8.1) and if move_base works with the current melodic-2.8 branch?

mintar, 2020-10-26:
@MartinJensen37 : I suspect that something in the MirMoveBase action changed between MiR 2.8.* and 2.8.3.

- The log you posted shows that all 5 `move_base` action topics (`move_base/result`, ...) are present on the MiR. Otherwise, you would have gotten a warning. So the "planner" is running.
- Still, the `mir_bridge` cannot connect to the action server within 2 seconds.

Did you reduce the time delay between your PC and the MiR?

Some steps that would help debugging:
- Stop the `mir_bridge` etc. and connect to the ROS core on the MiR's internal PC directly by running this command on your external PC: `export ROS_MASTER_URI=http://192.168.12.20:11311`
- Run `rostopic list -v` and paste the result here.
- Run `rosbag record move_base/result move_base/feedback move_base/status move_base/goal move_base/cancel -O mir_move_base.bag` and send the robot to a new pose goal from the web interface (no manual driving, but setting a new goal in the map). Check that the rosbaf is not empty (`rosbag info mir_move_base.bag`) and attach it here. I'm not sure whether setting a goal in the web interface actually triggers ROS messages on the move_base topics, but I hope it does.

MartinJensen37, 2020-10-26:
> I'm still a bit confused which MiR versions now work and which don't.
> @MartinJensen37 is running MiR software 2.8.3, and move_base isn't working (but cmd_vel is).

Yes, I'm currently using the 2.8.3 and the `/cmd_vel` works but the `/move_base_simple/goal` doesn't.

MartinJensen37, 2020-10-26:
> @MartinJensen37 : I suspect that something in the MirMoveBase action changed between MiR 2.8.* and 2.8.3.
>
> * The log you posted shows that all 5 `move_base` action topics (`move_base/result`, ...) [...]

I'll try and do this now. I'll report back with some screenshots and a status for the rosbag.

mintar, 2020-10-26:
> I'll report back with some screenshots 

As an aside: You can also just copy-paste text from the terminal, that's better than taking screenshots.

MartinJensen37, 2020-10-26:
This is going to be a bit long, but here is the output from the `rostopic list -v`: 
```
Published topics:
 * /map_metadata [nav_msgs/MapMetaData] 1 publisher
 * /move_base_node/local_costmap/obstacles [nav_msgs/GridCells] 1 publisher
 * /move_base_node/MIRPlannerROS/local_plan [nav_msgs/Path] 1 publisher
 * /move_base/feedback [mir_actions/MirMoveBaseActionFeedback] 1 publisher
 * /camera_floor_right/driver/color/image_raw/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /move_base_node/global_costmap/unknown_space [nav_msgs/GridCells] 1 publisher
 * /camera_floor_left/driver/detect_rack_obstacles [sensor_msgs/PointCloud2] 1 publisher
 * /scan [sensor_msgs/LaserScan] 2 publishers
 * /camera_floor_right/driver/color/image_raw [sensor_msgs/Image] 1 publisher
 * /move_base_node/MIRPlannerROS/cost_cloud [sensor_msgs/PointCloud2] 1 publisher
 * /camera_floor_left/driver/color/camera_info [sensor_msgs/CameraInfo] 1 publisher
 * /LightCtrl/us_list [sensor_msgs/Range] 1 publisher
 * /robot_mode [mirMsgs/RobotMode] 1 publisher
 * /mirwebapp/laser_map_metadata [mirMsgs/LocalMapStat] 1 publisher
 * /move_base_node/MIRPlannerROS/global_plan [nav_msgs/Path] 1 publisher
 * /camera_floor_left/driver/color/image_raw/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /move_base/status [actionlib_msgs/GoalStatusArray] 1 publisher
 * /robot_state [mirMsgs/RobotState] 1 publisher
 * /b_raw_scan [sensor_msgs/LaserScan] 1 publisher
 * /particlevizmarker [visualization_msgs/MarkerArray] 1 publisher
 * /one_way_map [nav_msgs/OccupancyGrid] 1 publisher
 * /mirEventTrigger/events [mirMsgs/Events] 1 publisher
 * /wifi_diagnostics/roam_events [mir_wifi_msgs/WifiRoamEvent] 2 publishers
 * /camera_floor_right/driver/depth/image_rect_raw [sensor_msgs/Image] 1 publisher
 * /tf [tf2_msgs/TFMessage] 4 publishers
 * /move_base_node/global_costmap/obstacles [nav_msgs/GridCells] 1 publisher
 * /camera_floor_left/driver/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /move_base_node/traffic_costmap/obstacles [nav_msgs/GridCells] 1 publisher
 * /session_importer_node/info [mirSessionImporter/SessionImportInfo] 1 publisher
 * /MC/currents [sdc21x0/MotorCurrents] 1 publisher
 * /active_mapping_guid [std_msgs/String] 1 publisher
 * /mirwebapp/web_path [mirMsgs/WebPath] 1 publisher
 * /SickPLC/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /move_base_node/global_costmap/robot_footprint [geometry_msgs/PolygonStamped] 1 publisher
 * /MissionController/CheckArea/visualization_marker [visualization_msgs/Marker] 1 publisher
 * /camera_floor_left/driver/depth/image_rect_raw/compressed/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /move_base_node/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /camera_floor_left/driver/color/image_raw [sensor_msgs/Image] 1 publisher
 * /mir_amcl/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /move_base_node/current_goal [geometry_msgs/PoseStamped] 1 publisher
 * /camera_floor_right/filter/visualization_marker [visualization_msgs/Marker] 1 publisher
 * /camera_floor_left/driver/color/image_raw/compressed/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /camera_floor_right/driver/depth/camera_info [sensor_msgs/CameraInfo] 1 publisher
 * /camera_floor_left/driver/depth/camera_info [sensor_msgs/CameraInfo] 1 publisher
 * /data_events/positions [mir_data_msgs/PositionEvent] 1 publisher
 * /move_base_node/visualization_marker [visualization_msgs/Marker] 1 publisher
 * /MissionController/goal_position_guid [std_msgs/String] 1 publisher
 * /camera_floor_right/driver/depth/image_rect_raw/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /mir_log [rosgraph_msgs/Log] 1 publisher
 * /camera_floor_left/driver/color/image_raw/compressed [sensor_msgs/CompressedImage] 1 publisher
 * /move_base_node/global_costmap/inflated_obstacles [nav_msgs/GridCells] 1 publisher
 * /move_base_node/MIRPlannerROS/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /camera_floor_right/driver/color/camera_info [sensor_msgs/CameraInfo] 1 publisher
 * /wifi_diagnostics/wifi_ap_time_stats [mir_wifi_msgs/APTimeStats] 1 publisher
 * /scan_filter/visualization_marker [visualization_msgs/Marker] 1 publisher
 * /robot_status [mirMsgs/RobotStatus] 1 publisher
 * /move_base_node/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /f_scan [sensor_msgs/LaserScan] 1 publisher
 * /camera_floor_left/driver/detect_pallet_obstacles [sensor_msgs/PointCloud2] 1 publisher
 * /wifi_diagnostics/cur_ap [mir_wifi_msgs/APInfo] 1 publisher
 * /data_events/area_events [mir_data_msgs/AreaEventEvent] 1 publisher
 * /odom_enc [nav_msgs/Odometry] 1 publisher
 * /move_base_node/MIRPlannerROS/updated_global_plan [mirMsgs/PlanSegments] 1 publisher
 * /localization_score [std_msgs/Float64] 1 publisher
 * /mir_status_msg [std_msgs/String] 3 publishers
 * /camera_floor_right/background [sensor_msgs/PointCloud2] 1 publisher
 * /move_base/goal [mir_actions/MirMoveBaseActionGoal] 2 publishers
[... 238 lines cut ...]
 * /check_area/polygon [geometry_msgs/PolygonStamped] 1 subscriber
```

And then I have the `rosbag info`. I gave 3 goals through the web interface and I can see 3 messages:

```
path:        mir_move_base.bag
version:     2.0
duration:    1:46s (106s)
start:       Oct 26 2020 13:05:15.30 (1603713915.30)
[... cut ...]
             mir_actions/MirMoveBaseActionResult   [bba42cc37afb0f3bc0d230821314e0cd]
topics:      move_base/cancel       3 msgs    : actionlib_msgs/GoalID                
             move_base/feedback   511 msgs    : mir_actions/MirMoveBaseActionFeedback
             move_base/goal         3 msgs    : mir_actions/MirMoveBaseActionGoal    
             move_base/result       3 msgs    : mir_actions/MirMoveBaseActionResult  
             move_base/status     540 msgs    : actionlib_msgs/GoalStatusArray
```

MartinJensen37, 2020-10-26:
>Did you reduce the time delay between your PC and the MiR?
Yes, the delay currently is 0.2s approx. instead of 20-30s.

mintar, 2020-10-26:
> And then I have the rosbag info. I gave 3 goals through the web interface and I can see 3 messages:

Great! Could you please upload the rosbag?

MartinJensen37, 2020-10-26:
> Great! Could you please upload the rosbag?

Github doesn't allow me to upload the file in the comments, so maybe we can put it somewhere else? Where do you prefer?

mintar, 2020-10-26:
I don't care - Dropbox, Google Drive, Mega.nz... your choice.

MartinJensen37, 2020-10-26:
Here is a link for the .bag file then: https://drive.google.com/file/d/1qOQnfR9kd9vUqDBqvI4wAwpjtUgg9_OB/view?usp=sharing

mintar, 2020-10-26:
It isn't a public link, but I've requested access.

MartinJensen37, 2020-10-26:
> It isn't a public link, but I've requested access.

Oh sorry, it should be accessible now.

matthias-mayr, 2020-10-26:
We are running version 2.8.0.3. I checked our repo and we have no further changes that should be relevant for this issue.
When you have a solution, we can check it on our version as well.

mintar, 2020-10-27:
@MartinJensen37 Thanks for the bagfile. There was indeed a small change in the MirMoveBaseAction between 2.4.0 and 2.8.3 (2e79d947), but I don't think that has anything to do with the problem.

Can you try using an action client instead of the `move_base_simple/goal` topic? For example, you can try running this script:

```python
#!/usr/bin/env python

import rospy
from actionlib import SimpleActionClient, GoalStatus
from geometry_msgs.msg import PoseStamped
from move_base_msgs.msg import MoveBaseAction, MoveBaseGoal

def make_waypoints():
    waypoints = []

    p = PoseStamped()
    p.header.frame_id = 'map'
    p.pose.position.x = 41.39
    p.pose.position.y = 24.24
    p.pose.orientation.z = 0.7071067811865475  # sin(pi/4) (90 degrees yaw)
    p.pose.orientation.w = 0.7071067811865475
    waypoints.append(p)

    p = PoseStamped()
    p.header.frame_id = 'map'
    p.pose.position.x = 39.49
    p.pose.position.y = 24.37
    p.pose.orientation.z = 1.0
    p.pose.orientation.w = 0.0
    waypoints.append(p)

    return waypoints

def return_code_to_string(return_code):
    if return_code == GoalStatus.SUCCEEDED:
        return 'succeeded'
    elif return_code == GoalStatus.PREEMPTED:
        return 'preempted'
    elif return_code == GoalStatus.ABORTED:
        return 'aborted'
    else:
        return str(return_code)

def main():
    rospy.init_node('drive_to_waypoints')
    action_client = SimpleActionClient('move_base', MoveBaseAction)
    if not action_client.wait_for_server(rospy.Duration(30)):
        rospy.logerr('Timeout waiting for move_base action!')
        return
    rospy.loginfo('Connected to move_base action!')

    while True:
        for nr, waypoint in make_waypoints():
            rospy.loginfo("move to pose {}".format(nr))
            goal = MoveBaseGoal()
            goal.target_pose = waypoint
            ret = action_client.send_goal_and_wait(goal)
            rospy.loginfo('move_base result: %s', return_code_to_string(ret))

            if rospy.is_shutdown():
                action_client.cancel_all_goals()
                return

if __name__ == '__main__':
    try:
        main()
    except rospy.ROSInterruptException:
        pass
```

If that doesn't solve it, can you do the following, while using the `mir_bridge` as usual (i.e., don't set the ROS_MASTER_URI to the MiR PC, but to localhost instead)?

* post the output of `rosnode info /mir_bridge`
* do a `rosbag record move_base/result move_base/feedback move_base/status move_base/goal move_base/cancel move_base_simple/goal -O mir_move_base2.bag` while executing the script above or sending `move_base_simple/goal` messages, and upload the bagfile

MartinJensen37, 2020-10-27:
Okay, I've tried the script and it gets stuck while giving the goal at `action_client.send_goal_and_wait(goal)`. It just waits until I break the loop. I'm getting messages from the mir_bridge saying that a stream of messages is being sent to the topic `move_base/feedback` whenever I run the script. This is the content of the .bag file while running the script you sent.
`path:        mir_move_base2.bag
version:     2.0
duration:    45.1s
start:       Oct 27 2020 14:14:25.90 (1603804465.90)
end:         Oct 27 2020 14:15:10.96 (1603804510.96)
size:        177.3 KB
messages:    678
compression: none [1/1 chunks]
types:       actionlib_msgs/GoalStatusArray        [8b2b82f13216d0a8ea88bd3af735e619]
             move_base_msgs/MoveBaseActionFeedback [7d1870ff6e0decea702b943b5af0b42e]
topics:      move_base/feedback   452 msgs    : move_base_msgs/MoveBaseActionFeedback
             move_base/status     226 msgs    : actionlib_msgs/GoalStatusArray
` 
Here is the link to the .bag file where I run the script:https://drive.google.com/file/d/12ZSH0Ehqz4u_4-b2gKRqfxcUwhuw75jQ/view?usp=sharing
And here is the link to the .bag file where I try to send commands with `move_base_simple/goal`: https://drive.google.com/file/d/1Ja-o3IZy_RozYloIqfej6hK136Dpm4QR/view?usp=sharing
Using the `move_base_simple/goal` I could move the robot this time for some reason, however the movement does not correspond to the goal given, it mostly rotate with some slight linear movement. I suspect it has something to do with either delay or the transformations, but I'll try and report back when I've tested a bit more.

MartinJensen37, 2020-10-27:
I just looked at the frames and it seems like a lot of them aren't present, I'm not sure why this is but here is the .pdf:
[frames.pdf](https://github.com/dfki-ric/mir_robot/files/5445736/frames.pdf)

mintar, 2020-10-27:
What about the `rosnode info /mir_bridge`? While you're at it, please also post the output of `rostopic info move_base/goal` and `rostopic info move_base/cancel`.

It's very strange that there are no `move_base/goal` messages in the bagfile.

The robot is doing some very strange things. It always ends up in the same pose that it started from and reports "Goal reached", but that's not actually the goal that you sent.

The tf frames shouldn't have anything to do with the current problem (but they'll need adjustment, too). You could do the following to help:

* comment out the `robot_state_publisher` from `mir.launch`
* `rosbag record --duration=10 /tf /tf_static`

MartinJensen37, 2020-10-28:
Here is the `rosnode info /mir_bridge` output:
```
--------------------------------------------------------------------------------
Node [/mir_bridge]
Publications: 
 * /LightCtrl/us_list [sensor_msgs/Range]
 * /MC/currents [sdc21x0/MotorCurrents]
 * /MissionController/CheckArea/visualization_marker [visualization_msgs/Marker]
 * /SickPLC/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /SickPLC/parameter_updates [dynamic_reconfigure/Config]
 * /amcl_pose [geometry_msgs/PoseWithCovarianceStamped]
 * /b_raw_scan [sensor_msgs/LaserScan]
 * /b_scan [sensor_msgs/LaserScan]
 * /camera_floor/background [sensor_msgs/PointCloud2]
 * /camera_floor/depth/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /camera_floor/depth/parameter_updates [dynamic_reconfigure/Config]
 * /camera_floor/depth/points [sensor_msgs/PointCloud2]
 * /camera_floor/filter/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /camera_floor/filter/parameter_updates [dynamic_reconfigure/Config]
 * /camera_floor/floor [sensor_msgs/PointCloud2]
 * /camera_floor/obstacles [sensor_msgs/PointCloud2]
 * /camera_floor/transform/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /camera_floor/transform/parameter_updates [dynamic_reconfigure/Config]
 * /check_area/polygon [geometry_msgs/PolygonStamped]
 * /diagnostics [diagnostic_msgs/DiagnosticArray]
 * /diagnostics_agg [diagnostic_msgs/DiagnosticArray]
 * /diagnostics_toplevel_state [diagnostic_msgs/DiagnosticStatus]
 * /f_raw_scan [sensor_msgs/LaserScan]
 * /f_scan [sensor_msgs/LaserScan]
 * /imu_data [sensor_msgs/Imu]
 * /laser_back/driver/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /laser_back/driver/parameter_updates [dynamic_reconfigure/Config]
 * /laser_back/transform/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /laser_back/transform/parameter_updates [dynamic_reconfigure/Config]
 * /laser_front/driver/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /laser_front/driver/parameter_updates [dynamic_reconfigure/Config]
 * /laser_front/transform/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /laser_front/transform/parameter_updates [dynamic_reconfigure/Config]
 * /map [nav_msgs/OccupancyGrid]
 * /map_metadata [nav_msgs/MapMetaData]
 * /mir_amcl/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /mir_amcl/parameter_updates [dynamic_reconfigure/Config]
 * /mir_amcl/selected_points [sensor_msgs/PointCloud2]
 * /mir_log [rosgraph_msgs/Log]
 * /mir_serial_button [mir_msgs/Serial]
 * /mir_sound [std_msgs/String]
 * /mir_status [mir_msgs/MirStatus]
 * /mir_status_msg [std_msgs/String]
 * /mirwebapp/grid_map_metadata [mir_msgs/LocalMapStat]
 * /mirwebapp/laser_map_metadata [mir_msgs/LocalMapStat]
 * /move_base/cancel [actionlib_msgs/GoalID]
 * /move_base/feedback [move_base_msgs/MoveBaseActionFeedback]
 * /move_base/goal [move_base_msgs/MoveBaseActionGoal]
 * /move_base/result [move_base_msgs/MoveBaseActionResult]
 * /move_base/status [actionlib_msgs/GoalStatusArray]
 * /move_base_node/MIRPlannerROS/local_plan [nav_msgs/Path]
 * /move_base_node/SBPLLatticePlanner/plan [nav_msgs/Path]
 * /move_base_node/current_goal [geometry_msgs/PoseStamped]
 * /move_base_node/local_costmap/inflated_obstacles [nav_msgs/GridCells]
 * /move_base_node/local_costmap/obstacles [nav_msgs/GridCells]
 * /move_base_node/local_costmap/robot_footprint [geometry_msgs/PolygonStamped]
 * /odom_comb [nav_msgs/Odometry]
 * /odom_enc [nav_msgs/Odometry]
 * /robot_mode [mir_msgs/RobotMode]
 * /robot_pose [geometry_msgs/Pose]
 * /robot_state [mir_msgs/RobotState]
 * /rosout [rosgraph_msgs/Log]
 * /rosout_agg [rosgraph_msgs/Log]
 * /scan [sensor_msgs/LaserScan]
 * /scan_filter/visualization_marker [visualization_msgs/Marker]
[... 110 lines cut ...]
```

And the info for the two topics: 
`move_base/goal`
```
Type: move_base_msgs/MoveBaseActionGoal

Publishers: 
 * /mir_bridge (http://martin-ThinkPad-X250:33733/)

Subscribers: 
 * /mir_bridge (http://martin-ThinkPad-X250:33733/)

```
`move_base/cancel`
```
Type: actionlib_msgs/GoalID

Publishers: 
 * /mir_bridge (http://martin-ThinkPad-X250:33733/)

Subscribers: 
 * /mir_bridge (http://martin-ThinkPad-X250:33733/)
```
I might've connected to the robot wrong yesterday since I cannot reproduce the movement and all the frames are there now. So I tried running your script again with no luck unfortunately.

mintar, 2020-10-28:
Hmm, that all looks normal. I don't think it's a problem with the MiR version, because the [implementation of `wait_for_server`](https://github.com/ros/actionlib/blob/melodic-devel/src/actionlib/action_client.py#L582) just checks if all publishers and subscribers are correctly connected. This happens before even sending anything to the MiR, so the MiR version should be irrelevant for the "Could not connect to'move_base' server" problem.

Perhaps the following helps. In all terminals, before starting any ROS nodes, run the following commands:

```bash
export ROS_IP=<your wifi IP>
export ROS_MASTER_URI=http://<your wifi IP>:11311
```

MartinJensen37, 2020-10-28:
Now I exported my IP while connected to the WiFi of the robot. 
```
export ROS_IP=192.168.12.xxx
export ROS_MASTER_URI=http://192.168.12.xxx:11311
```
I still get the message:
```
[INFO] [1603876175.006362]: [/mir_bridge] starting to stream messages on topic 'amcl_pose'
[INFO] [1603876175.012143]: [/mir_bridge] starting to stream messages on topic 'move_base_node/MIRPlannerROS/local_plan'
[INFO] [1603876175.013758]: [/mir_bridge] starting to stream messages on topic 'move_base_node/SBPLLatticePlanner/plan'
[INFO] [1603876175.015960]: [/mir_bridge] starting to stream messages on topic 'MissionController/CheckArea/visualization_marker'
[WARN] [1603876191.260923]: Could not connect to 'move_base' server after two seconds. Dropping goal.
[WARN] [1603876191.263091]: Did you activate 'planner' in the MIR web interface?
```
When I run Rviz and give the first goal it doesn't come with the could not connect to `move_base` but the second time I do it the message above pops up.

mintar, 2020-10-28:
I'm out of ideas, sorry.

MartinJensen37, 2020-10-28:
No problem, I'll try a bit more with some adjustments and I'll write if anything works. Thanks for all the help!

mintar, 2021-06-11:
I have just pushed a large update to the `melodic` and `noetic` branches. The branch `melodic-2.8` is fully merged and no longer necessary. This issue is fixed now.

closed 2021-06-11 by mintar