DFKI-NI/mir_robot #46: issue in the source for gazebo simulation in melodic ver.

LeonJung, 2020-02-06:
Hi,

I found that current melodic source base - gazebo simulation doesn't work (by following the commands described in README) with navigation stack, 
since the gazebo xacro file for sick_s300 urdf doesn't have (I guess) correct frameName for libgazebo_ros_gpu_laser.so.

And also is there anything to merge two sick lidar datas (front / back) together to output /scan?
since the navigation stack wants /scan topic, not /b_scan and /f_scan.

I modified those two things like,
putting ${prefix} in front of ${link},
and changed topicName from ${topic} to scan (I know that this will make /b_scan data and /f_scan data be mixed and published, so it may cause some trouble on localization)
and now the simulation works perfect with navigation stack.

or if there is other solutions that I couldn't catch, please suggest that.

Cheers.

mintar, 2020-02-19:
> I found that current melodic source base - gazebo simulation doesn't work (by following the commands described in README) with navigation stack,
> since the gazebo xacro file for sick_s300 urdf doesn't have (I guess) correct frameName for libgazebo_ros_gpu_laser.so.

I have just tested it, and it all works correctly. If it doesn't work on your machine, you probably don't have a graphics card that works with Gazebo. Please try again on a PC with a different GPU.

> And also is there anything to merge two sick lidar datas (front / back) together to output /scan?
> since the navigation stack wants /scan topic, not /b_scan and /f_scan.

Yes, that's already included and running:

https://github.com/dfki-ric/mir_robot/blob/bc6c4ff4f9a980ec25f2159af72295411b7d09f2/mir_gazebo/launch/mir_gazebo_common.launch#L38-L40

> I modified those two things like,
> putting ${prefix} in front of ${link},
> and changed topicName from ${topic} to scan (I know that this will make /b_scan data and /f_scan data be mixed and published, [...]

There is no need to modify anything. It already works the way you want it. :)

mintar, 2020-05-01:
I've committed a fix for this problem!

closed 2020-05-01 by mintar