DFKI-NI/mir_robot #51: Long term support for MIR software versions 2.8 or above

FaseehCS, 2020-03-26:
Hi, 

I was wondering if there was going to be a long term support for MIR software version 2.7 or higher?
The current repo is for 2.3.1 or lower. Although, some of the issues for later software versions have been address in other Issues section but there might some other issues which have not yet been encountered.
Thanks

mintar, 2020-03-26:
I don't have MiR software 2.8 available for testing. The newest version that I have is 2.6.

At some point in the future, I plan on upgrading our robot to 2.6 and updating this repo with all necessary changes, so that it also supports version 2.6; this should solve most problems for 2.8 as well, I believe. This is however not high priority for me, I'm swamped with other projects. Also, due to the Coronavirus pandemic, I'm working from home at the moment and don't have access to the MiR robot, so it'll have to wait until afterwards. Sorry for the inconvenience.

mintar, 2020-06-30:
Let's continue the discussion in #45.

closed 2020-06-30 by mintar