DFKI-NI/mir_robot #52: Gazebo demo - mapping

ecarrasc, 2020-04-30:
Hi,

I was following the instructions for mapping with the mir100 and I encountered some problems. First, under ### mapping: I think it should be 

`roslaunch mir_navigation hector_mapping.launch` without the xml extension.

But this is a minor issue. I have troubles with the command under # navigation:

`roslaunch mir_navigation move_base.xml with_virtual_walls:=false`

I get this output in the terminal

`[ WARN] [1588239096.440906321, 69.100000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 69.1 timeout was 0.1.
[ INFO] [1588239096.528079368, 69.160000000]: Using plugin "navigation_map"
[ INFO] [1588239096.548217776, 69.168000000]: Requesting the map...
[ INFO] [1588239096.822569493, 69.372000000]: Resizing costmap to 2048 X 2048 at 0.050000 m/pix
[ INFO] [1588239096.963045774, 69.469000000]: Received a 2048 X 2048 map at 0.050000 m/pix
[ INFO] [1588239096.968869319, 69.472000000]: Using plugin "obstacles"
[ INFO] [1588239096.982339040, 69.481000000]:     Subscribed to Topics: b_scan_marking b_scan_clearing f_scan_marking f_scan_clearing
[ INFO] [1588239097.203873307, 69.606000000]: Using plugin "inflation"
[ INFO] [1588239097.327970419, 69.656000000]: Name is SBPLLatticePlanner
[ INFO] [1588239099.975276407, 70.739000000]: Planning with ARA*
[ INFO] [1588239099.976435074, 70.740000000]: [sbpl_lattice_planner] Initialized successfully
[ INFO] [1588239100.063663768, 70.801000000]: Using plugin "obstacles"
[ INFO] [1588239100.086832803, 70.810000000]:     Subscribed to Topics: b_scan_marking b_scan_clearing f_scan_marking f_scan_clearing
[ INFO] [1588239100.318272605, 70.926000000]: Using plugin "inflation"
[ INFO] [1588239100.438672378, 70.980000000]: Created local_planner nav_core_adapter::LocalPlannerAdapter
[ INFO] [1588239100.441480870, 70.984000000]: Loading plugin dwb_local_planner::DWBLocalPlanner
[ INFO] [1588239100.493732191, 71.007000000]: Using Trajectory Generator "dwb_plugins::StandardTrajectoryGenerator"
[ INFO] [1588239100.597755627, 71.056000000]: Using Goal Checker "dwb_plugins::SimpleGoalChecker"
[ INFO] [1588239100.618203343, 71.068000000]: Using critic "RotateToGoal" (dwb_critics::RotateToGoalCritic)
[ INFO] [1588239100.627164351, 71.071000000]: Using critic "ObstacleFootprint" (dwb_critics::ObstacleFootprintCritic)
[ INFO] [1588239100.634960493, 71.075000000]: Using critic "PathAlign" (dwb_critics::PathAlignCritic)
[ INFO] [1588239100.652369691, 71.084000000]: Using critic "PathDistPruned" (mir_dwb_critics::PathDistPrunedCritic)
[ INFO] [1588239100.657793369, 71.087000000]: Using critic "PathProgress" (mir_dwb_critics::PathProgressCritic)
[ INFO] [1588239100.783721242, 71.172000000]: Recovery behavior will clear layer obstacles
[ INFO] [1588239100.785131502, 71.175000000]: Recovery behavior will clear layer obstacles
[ INFO] [1588239101.025901727, 71.326000000]: odom received!
`
Then I run 

`rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz`

In the rviz window no black lines or dots are drawn and while commanding the robot to move this does move in gazebo but nor in rviz.

I tested this on Ubuntu 16.04 / Kinetic Kame respectively 18.04 / Melodic, both running on VMWare Fusion 11.5 on a Mac.

I tried to find out what it could be, but with no success.

Thanks in advance

ecarrasc, 2020-04-30:
OK, I just saw that this issue or similar was already treated in #30 (closed). I will give it a try.

ecarrasc, 2020-04-30:
In mapping case it does not work, but for the Gazebo demo with an existing map it works.
Why is that? For the navigation it requires the /scan, too, right?

mintar, 2020-05-01:
>  I think it should be roslaunch mir_navigation hector_mapping.launch without the xml extension.

Thanks, fixed!

> In the mapping case it does not work within the VM. Is there any setting that can be changed in order to get a correct /scan message?

I think I have solved the issue. I've committed a fix for this problem, so please pull the latest code and let me know whether it works!

ecarrasc, 2020-05-01:
It works! After starting the gazebo part I echoed /scan once and I got a plausible output. Then I went through the other commands and it is working. I am adding a screenshot of rviz.

[image]

Thanks!

mintar, 2020-05-01:
Thanks for the feedback. Glad I could help!

closed 2020-05-01 by mintar