DFKI-NI/mir_robot #53: demo mapping error

Vicky49364, 2020-05-04:
Hi! I was following the mapping procedure. But when I run Rviz the map frame does not exist. I am using melodic, gazebo9. 
[image]

mintar, 2020-05-04:
I've just tested it myself, and it works without problems, so it's something on your end. If you know how to use Docker, you can install nvidia-docker and try it yourself with the following Dockerfile:

```Dockerfile
FROM ros:melodic-ros-base

# install the GL Vendor-Neutral Dispatch library and Gazebo
RUN apt-get update && apt-get install --no-install-recommends -y \
    libglvnd0 \
    build-essential \
    python-catkin-tools \
    ros-melodic-gazebo-* \
    && rm -rf /var/lib/apt/lists/*

# nvidia-container-runtime
ENV NVIDIA_VISIBLE_DEVICES \
    ${NVIDIA_VISIBLE_DEVICES:-all}
ENV NVIDIA_DRIVER_CAPABILITIES \
    ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}graphics

# Some QT-Apps don't not show controls without this
ENV QT_X11_NO_MITSHM 1

# Create ROS workspace
RUN mkdir -p /ws/src/
WORKDIR /ws
RUN cd src && \
    git clone -b melodic https://github.com/dfki-ric/mir_robot

# Use rosdep to install all dependencies (including ROS itself)
RUN apt-get update && \
    rosdep install --from-paths src -i -y --rosdistro melodic && \
    rm -rf /var/lib/apt/lists/*

RUN /bin/bash -c "source /opt/ros/melodic/setup.bash && \
    catkin init && \
    catkin config --install -j 1 -p 1 && \
    catkin build --limit-status-rate 0.1 --no-notify && \
    catkin build --limit-status-rate 0.1 --no-notify --make-args tests"

RUN /bin/bash -c "echo 'source /ws/install/setup.bash' >> /root/.bashrc"
```

Now build and run it like this:

```bash
docker build -t ros-melodic-gazebo-mir-robot .

xhost +local:root

nvidia-docker run -it \
    --privileged \
    --network=host \
    --env="DISPLAY" \
    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    --volume="$HOME/code/ros-docker-gazebo/host_docker:/home/user/host_docker" \
     ros-melodic-gazebo-mir-robot

xhost -local:root
```

If it works in Docker but not on your machine, you can now try and figure out where the difference is. If it also doesn't work in Docker, something is *very* strange. Please tell me in that case. Good luck! :)

Vicky49364, 2020-05-12:
Hi. Sorry for the late reply. It worked. Thank you! @mintar

closed 2020-05-12 by Vicky49364