DFKI-NI/mir_robot #54: Mir is not moving in a straight line. (mir+ur5)

Emily194, 2020-05-23:
Hi. I am currently struggling with this issue and I have no idea why mir is not moving in a straight line. Do you know what could be the possible cause of error? Thank you!
[image]

mintar, 2020-05-26:
Are you sure that the path is curving? Or is the start simply offset by a bit? It's a bit hard to tell by the video.

In my experiments, there is a slight offset at the start. This is caused by small perturbations in the simulation while the robot is at rest. However, the path is not significantly curved. Are any warnings printed to the terminal when you start your Gazebo simulation?

[image]

[image]

mintar, 2020-05-26:
Also, I've plotted the data via plotjuggler, and the path also looks to be straight:

[image]

closed 2020-06-29 by mintar