DFKI-NI/mir_robot #59: Add documentation about MIR and mir_bridge functionalities

matthias-mayr, 2020-07-09:
We recently got a new MIR200 and since it comes with a new software version, it was not directly compatible with this ROS driver.

We spent quite some time to figure out, how the MIR would work, how it connects with rosbridge and why it was not working right away. With the insight we obtained up to now, we could have saved quite some of it, if there would have been some more documentation, e.g.

- That MIR is actually running ROS internally (although that didn't really take long to figure out)
- A figure of the architecture like `ROS - mir_driver - rosbridge - rosbridge - MIR`
- That one [can actually just connect to the roscore on MIR](https://github.com/dfki-ric/mir_robot/issues/45#issuecomment-655979633). I misread the network specifications and thought that all the ports would be closed, so I though our only access would be through rosbridge.

All the manuals I got with the MIR leave out, that it runs ROS and a lot of other tips and tricks that appear in various issues here. It seems to me that it would be a lot of help for people if this implicit knowledge would be expressed somewhere.
What do you think?

mintar, 2020-07-09:
Yes, any updates to the documentation would be more than welcome!

matthias-mayr, 2020-07-09:
Do you prefer a set of markdown files or to start a wiki?

mintar, 2020-07-09:
I have a slight preference for a set of markdown files.

liesrock, 2020-10-09:
Hi @matthias-mayr, @mintar I can help on this as well.

@matthias-mayr can you provide the above documentation even if is just a draft?

matthias-mayr, 2020-10-14:
This issue is still in the back of my head, but I am totally busy until the first week of November.

matthias-mayr, 2021-02-02:
[image]

For convenience a jpg export here. Comments are appreciated.