DFKI-NI/mir_robot #60: Goals send via /move_base_simple do not work with MIR software version > 2.7

FaseehCS, 2020-07-10:
When I send goal to `/move_base_simple`, e.g through Rviz, MIR starts doing random orientations and does not go to the goal. There are two behaviours I found. In both behaviours, MIR just moves a little and starts oscillating. I have pasted the system log for both the behaviours. In the first behaviour, there was no error but in the second behaviour, I received an error
1) 
```
MC | Setting parameter using command: ^KD 2 2 | 17:53:07
-- | -- | --
  | MC | Setting parameter using command: ^KI 2 7 | 17:53:07
  | MC | Setting parameter using command: ^KP 2 4 | 17:53:07
  | MC | Setting parameter using command: ^KD 1 2 | 17:53:07
  | MC | Setting parameter using command: ^KI 1 7 | 17:53:07
  | MC | Setting parameter using command: ^KP 1 4 | 17:53:07
  | move_base_node | MirLocalPlanner: State change 'TurnToGoal' -> 'Stopping' | 17:53:07
  | move_base_node | MirLocalPlanner: State change 'GoalSearch' -> 'TurnToGoal' | 17:53:04
  | move_base_node | MirLocalPlanner: State change 'FollowPath' -> 'GoalSearch' | 17:53:03
  | move_base_node | MirLocalPlanner: State change 'TurnToTarget' -> 'FollowPath' | 17:53:03
  | move_base_node | MirLocalPlanner: State change 'FollowPath' -> 'TurnToTarget' | 17:52:59
  | move_base_node | Clearing segment with index = 0 | 17:52:59
  | move_base_node | MirLocalPlanner: State change 'TurnToTarget' -> 'FollowPath' | 17:52:58
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'TurnToTarget' | 17:52:58
  | move_base_node | MirLocalPlanner: State change 'TurnToTarget' -> 'Stopping' | 17:52:58
  | move_base_node | mirLocalPlanner: Got new plan - Length: 0.235928, Segments: 2 - Starting with forward motion | 17:52:58
  | move_base_node | MirLocalPlanner: Got new plan from (43.9417, 5.9917, 0.7854) to (43.95, 6.1, 0.0322469) | 17:52:58
  | move_base_node | SBPLPlanner: Succesfully created plan - Total time: 0.034 seconds - Total expands: 362 - Expands pr sec: 10672.7 - Final epsilon: 1.00 | 17:52:58
  | move_base_node | SBPLPlanner: Making plan from (43.94, 5.99, 37.5) to (43.95, 6.10, 1.8) | 17:52:58
  | move_base_node | ** MirEscapeRecovery: Succeded! ** | 17:52:58
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:57
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:56
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:55
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:54
  | move_base_node | ** MirEscapeRecovery started! ** | 17:52:53
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'TurnToTarget' | 17:52:53
  | move_base_node | MirLocalPlanner: State change 'TurnToTarget' -> 'Stopping' | 17:52:53
  | move_base_node | mirLocalPlanner: Got new plan - Length: 0.349652, Segments: 1 - Starting with backwards motion | 17:52:53
  | move_base_node | MirLocalPlanner: Got new plan from (44.225, 6.175, 0.3927) to (43.95, 6.1, 0.0322469) | 17:52:53
  | move_base_node | SBPLPlanner: Succesfully created plan - Total time: 0.038 seconds - Total expands: 1218 - Expands pr sec: 31990.3 - Final epsilon: 1.00 | 17:52:53
  | move_base_node | SBPLPlanner: Making plan from (44.23, 6.19, 33.8) to (43.95, 6.10, 1.8) | 17:52:53
  | move_base_node | ** MirEscapeRecovery: Succeded! ** | 17:52:53
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:53
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:52
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:51
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:52:50
  | move_base_node | ** MirEscapeRecovery started! ** | 17:52:50
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'TurnToTarget' | 17:52:50
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'Stopping' | 17:52:50
  | move_base_node | mirLocalPlanner: Got new plan - Length: 0, Segments: 1 - Starting with forward motion | 17:52:50
  | move_base_node | MirLocalPlanner: Got new plan from (43.95, 6.1, 0.0322469) to (43.95, 6.1, 0.0322469) | 17:52:50
  | move_base_node | SBPLPlanner: Succesfully created plan - Total time: 0.008 seconds - Total expands: 583 - Expands pr sec: 76093.4 - Final epsilon: 1.00 | 17:52:50
  | move_base_node | SBPLPlanner: Making plan from (43.93, 6.10, 1.7) to (43.95, 6.10, 1.8) | 17:52:50
```
2)

```
MC | Setting parameter using command: ^KD 2 2 | 17:55:36
-- | -- | --
  | MC | Setting parameter using command: ^KI 2 7 | 17:55:36
  | MC | Setting parameter using command: ^KP 2 4 | 17:55:36
  | MC | Setting parameter using command: ^KD 1 2 | 17:55:36
  | MC | Setting parameter using command: ^KI 1 7 | 17:55:36
  | MC | Setting parameter using command: ^KP 1 4 | 17:55:36
  **| move_base_node | Aborting because the robot appears to be oscillating over and over. Even after executing all recovery behaviors | 17:55:36**
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'TurnToTarget' | 17:55:36
  | move_base_node | MirLocalPlanner: State change 'TurnToTarget' -> 'Stopping' | 17:55:36
  | move_base_node | mirLocalPlanner: Got new plan - Length: 0.337848, Segments: 1 - Starting with backwards motion | 17:55:36
  | move_base_node | MirLocalPlanner: Got new plan from (44.225, 6.025, -0.392685) to (43.95, 6.1, 0.0322469) | 17:55:36
  | move_base_node | SBPLPlanner: Succesfully created plan - Total time: 0.031 seconds - Total expands: 1104 - Expands pr sec: 36003.7 - Final epsilon: 1.00 | 17:55:36
  | move_base_node | SBPLPlanner: Making plan from (44.23, 6.01, -34.2) to (43.95, 6.10, 1.8) | 17:55:36
  | move_base_node | ** MirEscapeRecovery: Succeded! ** | 17:55:36
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:55:36
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:55:34
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:55:33
  | move_base_node | ** MirEscapeRecovery: Trying to escape.. ** | 17:55:32
  | move_base_node | ** MirEscapeRecovery started! ** | 17:55:32
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'TurnToTarget' | 17:55:32
  | move_base_node | MirLocalPlanner: State change 'Stopping' -> 'Stopping' | 17:55:32
  | move_base_node | mirLocalPlanner: Got new plan - Length: 0, Segments: 1 - Starting with forward motion | 17:55:32
  | move_base_node | MirLocalPlanner: Got new plan from (43.95, 6.1, 0.0322469) to (43.95, 6.1, 0.0322469) | 17:55:32
  | move_base_node | SBPLPlanner: Succesfully created plan - Total time: 0.009 seconds - Total expands: 583 - Expands pr sec: 65685.9 - Final epsilon: 1.00 | 17:55:32
  | move_base_node | SBPLPlanner: Making plan from (43.94, 6.10, 1.5) to (43.95, 6.10, 1.8) | 17:55:32
```

Did you encounter this problem in the previous versions of MIR<= 2.3? If yes, any solutions?

mintar, 2020-07-11:
Please see #45. For now, you would have to use the branch `melodic-2.8`, but I'm planning to incorporate this into the main `kinetic`/`melodic`/`noetic` branches at the end of next month after vacation time is over.

matthias-mayr, 2020-07-13:
_Just for clarification: this happens with the changes in `melodic-2.8`.
It was posted separately since such issues can be hard to track down and #45 might not benefit from the introduction of another subtopic._

Another piece of information that can be added is that MIR works as expected if one sends goals through the `/move_base` action. So in the worst worst case mir_bridge can subscribe to `/move_base_simple` on the computer side and translate it to an action.

mintar, 2020-07-13:
Ok, thanks for the clarification! I'm reopening this issue.

> So in the worst worst case mir_bridge can subscribe to /move_base_simple on the computer side and translate it to an action.

Yes, this is probably necessary. PR welcome!

mintar, 2020-07-20:
Closed via #62 .

PR #62 (2020-07-20): Fix: Converts move_base_simple/goal into a move_base action.
At least MIR software version 2.8 did not react properly to goal messages. This implements a workaround.
Closes #60.

I wasn't entirely sure where to plug this in. `dict_filter` always output an altered message and in this case, we need to forward the message to different topics. This way it should be transparent for debugging, e.g. with `rqt_graph`.
There is also the nice side effect that it can print an error message if `planner` on the MIR is not activated.

Let me know what you think. This PR allows edits.
commit 4d9cefc6 (2020-07-20): Fix: Converts move_base_simple/goal into a move_base action. (#62)

At least MIR software version 2.8 does not react properly to move_base_simple/goal messages. This implements a workaround.
Closes #60.
closed 2020-07-20 by mintar