DFKI-NI/mir_robot #61: mir cannot move in rviz

Liping65, 2020-07-12:
Hi! I have created a moveit package for mir with a manipulator. However, when I add this in my launch file mir is not able to move around in rviz when I launch the move_group.
`````
     <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
          <rosparam param="source_list">[mir/joint_states]</rosparam>
          <param name="rate" value="200.0" />
      </node>  
`````

closed 2020-07-13 by Liping65

saralopez7, 2021-03-17:
Hi, could you share the package you created? I currently struggling with the same problem - creating a mobile robot manipulator using a MiR base and a UR arm.

mintar, 2021-04-29:
@saralopez7: If you're still struggling with this, you could have a look at the package that we've created for our own mobile manipulator "Mobipick" (MiR100 base + UR5 arm): https://github.com/DFKI-NI/mobipick