DFKI-NI/mir_robot #63: MiR100 sensors in Gazebo

DanWang1230, 2020-08-04:
Hi there, thank you so much for developing and sharing this ROS driver! It helps a lot.

We have bought a new mir100 recently. From the user manual, we see that the robot has multiple sensors: two laser scanners, two 3D cameras, four ultrasound sensors, and internal sensors (e.g., IMU, motor encoder, accelerometer, wheel encoders). Are all of them integrated into the mir gazebo? From the urdf file, we noticed some may be missing?

We are also trying to pair the sensors with the ros topics. For example, laser scanners --> /f_scan and /b_scan and IMU --> /imu_data. How about the other sensors, especially the 3D camera?

xCzsr, 2020-08-08:
I have access to a MiR100. I too was having the same issue with accessing the Intel RealSense camera data. 
Like @DanWang1230 I noticed that you didn't have a Intel RealSense sensor plugins for the MiR100 urdf file.  

One thing that I have done is, on my local computer I do:
$ export ROS_MASTER_URI=http://MIR_IP:11311
$ rostopic list 

rostopic list gives me all of the current topics that are on the MiR100. 
I then find the message type of the topic and I then add the topics onto mir_bridge.py. 
No luck with this method.

Awesome work by the way! This repo is great, keep up the great work!

mintar, 2021-06-11:
If somebody wants to add more sensors to the Gazebo simulation, pull requests are welcome!