DFKI-NI/mir_robot #64: Tf frame remapping

ibMH, 2020-08-06:
# Tf remapping error:
Since the mir_bridge seems to do a frame reparenting everytime its starts to stream messages to the tf topic it sometimes happens that tf_prefixes occure repeated:
[image]
Especially the odom frame is subjected to this error. 

# Reproduction:
This error is hardly reproduceable, since it does not occure every time a tf listener is launched. But the following steps lead to it very often:

1. Launch the mir.launch in a given namespace and with the tf_prefix arg set
2. Wait til everythign is setup
3. Execute a program that looks up mir transformations (In my case I started the tf tree plugin for rqt (see above))
4. If you got more than one line that says  "starting to stream messages on topic 'tf'":

[image]
 the error should be there.

mintar, 2020-08-10:
Something very strange is going on. There are several things I noticed:

* The duplicate message "starting to stream messages on topic 'tf'" is probably caused by a race condition, when two TF listeners try to connect to the `mir_driver` at the same time. This could probably be solved by adding a [threading.Lock](https://docs.python.org/2/library/threading.html#lock-objects) to the `PublisherWrapper.peer_subscribe()` method. I don't see how this could cause the double remapping bug though.
* Most of the frames published by `tf_remove_state_publisher_frames` shouldn't be there. To fix this, you'll have to edit mir_driver/launch/mir.launch and add `mur/` in front of all the frames, i.e., change `base_link` to `mur/base_link` and so on.

The only explanation that I can think of is if you are using the wrong `ROS_MASTER_URI`. Make sure that you use something like `export ROS_MASTER_URI=http://localhost:11311`, and **not** the MiR IP. If that doesn't fix it, could you please run `rosparam dump params.yaml` and paste the file `params.yaml` here?

ibMH, 2020-08-18:
Parameter file:
[params.txt](https://github.com/dfki-ric/mir_robot/files/5088928/params.txt)

- I can't see how a race condition could occure between two subscribtions to tf. I do them manually from command prompt (delay >3 sec). I guees a problem with multiple subscribers to the topic? Seems like the first tf_listener does everything well and as the second occures the tf dict filters dont work properly anymore.
- I was able to fix the recursive reparenting. I just checked if a frame_id already contains the tf_prefix and if so no new prefix is added. Therefore the tf_prefix/tf_prefix/tf_prefix/frame error seems not to occure anymore.

mintar, 2021-06-11:
I have just pushed a large update to the `melodic` and `noetic` branches. This issue should be fixed now.

closed 2021-06-11 by mintar