DFKI-NI/mir_robot #66: Publishing on topic /cmd_vel not working

alibinjunaid, 2020-09-25:
I have followed all the issues related to publishing on the /cmd_vel topic. There is a melodic-2.8 branch for the recent software version on mir but it does not work. Seems like it works on mir SW v2.7.2 but does not work 2.8.3.

Any help or update would be appreciated!

mintar, 2020-09-25:
I don't have MiR software 2.8.3 available for testing, so I can't help you here. Maybe this is related to #30?

liesrock, 2020-10-12:
@alibinjunaid you can take a look here: https://github.com/dfki-ric/mir_robot/issues/30#issuecomment-707203907

mintar, 2020-10-23:
Did this solve your problem, @alibinjunaid ?

mintar, 2020-10-23:
@liesrock : Can you confirm that this is working for you on MiR version 2.8.3?

liesrock, 2020-10-24:
@mintar I did the tests with 2.7.9 , but in the near future I will test the above also with 2.8.3

closed 2021-05-28 by mintar