DFKI-NI/mir_robot #70: Running the mir_driver on software version 2.8.3

MartinJensen37, 2020-10-23:
I'm rather new to the MiR platform and using ROS bridge so I'll post more information if it is needed. I'm trying to install the melodic 2.8 from source right now. However I get following cmake error:  
[image]
To overcome this I can run catkin_make_isolated, this builds the package then once I run the `roslaunch mir_driver mir.launch` the everything seems to work fine I can see all the sensor data in rviz, however I'm unsure how to give commands to the MiR itself. In the move_base/goal topic the msg type is still MoveBaseActionGoal shouldn't it be MirMoveBaseActionGoal or is there an underlying conversion to MirMoveBaseActionGoal? I have also tried using move_base_simple/goal through rviz this doesn't seem to work.

Another thing is syncing the transformations, when I run roswtf or tf_monitor I can see the following delay(at base_footprint) of 95 secs.
[image]

I hope this isn't too confusing! Thanks in advance!

mintar, 2020-10-23:
> I'm rather new to the MiR platform and using ROS bridge so I'll post more information if it is needed. I'm trying to install the melodic 2.8 from source right now. However I get following cmake error:

The warning about non-standard content is because the mir_robot/CMakeLists.txt specifies a minimum CMake version of 3.5.1, but  2.8.3 is expected. This can be ignored. If I change it back, I suspect that the buildfarm will start complaining again that cmake_policy CMP0048 is not set... `¯\_(ツ)_/¯`

Strangely enough, I couldn't reproduce the warning. Here are the exact commands I used:

```bash
sudo apt-get update
sudo apt-get install -y build-essential python-catkin-tools python-rosdep cmake python-catkin-lint
source /opt/ros/melodic/setup.bash
mkdir -p catkin_ws/src
cd catkin_ws/src
git clone https://github.com/dfki-ric/mir_robot.git -b melodic-2.8
cd ..
rosdep update
sudo rosdep install -i --from-paths src -y
catkin_make
```

The *error* about non-catkin packages means that you've copied a pure CMake based project into your catkin workspace. That has nothing to do with the `mir_robot` repo; all packages here are catkin packages.

> In the move_base/goal topic the msg type is still MoveBaseActionGoal shouldn't it be MirMoveBaseActionGoal or is there an underlying conversion to MirMoveBaseActionGoal?

There's an underlying conversion from MoveBaseAction to MirMoveBaseAction.

> I have also tried using move_base_simple/goal through rviz this doesn't seem to work.

That's a bummer, it's working for other people (see #62). A couple of questions:

* Can you publish to other topics? E.g., if you publish on the `cmd_vel` topic (for example by using `rqt_robot_steering`, the robot should move.
* Is your PC directly connected to the MiR, or is there a third PC in between?
* Are you **sure** that you are using the `melodic-2.8` branch?
* Is the move_base "planner" enabled on the MiR web interface? (System --> Processes --> planner)
* Are there any warnings printed when you run the mir_bridge?

BTW, there's another person reporting problems with 2.8.3: #66

> Another thing is syncing the transformations, when I run roswtf or tf_monitor I can see the following delay(at base_footprint) of 95 secs.
> [image]

The only solution for this is to log into the MiR web interface and try to synchronize the MiR system time with your PC's time as well as you can. Also see: https://github.com/dfki-ric/mir_robot/issues/45#issuecomment-652397685

MartinJensen37, 2020-10-24:
Hello Martin! Thanks for the answer. Here are some of the answers to your questions 

- Can you publish to other topics? Yes, I can send commands over /cmd_vel and the robot runs fine right now.

- Is your PC directly connected to the MiR, or is there a third PC in between? I'm directly connecting to the MiR's WiFi with my 
  laptop. 
- Are you sure that you are using the melodic-2.8 branch? I'm pretty sure I'm using this branch since I ran: `git clone https://github.com/dfki-ric/mir_robot.git -b melodic-2.8`

- Is the move_base "planner" enabled on the MiR web interface? (System --> Processes --> planner) Yes, I read the other thread #45 and made sure the planner was running. 

- Are there any warnings printed when you run the mir_bridge? So far there are no major warnings, but here are some screenshots of the warnings about some of the topics just to be sure: 
[image]
[image]
[image]

I have access to the robot again on Monday, so there I'll try to sync up the robot to decrease the delay and then I'm gonna try to publish the MoveBaseActionGoal to the move_base/goal again. What frames should I publish the goal? So far I think I've tried /odom, /map and /base_footprint just to be sure.

mintar, 2021-06-11:
I have just pushed a large update to the `melodic` and `noetic` branches. The branch `melodic-2.8` is fully merged and no longer necessary. This issue should be fixed now.

closed 2021-06-11 by mintar