DFKI-NI/mir_robot #71: gazebo demo existing map - Robot cannot avoid obstacle

ecarrasc, 2020-10-28:
Hi
I hope I can describe the issue correctly.
I am working with
Ubuntu 20.04
ROS noetic

I installed the mir_robot package as follows
```console
sudo apt-get install ros-noetic-mir-robot
```
I followed the instructions given in the webpage, section Gazebo (demo) existing map.

In rviz I set a new goal and after the new route was planned I added an obstacle in Gazebo.

[image]

I have done this on Ubuntu 16.04 with ROS Kinetic and it worked. In the gif it seems to be a narrow corridor where the robot get stuck but this behavior is also observable in a situation with plenty of space around the robot.

Here are some extracts from the console messages of each command. I hope this help.

```shell
roslaunch mir_gazebo mir_maze_world.launch
```

```shell
SUMMARY
========

CLEAR PARAMETERS
 * /ekf_localization_node/

PARAMETERS
 * /ekf_localization_node/acceleration_gains: [0.8, 0.0, 0.0, 0...
 * /ekf_localization_node/acceleration_limits: [1.3, 0.0, 0.0, 0...
 * /ekf_localization_node/base_link_frame: base_footprint
 * /ekf_localization_node/control_config: [True, False, Fal...
 * /ekf_localization_node/control_timeout: 0.2
 * /ekf_localization_node/debug: False
 * /ekf_localization_node/debug_out_file: /path/to/debug/fi...
 * /ekf_localization_node/deceleration_gains: [1.0, 0.0, 0.0, 0...
 * /ekf_localization_node/deceleration_limits: [1.3, 0.0, 0.0, 0...
 * /ekf_localization_node/frequency: 40
 * /ekf_localization_node/imu0: imu_data
 * /ekf_localization_node/imu0_config: [False, False, Fa...
 * /ekf_localization_node/imu0_differential: False
 * /ekf_localization_node/imu0_nodelay: False
 * /ekf_localization_node/imu0_queue_size: 10
 * /ekf_localization_node/imu0_relative: True
 * /ekf_localization_node/imu0_remove_gravitational_acceleration: False
 * /ekf_localization_node/initial_estimate_covariance: [100.0, 0, 0, 0, ...
 * /ekf_localization_node/map_frame: map
 * /ekf_localization_node/odom0: odom_comb
 * /ekf_localization_node/odom0_config: [False, False, Fa...
 * /ekf_localization_node/odom0_differential: False
 * /ekf_localization_node/odom0_nodelay: False
 * /ekf_localization_node/odom0_queue_size: 10
 * /ekf_localization_node/odom0_relative: False
 * /ekf_localization_node/odom_frame: odom_comb
 * /ekf_localization_node/print_diagnostics: True
 * /ekf_localization_node/process_noise_covariance: [0.05, 0, 0, 0, 0...
 * /ekf_localization_node/publish_acceleration: False
 * /ekf_localization_node/publish_tf: True
 * /ekf_localization_node/sensor_timeout: 0.1
 * /ekf_localization_node/stamped_control: False
 * /ekf_localization_node/transform_time_offset: 0.0
 * /ekf_localization_node/transform_timeout: 0.0
 * /ekf_localization_node/two_d_mode: True
 * /ekf_localization_node/use_control: False
 * /ekf_localization_node/world_frame: odom_comb
 * /gazebo/enable_ros_network: True
[... 558 lines cut ...]
```shell
[ INFO] [1603900472.033690625]: rviz version 1.14.1
[ INFO] [1603900472.033731881]: compiled against Qt version 5.12.8
[ INFO] [1603900472.033741718]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1603900472.042244284]: Forcing OpenGl version 0.
[ INFO] [1603900472.501675053, 18.895000000]: Stereo is NOT SUPPORTED
[ INFO] [1603900472.501730403, 18.895000000]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1603900472.834719863, 19.228000000]: Creating 1 swatches
[ INFO] [1603900472.835066535, 19.228000000]: Creating 1 swatches
[ INFO] [1603900472.835212333, 19.228000000]: Creating 1 swatches
[ INFO] [1603900479.630093455, 26.012000000]: Setting goal: Frame:map, Position(4.763, 14.477, 0.000), Orientation(0.000, 0.000, 0.915, 0.404) = Angle: 2.311

[ WARN] [1603900479.819535290, 26.200000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 26.361000 according to authority unknown_publisher
[ WARN] [1603900568.596557960, 114.676000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 114.841000 according to authority unknown_publisher
[ WARN] [1603900613.495799647, 159.450000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 159.640000 according to authority unknown_publisher
[ WARN] [1603900635.236822789, 181.151000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 181.321000 according to authority unknown_publisher
[ WARN] [1603900640.521638531, 186.425000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 186.601000 according to authority unknown_publisher
[ WARN] [1603900662.085014383, 207.950000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 208.121000 according to authority unknown_publisher
[ WARN] [1603900668.572280576, 214.426000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 214.601000 according to authority unknown_publisher
[ WARN] [1603900686.059992710, 231.876000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 232.041000 according to authority unknown_publisher
[ WARN] [1603900691.924988817, 237.725000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 237.881000 according to authority unknown_publisher
```
I hope this helps.

There are a few extra warnings I get with ROS noetic compared to the Ubuntu 16.04/Kinetic system. I cannot interpret them.

mintar, 2020-10-29:
> I have done this on Ubuntu 16.04 with ROS Kinetic and it worked.

Are you sure? From what I can tell, the behavior is no different between Noetic and Kinetic. I'll try to explain what's going on here:

* The global plan is created by SBPL, and the plan is followed by `dwb_local_planner`.
* I've configured `dwb_local_planner` so that it sticks very strictly to the global path. The reason why I did this is that otherwise the "local planner" will try to find shortcuts and subsequently get stuck. For example, SBPL sometimes plans maneuvers like a [Three-point turn](https://en.wikipedia.org/wiki/Three-point_turn), and if we give the local planner too much freedom to deviate from the path, it will try to cut the corners in the three-point turn, and then get stuck (because there was a reason why SBPL planned the path that way).
* The downside of forcing the local planner to stick to the global plan is that now when you place an obstacle right on the path (as you did in the GIF), the local planner cannot deviate from the global plan in order to avoid it. So it just gets stuck and oscillates back and forth a couple of times. If the obstacle doesn't go away, after `oscillation_timeout` (currently [set to 30 seconds](https://github.com/dfki-ric/mir_robot/blob/ddd38930cdebdd57c31833cd2b4989d896e96f94/mir_navigation/config/move_base_common_params.yaml#L11)) has passed, it aborts and replans. With some luck, the new global plan from SBPL now finds a way around the obstacle (although this usually requires another round of replanning once the backside of the obstacle becomes visible).

> In the gif it seems to be a narrow corridor where the robot get stuck but this behavior is also observable in a situation with plenty of space around the robot.

Just to make sure we're on the same page: In the gif, there are some "virtual walls" (forbidden areas) that make the space for the robot smaller than it seems at first. But you're right, since I configured the local planner so that it sticks to the global path very tightly, the same behavior also happens in situations with plenty of space if the obstacle is smack in the middle of the robot's path.

mintar, 2020-10-29:
Of course, this behavior is less than ideal. I've spent a *lot* of time on this, and the current configuration is the best I could come up with. If anyone has a better solution, I'd be happy to accept it. But for me personally it's not a huge problem at the moment, because in our Gazebo simulations there are no dynamic obstacles. And on the physical robot we don't use this move_base configuration, but the one on the actual robot, which deals much better with dynamic obstacles.

ecarrasc, 2020-10-29:
Hi

Thank you very much for giving such an extensive explanation.
I tested it again, I had to wait for the time to elapse and it works. It is good to know that I have to wait more. I think your work is great. I had the chance to work with a real mir100 unit and the behavior avoiding obstacles is very good. I just was wondering why there is this difference to the simulated one. In one of the issues, I read something about MiR using a own planner implementation.

I suppose that the warnings I was receiving have to do more with the ROS version that I am using.

closed 2020-10-29 by ecarrasc