DFKI-NI/mir_robot #73: Spawn multiple MiR100's into maze

rigopeti96, 2020-11-10:
Hello!
I have send an e-mail few days ago, but I'm not sure you got it.

I would like to ask how can I insert multiple MiR100 robots into my world? I found the .launch files with comments but I did not really get it how to spawn more than one robot. Can you please tell me how is it possible?

Thank you so much for the help!
Peter

mintar, 2020-11-26:
I have just pushed some commits to the `melodic` and `kinetic` branches that allow you to spawn multiple robots.

Instructions are here: https://github.com/dfki-ric/mir_robot#gazebo-demo-multiple-robots

I'm closing this issue, but feel free to reopen it if you have more comments or questions.

closed 2020-11-26 by mintar

rigopeti96, 2020-12-08:
Hello @mintar !
Sorry for the late answer. It is almost perfect but there is a small bug which I cannot find. When I start the simulation with one MiR it works perfectly in gazebo. But when I add a second mir i can move only the first. It is spawned but i cannot do anything with the other MiRs. 
Also I got an error message: 
[ERROR] [1607417366.144749656]: No p gain specified for pid.  Namespace: /mir2/gazebo_ros_control/pid_gains/mir2/right_wheel_joint
I forgot to mention I use ROS melodic, Ubuntu 18.04.

Also in connection with this issue: when I want to use them with RViz, I got an error message window. It says the amcl.launch is stopped. That's why I cannot navigate them in RViz. Also when i try to move only one MiR in RViz it does not work as well. 

Can you please explain what can I do with this issue?
Thank you so much!

mintar, 2020-12-18:
About the "No p gain specified for pid." warning: I get that too, that's normal; you can ignore it for now.

About the laser scans: Probably the laser scans are not simulated correctly. Many people have reported this before. It's usually a problem with integrated Intel graphics chips. Please try on a different PC, preferably one with an NVidia graphics card. Any old NVidia card will do.

ahaanairy, 2021-04-12:
@rigopeti96  did you manage to get it working as i am having similar problems. how and where did you add the code for spawing and controlling multiple mirs in maze

rigopeti96, 2021-04-12:
@ahaanairy Sorry but I cannot help you because I was able to insert multiple MiRs but I could not control them. This is surely not the solution what you looking for I guess. But as you can see in description under Gazebo demo (multiple robots) section, you have to insert the code into mir_empty_world.launch file.

To be honest I did not work with the code since 3-4 month and I also not plan it because of multiple reasons. I suggest to write an issue or try to get in touch with the DFKI, I think they could help you much more than I did. May your successes multiply!

mintar, 2021-04-12:
I just tested the code again (in ROS kinetic), and it works. Just make sure that you follow the steps in the "Gazebo demo (multiple robots)" section exactly, including the modifications to `mir_empty_world.launch`.

mintar, 2021-04-12:
@ahaanairy 's problem was solved in #84.