DFKI-NI/mir_robot #76: rviz move_base_simple/goal not working

k9632441, 2020-12-10:
Hello, I am using MiR 100, software version 2.10.0

I found many closed issues about version > 2.8.0

And same issue happens with 2.10.0

I just have some questions.

1. how did you find the type of Mir's Messages and internel process?

2. Do you have any plan to support 2.10.0 software?

3. Did the 2.8.0 software support finished?

let me know about your answer.

If you give me the answer of question 1, I will try what you do with mir.

If you have plan to support software 2.10.0, I will help you to support that one by using our mir 100.

I just tried with melodic-2.8 branch, but it didn't work.

That is all for my questions.

Thanks for your help.

mintar, 2020-12-17:
> 1. how did you find the type of Mir's Messages and internel process?

The easiest way is this:

1. directly connect to the MiR's roscore without using mir_bridge: `export ROS_MASTER_URI=http://192.168.12.20:11311`
2. use [message_thief](https://github.com/uts-magic-lab/message_thief) to retrieve the message definitions

> 2. Do you have any plan to support 2.10.0 software?

Yes. But I only have the MiR update images until version 2.6.0, so I have to rely on the community to provide pull requests and test them, because I cannot do that myself.

> 3. Did the 2.8.0 software support finished?

I believe so. The `melodic-2.8` branch should be working. When I find the time, I plan to put some `if mir_version > 2.3.1` blocks around the changes necessary for 2.8, so it works on all versions. Then I'll merge this back into the master branch.

mintar, 2021-03-05:
I'm closing this due to lack of activity. Feel free to reopen if you have anything to add!

closed 2021-03-05 by mintar