DFKI-NI/mir_robot #77: How to create map with Gmapping

IST-DEV20, 2021-02-02:
move_base.xml error when I use Gmapping. How to fix this problem?

`[ INFO] [1612257815.699039618, 1934.027000000]: Name is SBPLLatticePlanner
terminate called after throwing an instance of 'SBPL_Exception'
  what():  ERROR: failed to read in motion primitive file
[move_base_node-1] process has died [pid 734, exit code -6, cmd /opt/ros/melodic/lib/move_base/move_base map:=/map odom:=/odom marker:=move_base_node/DWBLocalPlanner/markers __name:=move_base_node __log:=/home/its/.ros/log/7500f83e-6533-11eb-a8ca-f8ac65f551a4/move_base_node-1.log].
log file: /home/its/.ros/log/7500f83e-6533-11eb-a8ca-f8ac65f551a4/move_base_node-1*.log
`

thank you for reply

mintar, 2021-02-02:
Probably the resolution of your map and the resolution of your `*.mprim` file doesn't match. See here: https://github.com/ros-planning/navigation_experimental/issues?q=is%3Aissue+failed+to+read+in+motion+primitive+file

mintar, 2021-03-05:
I'm closing this because of lack of activity. Feel free to reopen if you have anything to add!

closed 2021-03-05 by mintar