DFKI-NI/mir_robot #80: MIR Navigation start_planner issues.

klutten99, 2021-03-03:
I get this error when trying to run the `roslaunch mir_navigation start_planner.launch`.
`[FATAL] [1614779970.506633859, 19.521000000]: Failed to create the nav_core_adapter::LocalPlannerAdapter planner, are you sure it is properly registered and that the containing library is built? Exception: Could not find library corresponding to plugin mir_dwb_critics::PathDistPrunedCritic. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[move_base_node-3] process has died [pid 10889, exit code 1, cmd /opt/ros/kinetic/lib/move_base/move_base map:=/map odom:=odom_comb marker:=move_base_node/DWBLocalPlanner/markers __name:=move_base_node __log:=/home/user/.ros/log/8d7433ac-7c28-11eb-9552-0242c0a8d007/move_base_node-3.log].
log file: /home/user/.ros/log/8d7433ac-7c28-11eb-9552-0242c0a8d007/move_base_node-3*.log`

When I look in the files it looks like I am missing some header files aswell not sure if it is just reference errors in my ide or if this is something faulty aswell. I also had to use the catkin_make_isolated instead of the normal catkin_make since it complained about the workspace containing non-catkin packages in it.

CMake Error:
`Base path: /home/user/catkin_ws
Source space: /home/user/catkin_ws/src
Build space: /home/user/catkin_ws/build
Devel space: /home/user/catkin_ws/devel
Install space: /home/user/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/user/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/user/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/user/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/user/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/user/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - mir_robot (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'mir_robot' contains non standard content. Use the content of the following file instead: /home/user/catkin_ws/build/catkin_generated/metapackages/mir_robot/CMakeLists.txt
-- ~~  - mir_msgs
-- ~~  - sdc21x0
-- ~~  - mir_description
-- ~~  - mir_gazebo
-- ~~  - mir_navigation
-- ~~  - mir_actions
-- ~~  - mir_driver
-- ~~  - mir_dwb_critics
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/user/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:318: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed`

mintar, 2021-03-04:
Both errors are weird, and I can't reproduce them. What's your ROS version? Which Git Commit are you on? (`roscd mir_robot && git log | head -n1`)

klutten99, 2021-03-04:
I am on the latest commit also tried the one before that and still had the same issues. 
`user:~$ roscd mir_robot && git log | head -n1commit 9e3ec2cd0102b1140f09ffc765ae5f61b406a59e`
I am using the kinetic version. But when I do "rosversion ros" I get <unversioned> I am currently using ROS through app.theconstructsim.com. Not sure if that would be an issue? Everything else works with the robot and I can control it with the speed controls.

mintar, 2021-03-04:
Can you run `echo $ROS_DISTRO`?

mintar, 2021-03-04:
Also:

```bash
cat /home/user/catkin_ws/build/catkin_generated/metapackages/mir_robot/CMakeLists.txt
roscd mir_robot
cat CMakeLists.txt
```

klutten99, 2021-03-04:
> 
> 
> Can you run `echo $ROS_DISTRO`?
kinetic

cat /home/user/catkin_ws/build/catkin_generated/metapackages/mir_robot/CMakeLists.txt
cmake_minimum_required(VERSION 2.8.3)
project(mir_robot)
find_package(catkin REQUIRED)
catkin_metapackage()

roscd mir_robot
cat CMakeLists.txt
cmake_minimum_required(VERSION 3.5.1)
project(mir_robot)
find_package(catkin REQUIRED)
catkin_metapackage(

mintar, 2021-03-04:
I don't see the problem. The only difference is the cmake_minimum_required, and that should be okay. At least catkin_make doesn't complain on my machine. It looks like this is not a bug in mir_robot, so I'm afraid I can't help you.

klutten99, 2021-03-04:
Could I be that I do not have all the required packages installed? Any easy way to check this?

klutten99, 2021-03-05:
Issue was that I didnt do the Preliminaries so I was missing packages thank you for the help!

closed 2021-03-05 by klutten99