DFKI-NI/mir_robot #83: Loading own map wtih multiple robots

ljaniec, 2021-04-01:
Hi,

This project is a great help in my work, but I am missing maybe something obvious for more experienced ROS/Gazebo users. 

How could I load my own previously prepared in Gazebo map of walls and obstacles (and second one, with virtual obstacles for forbidden areas) for multiple MiR 100? I have a plan of the floor too for it as a PNG black & white image.
Is 

Best regards
Lukasz

mintar, 2021-04-01:
You're talking about the Gazebo simulation, not the real robots, correct? In that case, you can just give the path to the `map.yaml` files to the `map_file` and `virtual_walls_map_file` arguments of `mir_navigation/launch/start_planner.launch`. Look at `mir_gazebo/maps/maze.yaml` for the format.

ljaniec, 2021-04-01:
Yes, I meant in Gazebo simulation. I will try your hints and report with its effects later in an edit.

mintar, 2021-04-01:
Please don't edit, just add a new comment. I don't get email notifications for edits.

closed 2021-05-28 by mintar