DFKI-NI/mir_robot #84: controlling 2 mirs

ahaanairy, 2021-04-06:
I added the code in the empty world file to launch another mir robot into the world. After following the steps to load it in the world the rqt steering opens for both the robots but i can only control the first mir and not the second one.
It does not move.
Maybe i am adding the code incorrectly. Do i need to add additional code in other files or if just the one then what position or line?

My main aim is to programmatically control both the amrs using my code. I have figured this part out for the one mir robot but it brings up lots of problems when trying to control 2. Essentially i am just sending pose goals and the orientation to mir and mir2 move base topics

mintar, 2021-04-12:
Did you follow the steps from the README *exactly*? I just tested everything (on Kinetic), and it works.

ahaanairy, 2021-04-12:
I believe i am following the steps correctly but then again i am confused as to what it requires me to do 
Since you have tested it could you post a screenshot of the changes you made to the empty world file on here?

mintar, 2021-04-12:
Here's the full modified file:

```xml
<?xml version="1.0" ?>
<launch>
  <arg name="gui" default="true" />
  <arg name="world_name" default="worlds/empty.world"/> <!-- Note: the world_name is with respect to GAZEBO_RESOURCE_PATH environmental variable, but can also be an absolute path -->

  <arg name="robot_x"   default="0.0" />
  <arg name="robot_y"   default="0.0" />
  <arg name="robot_yaw" default="0.0" />

  <arg name="tf_prefix" default="" doc="tf_prefix to be used by gazebo plugins and in the robot's urdf etc." />

  <arg name="namespace" default="$(arg tf_prefix)" doc="Namespace to push all topics into."/>

  <group if="$(eval namespace != '')">
    <group>
      <remap from="$(arg namespace)/joint_states"                   to="$(arg namespace)/mir/joint_states" />
      <remap from="$(arg namespace)/mobile_base_controller/cmd_vel" to="$(arg namespace)/cmd_vel" />
      <remap from="$(arg namespace)/mobile_base_controller/odom"    to="$(arg namespace)/odom_comb" />

      <remap from="mir2/joint_states"                   to="mir2/mir/joint_states" />
      <remap from="mir2/mobile_base_controller/cmd_vel" to="mir2/cmd_vel" />
      <remap from="mir2/mobile_base_controller/odom"    to="mir2/odom_comb" />

      <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="world_name" value="$(arg world_name)"/>
        <arg name="paused" value="true" />
        <arg name="gui" value="$(arg gui)" />
      </include>
    </group>

    <!-- This namespace has two effects:
         1. pushing all topics into the namespace
         2. adding the namespace prefix to the header.frame_id field of the laser scanner messages. -->
    <group ns="$(arg namespace)">
      <!-- spawn robot and bring up controllers etc. -->
      <include file="$(find mir_gazebo)/launch/mir_gazebo_common.launch">
        <arg name="robot_x"   value="$(arg robot_x)" />
        <arg name="robot_y"   value="$(arg robot_y)" />
        <arg name="robot_yaw" value="$(arg robot_yaw)" />
        <arg name="tf_prefix" value="$(arg tf_prefix)" />
        <arg name="prepend_prefix_to_laser_frame" value="false" />
      </include>
    </group>
  </group>

  <!-- Duplicate of the above in case namespace is empty. This is necessary to
       avoid the "<group> tag has an empty 'ns' attribute" parsing error. -->
  <group unless="$(eval namespace != '')">
    <group>
      <remap from="joint_states"                   to="mir/joint_states" />
      <remap from="mobile_base_controller/cmd_vel" to="cmd_vel" />
      <remap from="mobile_base_controller/odom"    to="odom_comb" />

      <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="world_name" value="$(arg world_name)"/>
        <arg name="paused" value="true" />
        <arg name="gui" value="$(arg gui)" />
      </include>
    </group>

    <!-- spawn robot and bring up controllers etc. -->
    <include file="$(find mir_gazebo)/launch/mir_gazebo_common.launch">
      <arg name="robot_x"   value="$(arg robot_x)" />
      <arg name="robot_y"   value="$(arg robot_y)" />
      <arg name="robot_yaw" value="$(arg robot_yaw)" />
      <arg name="tf_prefix" value="$(arg tf_prefix)" />
      <arg name="prepend_prefix_to_laser_frame" value="true" />
    </include>
  </group>
</launch>
```

ahaanairy, 2021-04-12:
I used this code and now i can control both the mirs using their individual rqt steering but my aim is to programmatically control the mirs and send them to a specific location.

I have made a cpp  program to do that for the first mir but when i try doing it for the second it keeps waiting for the move base action client to show i.e /mir2/move_base does not turn on i think 

it also displays this error:
Timed out waiting for transform from mir2base_footprint to map to become available before running costmap, tf error: canTransform: source_frame mir2base_footprint does not exist.. canTransform returned after 0.1 timeout was 0.1.

what can i do to make it work?

mintar, 2021-04-12:
You incorrectly started the "start_planner.launch" file with `prefix:=mir2`. It should have been `prefix:=mir2/`, as stated in the README. You have to make sure to copy-paste the instructions from the README *exactly*.

MohammadAlnashef, 2021-04-17:
Hi!
I'm also trying to do multiple robots, but when I went to spawn second MiR into Gazebo it's return "Failure - model name mir already exist." Any hint pls?!

ahaanairy, 2021-04-17:
@MohammadAlnashef  you would just need to copy paste the above code into the mir_empty_world.launch file and run the simulation by copy pasting the commands exactly as there in ReadMe . Be careful to not miss out the "/".

ahaanairy, 2021-04-28:
@mintar 

<group>
   <include file="$(find mir_gazebo)/launch/mir_gazebo_common.launch" ns="mir2">
      <arg name="tf_prefix" value="mir2" />
      <arg name="robot_x" value="-2" />
      <arg name="robot_y" value="-2" />
      <arg name="model_name" value="mir2" />
    </include>

    <include file="$(find mir_navigation)/launch/amcl.launch" >
      <arg name="tf_prefix" value="mir2" />
      <arg name="initial_pose_x" value="8.0" />
      <arg name="initial_pose_y" value="8.0" />
    </include> 

    <include file="$(find mir_navigation)/launch/start_planner_1.launch" >
        <arg name="map_file" value="$(find mir_gazebo)/maps/maze.yaml" />
        <arg name="virtual_walls_map_file" value="$(find mir_gazebo)/maps/maze_virtual_walls.yaml" />
        <arg name="prefix" value="mir2/" />
    </include> 
</group>

will this not work when trying to launch the mir2

I am trying to launch both mirs in one launch file but the latticeplanner on;y recognizes move base for one mir and does not move the other one.

closed 2021-05-28 by mintar