DFKI-NI/mir_robot #85: launching multiple robots using single launch file

ahaanairy, 2021-04-13:
Hi i wish to launch multiple mir robots into the maze world using a single launch file:
I have created a launch file containing the following code : 
<?xml version="1.0" ?>
<launch>

    <include file="$(find mir_gazebo)/launch/mir_maze_world.launch" >
      <arg name="tf_prefix" value="mir" />
    </include>

    <include file="$(find mir_navigation)/launch/amcl.launch" >
      <arg name="tf_prefix" value="mir" />
      <arg name="initial_pose_x" value="10.0" />
      <arg name="initial_pose_y" value="10.0" />
    </include>

    <include file="$(find mir_navigation)/launch/start_planner.launch" ns="mir">
        <arg name="map_file" value="$(find mir_gazebo)/maps/maze.yaml" />
        <arg name="virtual_walls_map_file" value="$(find mir_gazebo)/maps/maze_virtual_walls.yaml" />
        <arg name="prefix" value="mir/" />
    </include>
    <include file="$(find amr_sub)/launch/mir_2.launch" />
</launch>
 
 the mir_2.launch contains the following code 
  <group ns="mir2">
    <include file="$(find mir_gazebo)/launch/mir_gazebo_common.launch" ns="mir2">
      <arg name="tf_prefix" value="mir2" />
      <arg name="robot_x" value="-2" />
      <arg name="robot_y" value="-2" />
      <arg name="model_name" value="mir2" />
    </include>

    <include file="$(find mir_navigation)/launch/amcl.launch" ns="mir2">
      <arg name="tf_prefix" value="mir" />
      <arg name="initial_pose_x" value="8.0" />
      <arg name="initial_pose_y" value="8.0" />
    </include>

    <include file="$(find mir_navigation)/launch/start_planner.launch" ns="mir2">
        <arg name="prefix" value="mir2/" />
        <arg name="map_file" value="$(find mir_gazebo)/maps/maze.yaml" />
        <arg name="virtual_walls_map_file" value="$(find mir_gazebo)/maps/maze_virtual_walls.yaml" />
    </include>
  </group>

</launch>

when i run the mir.launch which runs the first code it gives me an error : 

roslaunch file contains multiple nodes named [/static_map_server].
Please check all <node> 'name' attributes to make sure they are unique.
 
how can i fix this error

mintar, 2021-04-13:
This is a general ROS question not specific to this repo. Please ask on https://answers.ros.org/ .

closed 2021-04-13 by mintar

ahaanairy, 2021-04-17:
would you be able to give some advice on how to prevent the multiple nodes from happening. Upon looking at maps.xml,  I see the static map server node which is causing problems. The ns argument is just a "/" . Would it help if i pass the mir name onto that namespace argument.