DFKI-NI/mir_robot #88: Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb

fjp, 2021-05-20:
When running the [Gazebo demo (existing map)](https://github.com/dfki-ric/mir_robot#gazebo-demo-existing-map) a lot of the following warnings are logged:

```
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 23.567000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
```

I guess this is related to https://github.com/ros/geometry2/issues/467. To reduce the number of `TF_REPEATED_DATA` warnings the laser scan publish rate needs to be reduced, because amcl depends on rate of incoming laser messages.

Can we ignore this warning? And do you know if this is a problem with amcl itself or is the laser scan publish rate set to high?

mintar, 2021-05-28:
> Can we ignore this warning? And do you know if this is a problem with amcl itself or is the laser scan publish rate set to high?

This is a bug in AMCL. The laser scan publish rate is ok. The warning can be ignored; AMCL is working just as it was in previous ROS versions. The only difference is that in Noetic, the warning is now shown. The bug was there all the time. We should wait for it to be fixed in AMCL.

Also, can you update all ROS packages, recompile and retry? I think they changed something; at least for me, the warning is only displayed once.

fjp, 2021-05-29:
Hi @mintar thanks for the explanation. 

> Also, can you update all ROS packages, recompile and retry? I think they changed something; at least for me, the warning is only displayed once.

I updated all my ROS Noetic (1.15.11) packages (`sudo apt update` and `sudo apt upgrade`). Then recompiled a fresh clone of `mir_robot` in a new catkin workspace but the warning messages are still quite frequently displayed:

```
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 32.281000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 33.720000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 57.881000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 80.281000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 102.041000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 105.001000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
``` 

I then cloned https://github.com/ros-planning/navigation (noetic-devel 2b807bd), which includes the amcl package, into the same catkin workspace. This seems to reduced printing the warning but doesn't get rid of it unfortunately:

```
[ INFO] [1622276763.352941573, 20.438000000]: Initializing likelihood field model; this can take some time on large maps...
[ INFO] [1622276763.445606552, 20.529000000]: Done initializing likelihood field model.
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 67.081000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 135.800000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 158.601000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 181.164000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 188.684000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 191.644000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 205.404000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 303.164000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 306.124000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 352.444000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom_comb at time 369.324000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
```

Do you know which package I have to build to get the message displayed only once?

mintar, 2021-05-29:
> Do you know which package I have to build to get the message displayed only once?

No, unfortunately not. The messages seem to appear at irregular intervals, so it looks like it's hard to debug (might be a race condition, which again depends on the speed of your computer vs. mine etc.). By the way, I didn't build the navigation package from source, only `mir_robot`.

I'd say let's ignore this and wait forit to be fixed in `amcl` (if that ever happens). Or if you're feeling adventurous, try to fix the `amcl` bug yourself. Basically, once `amcl` has published a TF for a given time stamp `t`, it must never publish it again. So there are two options: Either publish the first message and then drop all subsequent ones, or delay publishing the message until `amcl` is certain that this is the definitive one. Both have their advantages and disadvantages. I'm surprised that there's no issue about this in the `navigation` repository already.

Since this issue is not a bug in `mir_robot`, I suggest that you open an issue for `amcl` instead.

fjp, 2021-05-29:
Thank you for the hints. I will open the issue in the navigation package/`amcl`.

closed 2021-05-29 by fjp