DFKI-NI/mir_robot #91: Orientation missing in Path while using driver / reverse trajectories unable to execute in simulation

rodubero, 2021-05-27:
Hi,

First of all, thanks a lot for your public contribution. Even our MIR-100 is a great model to try interesting stuff, MIR does not provide any support for ROS to one of the most common clients: Universities and research institutions. The MIR support team shamelessly answer _" Sorry to disappoint you, but what you asked is not possible to have in the rest API"_ and that's it. Without your project, probably our MIR-100 would be used only to bring us drinks, not in research.

Currently, I am working on using the data available in the robot for interfacing purposes. I am trying to use the path plan published by the topic **/move_base_node/SBPLLatticePlanner/plan** to enhance the real-time visualization. Unfortunately, the topic while connected as a driver does not provide any data in the orientation key ([0,0,0,0] always). I am not sure where to check what is missing, because, in simulation, the SBPL does provide z and w in the orientation quaternion. This is the first issue.

To overcome the first issue I am trying to simulate the path while listening to goals to obtain the orientation from that, but I noticed while simulating the robot is unable to execute paths that require reverse displacements (i.e. undocking from charging to reach another place), this is the second issue. Again, here I am unable to solve the problem because I am not sure where exactly to check and maybe add some extra planning process for these cases which require short reverse trajectories.

Our robot has the software v2.8.3.1

Any light will be highly appreciated. 

Thanks again.

Issue 1:
[image]

Issue 2:
[image]

mintar, 2021-05-28:
>  I am not sure where to check what is missing, because, in simulation, the SBPL does provide z and w in the orientation quaternion. This is the first issue.

The MiR runs heavily customized versions of SBPL and move_base. For some unknown reason, it is not publishing the orientation. There is nothing you or I can do about it.

> To overcome the first issue I am trying to simulate the path while listening to goals to obtain the orientation from that,

I don't think this is going to work. As I said above, the motion planning on the MiR is heavily customized. I tried to reproduce it as faithfully as possible in simulation, but there will be large differences in the produced paths, so you can't just copy the orientations over.

> but I noticed while simulating the robot is unable to execute paths that require reverse displacements (i.e. undocking from charging to reach another place), this is the second issue.

No, reverse motions are working fine in simulation. The robot strongly prefers forward motions (because the floor 3D camera for obstacle avoidance is mounted in front), but if it absolutely requires a backward motion, it will do it.

The problem in your screenshot is that the start pose of the robot is "in collision" with the dock. If it was in the same position but rotated 180° (so that a forward motion is required) it would also not be able to generate a plan. You need to first steer the robot backward a bit by sending commands to the `/cmd_vel` topic until it is out of collision. Then you can use `move_base`.

mintar, 2021-06-11:
I'm closing this issue due to inactivity. Feel free to reopen if you want to add something.

closed 2021-06-11 by mintar