DFKI-NI/mir_robot #94: Modifying move_base on internal MiR PC

mintar, 2021-09-30:
Hi @mintar. I have a MiR250 that I am trying to control through ROS (noetic). You mentioned to have accessed the move_base inside the mir computer? How is that done? Also, is there any way of setting the robot to ignore the SICK configurations and drive close to walls?

_Originally posted by @ravescovi in https://github.com/dfki-ric/mir_robot/issues/81#issuecomment-930566754_

mintar, 2021-09-30:
Hi @ravescovi,

> You mentioned to have accessed the move_base inside the mir computer? How is that done?

To modify the parameters of the internal move_base, I did the following:

1. Remove the black plastic cover on one of the MiR's corners, which will expose one HDMI and two USB ports. Plug in a monitor, keyboard and USB stick with a live Ubuntu system and boot into the live system.
2. Use `chroot` to add a user with sudo privileges to the internal MiR PC.
3. Reboot into the normal system, unplug everything.
4. SSH into the internal PC.
5. The parameters of move_base are stored inside a MySQL database. You can change those and reboot to take effect.

**However,** unless you are doing something very special and want to change one specific parameter, I would not recommend doing this. The parameters are extremely well-tuned, and messing around with them can have unforeseen consequences. At least make a backup before you change anything.

A better alternative would be not to use the move_base on the robot at all, but instead run your own instance of move_base on an external PC and only send `cmd_vel` commands to the robot. You can use (or modify) the move_base config from my package `mir_navigation`. The advantage is that you have full control; the disadvantage is that you miss some of the custom improvements of the internal move_base (faster planning, better path following). But I think I'd try this first.

> Also, is there any way of setting the robot to ignore the SICK configurations and drive close to walls?

There are two parts to this:

1. The internal move_base will not drive extremely close to walls, and there's no way to change this without causing collisions all the time. The parameters of the internal move_base are already tuned pretty perfectly. If you want to implement for example a docking behavior (where you actually want to "collide" with the charging station), you shouldn't use move_base for this. Just use move_base to drive in front of the charging station, and then write your own node that sends `cmd_vel` commands for the final approach.
2. The MiR has an actual safety system implemented on a Programmable Logic Controller (PLC), which reads the safety fields of the laser scanners and the speed from the wheel encoders. If the safety field is triggered, the PLC will stop the robot, no matter what `cmd_vel` commands you send (directly or via `move_base`). You cannot switch off this safety feature (which is good). The size of the safety field depends on the speed of the robot. If you drive very slowly, the safety field is very small, so you can drive almost into contact with a wall if you want.

closed 2022-03-02 by mintar

dimadobriy, 2023-11-04:
Hi there!

Could you please describe how to run own instance of move_base?

Regards

mintar, 2023-11-06:
I've described it all in my comment above. Do you have any specific questions?