DFKI-NI/mir_robot #101: robot_state_publisher hz

narutojxl, 2021-12-23:
Hi author,
When i launch mir_robot, i see the robot_state_publisher hz is very big, e.g.  10000.  
[image]

Adding `publish_frequency` does not take affect according to [ros wiki robot_state_publisher](http://wiki.ros.org/robot_state_publisher), so how to do to modify the published static tf  frequency, thanks for your help and time!
>
	  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" >
		<param name ="publish_frequency" value="50" />
	  </node>

mintar, 2021-12-23:
Don't worry. When rqt_tf_tree shows a publishing frequency of 10000, what that really means is that this is a static transform that is published on the tf_static topic (not the normal tf topic). Static transforms are only published once.You can check with "rostopic hz tf" that it's not really 10000 Hz.

narutojxl, 2021-12-23:
Thanks author very fast reply, `rostopic hz /tf` is about 32 hz when `publish_frequency = 50`,  and `rostopic hz /tf`  is 25 hz when  when `publish_frequency = 30`. Maybe because some other system load, the actual frequency is not the expected frequency. Thanks again.

closed 2021-12-23 by narutojxl