DFKI-NI/mir_robot #102: Switching modes

fdumaire, 2022-01-14:
Hi,
I am currently working with a MIR 250. I want to use the MIR in a teleoperative state, where I can steer it with the keyboard of my laptop. So far most of the functions are working, but I got a problem. When I put the MIR into manual mode and don't give an input after ca. 5sec. it leaves the manual mode and goes back into the automatic mode. Is there a workaround for my problem or an option how I can increase the duration in until it switches modes.

Appreciate the help, great work from you guys.

mintar, 2022-01-14:
Hi,

that seems to be an intentional feature of the web interface that was added in one of the later versions of the MiR software. I haven't found a way to turn it off, sorry. Please note that this repo is a community project to use the MiR Robots with ROS. It is not affiliated with Mobile Industrial Robots. Perhaps you could ask Mobile Industrial Robots support about it?

mclutz, 2022-05-13:
> Hi, I am currently working with a MIR 250. I want to use the MIR in a teleoperative state, where I can steer it with the keyboard of my laptop. So far most of the functions are working, but I got a [...]

Hi fdumaire, we have the same behavior on our MIR 250, have you found a way to solve the problem or work around it?  Thanks!