DFKI-NI/mir_robot #105: Spawning MiR Robot into AWS Small Warehouse World

rllyryan, 2022-02-28:
Dear to whom it may concern,

I am having problems trying to spawn the MiR robot into the small warehouse world provided by AWS. For starters, I don't really have a good idea on how to modify the launch files so that the corresponding world file is used.

Here is the link to the repository:
https://github.com/aws-robotics/aws-robomaker-small-warehouse-world 

Could I please get some advice/help on how to go about this? (For example, which files to edit, what lines to add, etc)

Thank you :)

mintar, 2022-03-02:
First, clone `aws_robomaker_small_warehouse_world` into your catkin workspace, build + source:

```bash
cd <your-catkin-ws>/src
git clone -b ros1 https://github.com/aws-robotics/aws-robomaker-small-warehouse-world.git aws_robomaker_small_warehouse_world
cd ..
catkin b
source devel/setup.bash
```

Now you can run the demos from the README like this:

### Gazebo demo (existing map)

```bash
roslaunch mir_empty_world.launch world_name:=$(rospack find aws_robomaker_small_warehouse_world)/worlds/no_roof_small_warehouse.world
rosservice call /gazebo/unpause_physics   # or click the "start" button in the Gazebo GUI
roslaunch mir_navigation amcl.launch
roslaunch mir_navigation start_planner.launch \
    map_file:=$(rospack find aws_robomaker_small_warehouse_world)/maps/005/map.yaml \
    with_virtual_walls:=false
rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz
```

### Gazebo demo (mapping)

```bash
roslaunch mir_empty_world.launch world_name:=$(rospack find aws_robomaker_small_warehouse_world)/worlds/no_roof_small_warehouse.world
rosservice call /gazebo/unpause_physics   # or click the "start" button in the Gazebo GUI
roslaunch mir_navigation hector_mapping.launch
roslaunch mir_navigation move_base.xml with_virtual_walls:=false
rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz
```

closed 2022-03-02 by mintar

rllyryan, 2022-03-02:
@mintar thank you so much for your assistance! Have a great week ahead!