DFKI-NI/mir_robot #108: Any idea how to convert MiR *.site map file into ROS map file?

Hubert51, 2022-04-12:
In MiR web interface, I can create a map describe my office. Then I exported the map and got a *.site file. Can we open it as human readable file or it is just MiR internal file type? 

Thanks!

mintar, 2022-04-13:
I think the `*.site` files are only MiR internal, and they are GPG encrypted, so there's not much we can do with them (although I could probably figure out the GPG key/passphrase). However, there's the option to export the map as a PNG somewhere in the web interface, and that's all I needed so far. I think the site file also includes the positions of charging stations and so on, but personally I haven't needed them.

closed 2022-05-10 by mintar

mintar, 2022-05-12:
Update: The `*.site` file is a GPG-encrypted file that contains a ZIP file, which contains a `session.json` file. That file contains all the maps, missions, forbidden zones and so on. The maps are base64-encoded PNG files.

Here is how you extract the `session.json` file in the terminal:

```bash
echo "MiRArea" | gpg --homedir /tmp --batch --no-tty --yes --passphrase-fd 0 -o mysite.zip -d mysite.site
unzip mysite.zip
```

And here is some example python code on how to extract the first map and some metadata of the first map in the file. The json file contains much more info (other maps, missions), so this is just an example to get you started.

```python
import json
import base64

with open('session.json', 'r') as json_file:
    session = json.load(json_file)

first_map = session['maps'][0]

with open('map.png', 'wb') as map_file:
    map_file.write(base64.b64decode(first_map['map']))

metadata = json.loads(base64.b64decode(first_map['metadata']))
with open('first_map_walls.png', 'wb') as base_map_walls_file:
    base_map_walls_file.write(base64.b64decode(metadata['base_map_walls']))

with open('first_map_metadata.json', 'w') as json_file:
    json.dump(metadata, json_file, indent=4, sort_keys=True)
```