DFKI-NI/mir_robot #119: General problems / missing details?

lliendo, 2022-12-23:
Hi,

I'm trying to run the Gazebo simulator but there are things that are not running as expected.

* First issue. The ROS `ros-noetic-rosbridge-server` package seems to be missing although the README.md states that installing the `ros-noetic-mir-robot` package is enough to get up and running. Here's a copy of the error that I'm getting when launching `roslaunch mir_driver mir.launch`:

```
SUMMARY
========

PARAMETERS
 * /mir_bridge/hostname: 192.168.12.20
 * /mir_bridge/tf_prefix: 
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /tf_prefix: 
 * /tf_remove_state_publisher_frames/remove_frames: ['base_link', 'fr...

NODES
  /
    b_rep117_laser_filter (mir_driver/rep117_filter.py)
    f_rep117_laser_filter (mir_driver/rep117_filter.py)
    fake_mir_joint_publisher (mir_driver/fake_mir_joint_publisher.py)
    mir_bridge (mir_driver/mir_bridge.py)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    tf_remove_state_publisher_frames (mir_driver/tf_remove_child_frames.py)

auto-starting new master
process[master]: started with pid [92]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cc06cf6a-82ae-11ed-9454-3c58c2b35ab0
process[rosout-1]: started with pid [102]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [109]
process[tf_remove_state_publisher_frames-3]: started with pid [110]
process[mir_bridge-4]: started with pid [111]
process[b_rep117_laser_filter-5]: started with pid [112]
process[f_rep117_laser_filter-6]: started with pid [117]
process[fake_mir_joint_publisher-7]: started with pid [119]
[INFO] [1671792279.095902]: [/mir_bridge] trying to connect to 192.168.12.20:9090...
[WARN] [1671792279.998512]: [/mir_bridge] still waiting for connection to 192.168.12.20:9090...
[WARN] [1671792280.998691]: [/mir_bridge] still waiting for connection to 192.168.12.20:9090...
[WARN] [1671792281.998505]: [/mir_bridge] still waiting for connection to 192.168.12.20:9090...
[WARN] [1671792282.998397]: [/mir_bridge] still waiting for connection to 192.168.12.20:9090...
[WARN] [1671792283.998458]: [/mir_bridge] still waiting for connection to 192.168.12.20:9090...
[WARN] [1671792284.998649]: [/mir_bridge] still waiting for connection to 192.168.12.20:9090...
```
It  seems that first I need to run: `roslaunch rosbridge_server rosbridge_websocket.launch` after doing `apt-get install ros-noetic-rosbridge-server`.

* Second issue. The above IP (192.168.12.20) is hardcoded in `/opt/ros/noetic/share/mir_driver/launch/mir.launch`. I don't understand why that particular IP has been chosen, but in the docs that is not mentioned. I'm assuming that after I've launched the `rosbridge_server` (which by default is listening on 0.0.0.0:9090) I need to update this configuration to be: `<arg name="mir_hostname" default="127.0.0.1" />` but not 100% sure about this. However I can succesfully launch both processes so far.

* Third issue. The console that is running the `roslaunch mir_driver mir.launch` is being flooded at a very high rate (actually my CPU load skyrockets after launching it) with these messages:

```
[INFO] [1671794925.991974]: [/mir_bridge] tf_static: sent 4 transforms: odict_keys(['/surface', '/imu_frame', '/us_1_frame', '/us_2_frame'])
[INFO] [1671794925.994877]: [/mir_bridge] tf_static: updated transform base_link->/surface.
[INFO] [1671794925.995985]: [/mir_bridge] tf_static: updated transform imu_link->/imu_frame.
[INFO] [1671794925.997328]: [/mir_bridge] tf_static: updated transform base_link->/us_1_frame.
[INFO] [1671794925.998473]: [/mir_bridge] tf_static: updated transform base_link->/us_2_frame.
[INFO] [1671794925.999635]: [/mir_bridge] tf_static: sent 4 transforms: odict_keys(['/surface', '/imu_frame', '/us_1_frame', '/us_2_frame'])
[INFO] [1671794926.002193]: [/mir_bridge] tf_static: updated transform base_link->/surface.
[INFO] [1671794926.003573]: [/mir_bridge] tf_static: updated transform imu_link->/imu_frame.
[INFO] [1671794926.005487]: [/mir_bridge] tf_static: updated transform base_link->/us_1_frame.
[INFO] [1671794926.007061]: [/mir_bridge] tf_static: updated transform base_link->/us_2_frame.
[INFO] [1671794926.009957]: [/mir_bridge] starting to stream messages on topic 'move_base/status'
[INFO] [1671794926.010236]: [/mir_bridge] tf_static: sent 4 transforms: odict_keys(['/surface', '/imu_frame', '/us_1_frame', '/us_2_frame'])
[INFO] [1671794926.015737]: [/mir_bridge] starting to stream messages on topic 'move_base/result'
[INFO] [1671794926.017013]: [/mir_bridge] tf_static: updated transform base_link->/surface.
[... 151 lines cut ...]
[ INFO] [1671795490.348355294]: Physics dynamic reconfigure ready.
[spawn_urdf-3] process has finished cleanly
log file: /home/lucas/.ros/log/f1597fdc-82b4-11ed-9fcc-3c58c2b35ab0/spawn_urdf-3*.log
```

Note the:
```
[ WARN] [1671795485.266056747]: Shutdown request received.
[ WARN] [1671795485.272054087]: Reason given for shutdown: [[/robot_state_publisher] Reason: new node registered with same name]
```
messages. I don't know if that is correct/incorrect or expected behaviour. Besides that the robot is not moving in the console where I'm running the `rosbridge_server` process I get these errors:

```
2022-12-23 11:38:08+0000 [-] [ERROR] [1671795488.502394]: [Client 1] publish: Message type geometry_msgs/Twist does not have a field header
2022-12-23 11:40:27+0000 [-] [ERROR] [1671795627.972498]: [Client 1] publish: Message type geometry_msgs/Twist does not have a field header
2022-12-23 11:40:27+0000 [-] [ERROR] [1671795627.990643]: [Client 1] publish: Message type geometry_msgs/Twist does not have a field header
2022-12-23 11:40:28+0000 [-] [ERROR] [1671795628.091006]: [Client 1] publish: Message type geometry_msgs/Twist does not have a field header
2022-12-23 11:40:28+0000 [-] [ERROR] [1671795628.191302]: [Client 1] publish: Message type geometry_msgs/Twist does not have a field header
2022-12-23 11:40:28+0000 [-] [ERROR] [1671795628.291005]: [Client 1] publish: Message type geometry_msgs/Twist does not have a field header
```

At this stage it's clear that either I'm doing something wrong, or some steps from the documentation are missing, or both or something else is going on... Unfortunately I'm not that experienced with ROS so is not easy to spot what the issue could be. As previously stated I'm running `roslaunch rosbridge_server rosbridge_websocket.launch` and `roslaunch mir_driver mir.launch` in a docker `noetic` container (the docker container is launched with the `--network host` option so I can communicate with those services from my regular environment) and the Gazebo environment is run outside docker. Any help is much appreciated.

Thanks,
Lucas.

mintar, 2022-12-23:
Please *only* run the commands in the "Gazebo demo (existing map)" section of the README. You are trying to run commands from the mir_driver package in simulation, but that package is only for the real robot.

mintar, 2022-12-23:
By the way, the README is very well tested, so if you feel that you have to deviate from it, chances are that you're doing something wrong. :)

mintar, 2022-12-23:
And if you find that the instructions are unclear (such as in this case), feel free to improve them and send a pull request, or tell me what was unclear so I can improve it. Feedback from actual users is always very valuable!