DFKI-NI/mir_robot #129: Goal to MiR doesn't work

arielc-brillianetor, 2023-06-12:
Hi,
I was able to connect to both MiR100 and MiR250 using the mir_robot package. When listening to the different topics I received the correct information, and publishing I was able to move both robots using the cmd_vel topic. But when I tried to send a move to position goal it didn't work: "move_base_simple/goal" subscriber didn't do anything.
If this happened to anyone else I would really appreciate the help,
Thanks a lot.

mintar, 2023-06-13:
This sounds like it might be related to #62. Which version of the MiR software are you running? For reference, we are running 2.13.4.1 (I believe), and "move_base_simple/goal" should be working I believe (but I haven't tested it in a long time).

arielc-brillianetor, 2023-06-13:
I'm running different versions depending on the robot:
MIR100 - 2.13.0.2
MIR250 - 2.13.4.1
I was using the correction made in #62 and I tried to publish both the topic as well as the action from the command line and that didn't work wither.
Thanks for helping out!

mintar, 2023-06-13:
What do you mean, you tried the correction made in #62? That's already merged, so no need for you to do anything.

Which version of this repo are you using? The one from the binary packages, or from source (and if from source, which git commit and did you make any changes)?

arielc-brillianetor, 2023-06-13:
I didn't make any changes. Just pointed out that I was using the corrected version.
I tried the binary packages on ubuntu 18.

mintar, 2023-06-13:
Aaah, so that's your problem then. Ubuntu 18.04 still has ROS melodic, and the latest release into melodic (1.0.8) was made almost 2 years ago, on 2021-06-11.

It would be best if you could instead run Ubuntu 20.04 with ROS noetic so you get the newest version.

Alternatively you could try to run the `noetic` branch from source on ROS melodic, but that's going to need some adjustments because melodic is still Python 2, whereas noetic is Python 3.

arielc-brillianetor, 2023-06-13:
Oh, good to know. Thanks' i'll try it out and let you know

mintar, 2023-06-14:
Just FYI: I've just confirmed that the "move_base_simple/goal" (published from RViz) works with our MiR100 robot, software version 2.13.5.3, mir_robot branch `noetic` (1d7f6c2).

arielc-brillianetor, 2023-06-14:
ok thanks. I'll update my ubuntu and ROS and let you know.

DanyResasco, 2023-06-20:
I have the same problem. I can't move the robot with /cmd_vel or /move_base. 
I have Ubunto 20 and the robot is mir250 with software version 2.14.

mintar, 2023-06-21:
@DanyResasco : If you can send me the MiR software upgrade image to 2.14, I can have a look.

DanyResasco, 2023-06-21:
Forgive my ignorance but How can I get the image?

mintar, 2023-06-21:
If you upgraded the MiR software yourself, you should have a file called something like `MiR_Software-2.13.3.mir`. This is the one I need. If your MiR already came preinstalled with this version (or a technician installed it), then you probably don't have this file.

DanyResasco, 2023-06-21:
Unfortunately the Mir came directly with the new software.

DanyResasco, 2023-06-21:
@mintar also this new version, the manual mode get disabled if u don't move the robot for few seconds. 
The webpage and some messages/actions are different from your package

mintar, 2023-06-21:
Are you sure you are using the most up to date noetic version of this package? There were some changes to the cmd_vel topic years ago that we had to adapt to (https://github.com/DFKI-NI/mir_robot/issues/45#issuecomment-651892529), but I would be surprised if they changed it again.

DanyResasco, 2023-06-21:
I installed using **sudo apt install ros-noetic-mir-robot** . I'm not sure how can I move it via ros if the manual mode gets disconnected all the time. 
For now, I changed my ROS_MASTER_URI to the MIR's IP and try to see the difference with your package, one example is if I use your package I don't have data from sensors, while with the ROS_MASTER_URI I can listen to the topics.

mintar, 2023-06-21:
> I installed using sudo apt install ros-noetic-mir-robot.

Ok, that's good then.

> I'm not sure how can I move it via ros if the manual mode gets disconnected all the time.

You can still move it via ROS even if the manual mode gets disconnected, that doesn't matter.

> For now, I changed my ROS_MASTER_URI to the MIR's IP and try to see the difference with your package, one example is if I use your package I don't have data from sensors, while with the ROS_MASTER_URI I can listen to the topics.

Ok, then the first step would be getting data from the sensors. It sounds like my package didn't connect properly to the MiR at all.

What's your MiR IP? If it's not the default (`192.168.12.20`), then you have to adjust `mir_hostname` accordingly (`roslaunch mir_driver mir.launch mir_hostname:=192.168.xxx.xxx`).

What's being printed in the console?

DanyResasco, 2023-06-21:
if I export the ROS_MASTER_URI=http:/192.168.12.20:11311 I can get the sensor data (rostopic echo) and display them in rviz. 
My problem right now is that the robot doesn't move. 
I tried with the command line **rostopic pub /cmd_vel** . I can see the msg is modified with the value I publish but the robot stays still.

DanyResasco, 2023-06-21:
@mintar wiht roslaunch mir_driver mir.launch I can see ROS_BRIDGE CONNECTED then the logs regarding which msgs is subscribing to (some are warnings because they changed the name)

DanyResasco, 2023-06-21:
@mintar my bad, I have the 2.13.5.3 MIR250 

`
[INFO] [1687369617.386966]: [/mir_bridge] publishing topic 'LightCtrl/us_list' [sensor_msgs/Range]
[WARN] [1687369617.387829]: [/mir_bridge] topic 'LightCtrl/us_list' is not published by the MiR!
[INFO] [1687369617.389313]: [/mir_bridge] publishing topic 'MC/currents' [sdc21x0/MotorCurrents]
[WARN] [1687369617.390120]: [/mir_bridge] topic 'MC/currents' is not published by the MiR!
[INFO] [1687369617.391713]: [/mir_bridge] publishing topic 'MissionController/CheckArea/visualization_marker' [visualization_msgs/Marker]
[INFO] [1687369617.393524]: [/mir_bridge] publishing topic 'SickPLC/parameter_descriptions' [dynamic_reconfigure/ConfigDescription]
[WARN] [1687369617.394519]: [/mir_bridge] topic 'SickPLC/parameter_descriptions' is not published by the MiR!
[INFO] [1687369617.396428]: [/mir_bridge] publishing topic 'SickPLC/parameter_updates' [dynamic_reconfigure/Config]
[WARN] [1687369617.397461]: [/mir_bridge] topic 'SickPLC/parameter_updates' is not published by the MiR!
[INFO] [1687369617.399200]: [/mir_bridge] publishing topic 'amcl_pose' [geometry_msgs/PoseWithCovarianceStamped]
[INFO] [1687369617.400791]: [/mir_bridge] publishing topic 'b_raw_scan' [sensor_msgs/LaserScan]
[INFO] [1687369617.402202]: [/mir_bridge] publishing topic 'b_scan' [sensor_msgs/LaserScan]
[INFO] [1687369617.403654]: [/mir_bridge] publishing topic 'camera_floor/driver/color/camera_info' [sensor_msgs/CameraInfo]
[WARN] [1687369617.404430]: [/mir_bridge] topic 'camera_floor/driver/color/camera_info' is not published by the MiR!
[INFO] [1687369617.406128]: [/mir_bridge] publishing topic 'camera_floor/background' [sensor_msgs/PointCloud2]
[WARN] [1687369617.406973]: [/mir_bridge] topic 'camera_floor/background' is not published by the MiR!
[INFO] [1687369617.408528]: [/mir_bridge] publishing topic 'camera_floor/depth/parameter_descriptions' [dynamic_reconfigure/ConfigDescription]
[WARN] [1687369617.409743]: [/mir_bridge] topic 'camera_floor/depth/parameter_descriptions' is not published by the MiR!
[INFO] [1687369617.411686]: [/mir_bridge] publishing topic 'camera_floor/depth/parameter_updates' [dynamic_reconfigure/Config]
[WARN] [1687369617.412913]: [/mir_bridge] topic 'camera_floor/depth/parameter_updates' is not published by the MiR!
[INFO] [1687369617.414906]: [/mir_bridge] publishing topic 'camera_floor/depth/points' [sensor_msgs/PointCloud2]
[WARN] [1687369617.415768]: [/mir_bridge] topic 'camera_floor/depth/points' is not published by the MiR!
[INFO] [1687369617.417264]: [/mir_bridge] publishing topic 'camera_floor/filter/visualization_marker' [visualization_msgs/Marker]
[WARN] [1687369617.418110]: [/mir_bridge] topic 'camera_floor/filter/visualization_marker' is not published by the MiR!
[INFO] [1687369617.419652]: [/mir_bridge] publishing topic 'camera_floor/floor' [sensor_msgs/PointCloud2]
[WARN] [1687369617.420488]: [/mir_bridge] topic 'camera_floor/floor' is not published by the MiR!
[INFO] [1687369617.421962]: [/mir_bridge] publishing topic 'camera_floor/obstacles' [sensor_msgs/PointCloud2]
[WARN] [1687369617.422870]: [/mir_bridge] topic 'camera_floor/obstacles' is not published by the MiR!
[INFO] [1687369617.424387]: [/mir_bridge] publishing topic 'check_area/polygon' [geometry_msgs/PolygonStamped]
[INFO] [1687369617.425930]: [/mir_bridge] publishing topic 'diagnostics' [diagnostic_msgs/DiagnosticArray]
[INFO] [1687369617.427488]: [/mir_bridge] publishing topic 'diagnostics_agg' [diagnostic_msgs/DiagnosticArray]
[INFO] [1687369617.429086]: [/mir_bridge] publishing topic 'diagnostics_toplevel_state' [diagnostic_msgs/DiagnosticStatus]
[INFO] [1687369617.430763]: [/mir_bridge] publishing topic 'f_raw_scan' [sensor_msgs/LaserScan]
[INFO] [1687369617.432406]: [/mir_bridge] publishing topic 'f_scan' [sensor_msgs/LaserScan]
[INFO] [1687369617.433915]: [/mir_bridge] publishing topic 'imu_data' [sensor_msgs/Imu]
[INFO] [1687369617.435397]: [/mir_bridge] publishing topic 'laser_back/driver/parameter_descriptions' [dynamic_reconfigure/ConfigDescription]
[WARN] [1687369617.436228]: [/mir_bridge] topic 'laser_back/driver/parameter_descriptions' is not published by the MiR!
[INFO] [1687369617.437751]: [/mir_bridge] publishing topic 'laser_back/driver/parameter_updates' [dynamic_reconfigure/Config]
[WARN] [1687369617.438630]: [/mir_bridge] topic 'laser_back/driver/parameter_updates' is not published by the MiR!
[INFO] [1687369617.440196]: [/mir_bridge] publishing topic 'laser_front/driver/parameter_descriptions' [dynamic_reconfigure/ConfigDescription]
[WARN] [1687369617.441082]: [/mir_bridge] topic 'laser_front/driver/parameter_descriptions' is not published by the MiR!
[INFO] [1687369617.442855]: [/mir_bridge] publishing topic 'laser_front/driver/parameter_updates' [dynamic_reconfigure/Config]
[WARN] [1687369617.443758]: [/mir_bridge] topic 'laser_front/driver/parameter_updates' is not published by the MiR!
[INFO] [1687369617.445737]: [/mir_bridge] publishing topic '/map' [nav_msgs/OccupancyGrid]
[INFO] [1687369617.447116]: [/mir_bridge] starting to stream messages on topic '/map'
[INFO] [1687369617.449050]: [/mir_bridge] publishing topic '/map_metadata' [nav_msgs/MapMetaData]
[INFO] [1687369617.450707]: [/mir_bridge] publishing topic 'mir_amcl/parameter_descriptions' [dynamic_reconfigure/ConfigDescription]
[WARN] [1687369617.451560]: [/mir_bridge] topic 'mir_amcl/parameter_descriptions' is not published by the MiR!
[... cut ...]
[INFO] [1687369617.614865]: [/mir_bridge] subscribing to topic 'move_base/cancel' [actionlib_msgs/GoalID]
[INFO] [1687369617.619834]: [/mir_bridge] subscribing to topic 'move_base/goal' [move_base_msgs/MoveBaseActionGoal]
[INFO] [1687369617.651313]: [/mir_bridge] starting to stream messages on topic 'move_base/status'
[INFO] [1687369617.657916]: [/mir_bridge] starting to stream messages on topic 'move_base/result'
[INFO] [1687369617.663084]: [/mir_bridge] starting to stream messages on topic 'move_base/feedback'
**ERROR 110 CONNECTION TIMED OUT**
`

mintar, 2023-06-23:
So that output above was with ROS_MASTER_URI=http:/**localhost**:11311, correct?

Is the emergency stop deactivated?

What colors do the MiR LEDs have?

Does the MiR web interface show any errors?

What's the output of `rostopic info /cmd_vel`?

closed 2023-11-28 by mintar