DFKI-NI/mir_robot #132: Problems when sending it to goal

lagartoxa, 2023-09-08:
Hi everyone,

I followed your README, but I still can't send the Mir to a goal with Rviz. The navigation launcher keeps giving me error message:
`[ERROR] [1694195735.450105552, 843.715000000]: Aborting because a valid plan could not be found. Even after executing all recovery behaviors`

(Actually I could send the Mir to a goal only once. All other tries gave me this same error, even after restarting the launcher and even the computer)

[image]

Steps I followed:
 - I installed mir_robot by apt (sudo apt install ros-noetic-mir-robot)
 - I ran the commands from the Gazebo demo (existing map) section:
```
### gazebo:
roslaunch mir_gazebo mir_maze_world.launch
rosservice call /gazebo/unpause_physics   # or click the "start" button in the Gazebo GUI

### localization:
roslaunch mir_navigation amcl.launch initial_pose_x:=10.0 initial_pose_y:=10.0
# or alternatively: roslaunch mir_gazebo fake_localization.launch delta_x:=-10.0 delta_y:=-10.0

# navigation:
roslaunch mir_navigation start_planner.launch \
    map_file:=$(rospack find mir_gazebo)/maps/maze.yaml \
    virtual_walls_map_file:=$(rospack find mir_gazebo)/maps/maze_virtual_walls.yaml
rviz -d $(rospack find mir_navigation)/rviz/navigation.rviz
```
 - I used the 2D Nav Goal button on Rviz to send the robot to a point in the map. Output from `rostopic echo /move_base_simple/goal` :
```
WARNING: no messages received and simulated time is active.
Is /clock being published?
header: 
  seq: 6
  stamp: 
    secs: 1084
    nsecs: 125000000
  frame_id: "map"
pose: 
  position: 
    x: 15.26579475402832
    y: 3.153439521789551
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.9999401232387972
    w: 0.010943031443757401
```

[image]

Is there something else I should do to make it work? I did the same test on two different computers, both Mac M1, running Ubuntu 20.04 ARM

Best Regards

mintar, 2023-09-12:
First off, thanks for the detailed bug report.

The error message you see means that the SBPL global planner could not find a path between the current robot pose and the goal pose. For example, because there is no obstacle-free connection between the two poses, the current robot pose is in collision or the target pose is in collision with an obstacle.

However, your situation is weird. I just reproduced the same steps as you, and for me, it finds paths between all the poses you gave.

I've tested this on my Ubuntu 20.04 x86_64, while you are using ARM; however, I don't think this should matter. So I'm out of ideas, sorry.

mintar, 2023-11-28:
Closing due to lack of activity.

closed 2023-11-28 by mintar