DFKI-NI/mir_robot #133: Migrate to modern Gazebo

azeey, 2023-12-21:
As you may know, Gazebo-classic (aka Gazebo11, see [Terminology](https://gazebosim.org/about#:~:text=Terminology)) is no longer being actively developed  and will not be available on ROS Jazzy ([REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)). I noticed that this project still uses Gazebo-classic. Are there any plans to migrate to modern Gazebo ([previously known as Ignition](https://community.gazebosim.org/t/a-new-era-for-gazebo/1356))?

mintar, 2024-02-06:
This project is always going to use the Gazebo version that's required by the respective ROS release. This means (classic) Gazebo11 for ROS1 Noetic and ROS2 <= Iron, and (modern) Gazebo Harmonic for ROS2 >= Jazzy.

Note that ROS2 support is still under development (see #96). The goal is to have a working `iron` branch and a `rolling` branch which will probably be kept in sync for the time being. Those branches will be using Gazebo11, since that's the required version for these ROS releases.

At some point in the future, ROS Rolling will probably switch over to Gazebo Harmonic in preparation of the Jazzy release. At that point, the `rolling` branch in our repo should also make that switch so that we have Gazebo Harmonic support ready for the Jazzy release.

Does that make sense? Did I miss something?

P.S.: Sorry for the late reply! I just stumbled over this while cleaning up my pre-christmas email.

azeey, 2024-02-06:
Thanks @mintar. That all makes sense and it's great to hear that you plan to migrate. However, in case you are not aware, migrating to modern Gazebo will require some changes to your launch files and potentially, your SDF files. I will leave the following links for future reference.

- Migration guide for SDF files: https://gazebosim.org/api/sim/8/migrationsdf.html 
- Migration guild for Gazebo plugins: https://gazebosim.org/api/sim/8/migrationplugins.html 
- Documentation for ros_gz: https://gazebosim.org/docs/harmonic/ros2_integration. 
- https://github.com/gazebosim/ros_gz_project_template and its [documentation](https://gazebosim.org/docs/harmonic/ros_gz_project_template_guide#guide-to-code-class-codespan-ros_gz_project_template-code-for-ros-2-and-gazebo-development).

mintar, 2024-02-07:
Thanks for the links! I'll close this issue for now since I believe your question was answered.

closed 2024-02-07 by mintar