DFKI-NI/mir_robot #136: Is that works with the MiR250?

liyifei1997, 2024-02-07:
First of all, thanks so much for your repositorie, that's very helpful.

Our lab want to buy a MiR250 and would like use ROS to control it. Not sure if anyone test it?
I saw you are test in MiR 100/200/500. But I didn't see these solutions in them web. 

Thanks so much.

mintar, 2024-02-08:
I'm pretty sure it works with the MiR250. I haven't tested it myself, but from a software perspective all MiR robots are very similar.

liyifei1997, 2024-02-08:
> I'm pretty sure it works with the MiR250. I haven't tested it myself, but from a software perspective all MiR robots are very similar.

Thanks so much, that's will be great!

liesrock, 2024-05-30:
I tested on two different MiR250 and I can confirm that the repo works reliably for both of them.

liyifei1997, 2024-05-30:
> I tested on two different MiR250 and I can confirm that the repo works reliably for both of them.

Thanks so much!

closed 2024-05-31 by mintar